patjak/facetimehd #104: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2

nicolone, 2016-10-20:
When the driver is loaded via ```modprobe facetimehd``` the following error is generated:
```Direct firmware load for facetimehd/1871_01XX.dat failed with error -2```

More details on my system:
```
$ sudo dmidecode -s system-product-name
MacBookPro11,5
$ uname -r
4.2.0-42-generic
$ ls -la /lib/firmware/facetimehd/
total 1428
drwxr-xr-x  2 root root    4096 Okt 20 16:05 .
drwxr-xr-x 73 root root   24576 Jul 20 12:04 ..
-rw-r--r--  1 root root 1425412 Okt 20 16:05 firmware.bin
$ lsmod | grep bdc_pci
$ lsmod | grep facetimehd
facetimehd             98304  0
videobuf2_dma_sg       20480  1 facetimehd
videobuf2_core         49152  1 facetimehd
videodev              172032  3 v4l2_common,facetimehd,videobuf2_core
```

The full dmesg log:
```
$ dmesg
...
[Okt20 16:38] media: Linux media interface: v0.10
[  +0,002931] Linux video capture interface: v2.00
[  +0,003979] facetimehd 0000:05:00.0: Found FaceTime HD camera with device id: 1570
[  +0,000066] facetimehd 0000:05:00.0: Setting 64bit DMA mask
[  +0,005036] facetimehd 0000:05:00.0: S2 PCIe link init succeeded
[  +0,000028] facetimehd 0000:05:00.0: Refclk: 25MHz (0xa)
[  +0,009943] facetimehd 0000:05:00.0: PLL reset finished
[  +0,000002] facetimehd 0000:05:00.0: Waiting for S2 PLL to lock at 450 MHz
[  +0,000012] facetimehd 0000:05:00.0: S2 PLL is locked after 10 us
[  +0,009940] facetimehd 0000:05:00.0: S2 PLL is in bypass mode
[  +0,019899] facetimehd 0000:05:00.0: DDR40 PHY PLL locked on safe settings
[  +0,000014] facetimehd 0000:05:00.0: STRAP valid
[  +0,000001] facetimehd 0000:05:00.0: Configuring DDR PLLs for 450 MHz
[  +0,000007] facetimehd 0000:05:00.0: DDR40 PLL is locked after 0 us
[  +0,000011] facetimehd 0000:05:00.0: First DDR40 VDL calibration completed after 2 us
[  +0,000011] facetimehd 0000:05:00.0: Second DDR40 VDL calibration completed after 2 us
[  +0,000001] facetimehd 0000:05:00.0: Using step size 150
[  +0,000005] facetimehd 0000:05:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[  +0,000005] facetimehd 0000:05:00.0: Virtual VTT enabled
[  +0,020379] facetimehd 0000:05:00.0: S2 DRAM memory address: 0x22159559
[  +0,000014] facetimehd 0000:05:00.0: Rewrite DDR mode registers succeeded
[  +0,000199] facetimehd 0000:05:00.0: Full memory verification succeeded! (0)
[  +0,464180] facetimehd 0000:05:00.0: Loaded firmware, size: 1392kb
[  +0,140572] facetimehd 0000:05:00.0: ISP woke up after 0ms
[  +0,000007] facetimehd 0000:05:00.0: Number of IPC channels: 7, queue size: 44865
[  +0,000003] facetimehd 0000:05:00.0: Firmware requested heap size: 3072kb
[  +0,009948] facetimehd 0000:05:00.0: ISP second int after 0ms
[  +0,000002] facetimehd 0000:05:00.0: Channel description table at 00800000
[  +0,010104] facetimehd 0000:05:00.0: magic value: 00000000 after 0 ms
[  +0,000002] facetimehd 0000:05:00.0: Enabling interrupts
[  +0,000150] FWMSG: 
[  +0,000023] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[  +0,000011] FWMSG: 
[  +0,000013] FWMSG:   Restart count : 0
[  +0,000096] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[  +0,000016] FWMSG:   System Clock : 200000000 Hz
[  +0,000014] FWMSG:   Processor mode : SUPERVISOR
[  +0,000016] FWMSG:   Cache architecture type : SEPARATE
[  +0,000014] FWMSG:   Cache type : WRITEBACK
[  +0,000014] FWMSG:   Data Cache Line : 32 [0x20]
[  +0,000015] FWMSG:   Boot arguments entries : 2
[  +0,000082] FWMSG:     0000: 0x00000000 0x00000000
[  +0,000021] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[  +0,000015] FWMSG:   Main memory :   base : 0x00000000
[... 86 lines cut ...]
[  +0,000004]  [<ffffffffc1518a44>] __vb2_queue_alloc+0x164/0x530 [videobuf2_core]
[  +0,000002]  [<ffffffffc1519008>] __reqbufs.isra.20+0x1f8/0x360 [videobuf2_core]
[  +0,000002]  [<ffffffffc151a6ac>] vb2_ioctl_reqbufs+0x6c/0xa0 [videobuf2_core]
[  +0,000005]  [<ffffffffc1540b63>] v4l_reqbufs+0x43/0x50 [videodev]
[  +0,000003]  [<ffffffffc153f7fd>] __video_do_ioctl+0x29d/0x310 [videodev]
[  +0,000003]  [<ffffffff811e1fbb>] ? __kmalloc+0x1cb/0x270
[  +0,000003]  [<ffffffffc153f2c6>] video_usercopy+0x336/0x5b0 [videodev]
[  +0,000002]  [<ffffffffc153f560>] ? video_ioctl2+0x20/0x20 [videodev]
[  +0,000003]  [<ffffffff810b0ab9>] ? update_curr+0x79/0x160
[  +0,000002]  [<ffffffff81242476>] ? fsnotify+0x316/0x4a0
[  +0,000001]  [<ffffffff810bea88>] ? __wake_up+0x48/0x60
[  +0,000003]  [<ffffffffc153f555>] video_ioctl2+0x15/0x20 [videodev]
[  +0,000002]  [<ffffffffc153a681>] v4l2_ioctl+0xb1/0xe0 [videodev]
[  +0,000003]  [<ffffffff81214205>] do_vfs_ioctl+0x295/0x480
[  +0,000001]  [<ffffffff81214469>] SyS_ioctl+0x79/0x90
[  +0,000003]  [<ffffffff817fa072>] entry_SYSCALL_64_fastpath+0x16/0x75
[  +0,000001] Code: ff ff ff 48 8b 73 10 8b 53 40 48 8b 7b 30 31 c9 4c 8b 45 b0 41 b9 d0 00 00 00 e8 c8 c7 f5 bf 85 c0 0f 85 47 01 00 00 48 8b 45 a8 <48> 8b 38 e8 24 20 0a c0 49 89 c5 48 89 03 8b 43 1c 4d 85 ed 44 
[  +0,000013] RIP  [<ffffffffc14800f4>] vb2_dma_sg_alloc+0x194/0x10a0 [videobuf2_dma_sg]
[  +0,000002]  RSP <ffff8803b0d9baf8>
[  +0,000000] CR2: 0000000000096000
[  +0,000002] ---[ end trace be3fed7893e5f349 ]---
```

Please contact me if you need further information to solve this issue.
Thanks for your work!

patjak, 2016-10-20:
That firmware error is harmless. You're missing the sensor calibration files but there is unfortunately no support for extracting them at the moment. The problem seems to be that your kernel is too old. If possible try upgrading to at least 4.4 and see if that helps.

seumasmac, 2016-10-21:
Is 4.4 the oldest kernel that works? The wiki says:

> Since this driver depends on vb2_dma_sg_init_ctx, it requires Linux kernel 3.19 or later.

so we should probably update it if the driver requires something a lot newer.

nicolone, 2016-10-23:
Upgrading to kernel 4.4 fixed the problem.
Thanks!

closed 2016-10-23 by nicolone

dcazdavi, 2018-06-28:
I'm seeing this same time and I'm running kernel 4.17.   How else can i test?

`<pre><font color="#4E9A06">[ 9648.712467] </font><font color="#C4A000">Linux video capture interface</font>: v2.00
<font color="#4E9A06">[ 9648.736927] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Found FaceTime HD camera with device id: 1570
<font color="#4E9A06">[ 9648.737225] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Setting 64bit DMA mask
<font color="#4E9A06">[ 9648.743066] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: S2 PCIe link init succeeded
<font color="#4E9A06">[ 9648.743098] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Refclk: 25MHz (0xa)
<font color="#4E9A06">[ 9648.753124] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: PLL reset finished
<font color="#4E9A06">[ 9648.753126] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Waiting for S2 PLL to lock at 450 MHz
<font color="#4E9A06">[ 9648.753139] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: S2 PLL is locked after 10 us
<font color="#4E9A06">[ 9648.763172] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: S2 PLL is in bypass mode
<font color="#4E9A06">[ 9648.783245] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: DDR40 PHY PLL locked on safe settings
<font color="#4E9A06">[ 9648.783258] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: STRAP valid
<font color="#4E9A06">[ 9648.783259] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Configuring DDR PLLs for 450 MHz
<font color="#4E9A06">[ 9648.783265] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: DDR40 PLL is locked after 0 us
<font color="#4E9A06">[ 9648.783274] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: First DDR40 VDL calibration completed after 2 us
<font color="#4E9A06">[ 9648.783284] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Second DDR40 VDL calibration completed after 2 us
<font color="#4E9A06">[ 9648.783285] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Using step size 152
<font color="#4E9A06">[ 9648.783289] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: VDL set to: coarse=0x10008, fine=0x1011a
<font color="#4E9A06">[ 9648.783294] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Virtual VTT enabled
<font color="#4E9A06">[ 9648.803852] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: S2 DRAM memory address: 0x22159559
<font color="#4E9A06">[ 9648.803864] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Rewrite DDR mode registers succeeded
<font color="#4E9A06">[ 9648.804058] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Full memory verification succeeded! (0)
<font color="#4E9A06">[ 9649.267961] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Loaded firmware, size: 1392kb
<font color="#4E9A06">[ 9649.310953] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: ISP woke up after 0ms
<font color="#4E9A06">[ 9649.310958] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Number of IPC channels: 7, queue size: 44865
<font color="#4E9A06">[ 9649.310961] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Firmware requested heap size: 3072kb
<font color="#4E9A06">[ 9649.320989] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: ISP second int after 0ms
<font color="#4E9A06">[ 9649.320991] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Channel description table at 00800000
<font color="#4E9A06">[ 9649.331172] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: magic value: 00000000 after 0 ms
<font color="#4E9A06">[ 9649.331173] </font><font color="#C4A000">facetimehd 0000:02:00.0</font>: Enabling interrupts
<font color="#4E9A06">[ 9649.331344] </font><font color="#C4A000">FWMSG</font>: 
<font color="#4E9A06">[ 9649.331364] </font><font color="#C4A000">FWMSG</font>: - APOLLO-ISP-APPLE ------------------------------------------------------------
<font color="#4E9A06">[ 9649.331374] </font><font color="#C4A000">FWMSG</font>: 
<font color="#4E9A06">[ 9649.331385] </font><font color="#C4A000">FWMSG</font>:   Restart count : 0
<font color="#4E9A06">[ 9649.331405] </font><font color="#C4A000">FWMSG</font>:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
<font color="#4E9A06">[ 9649.331493] </font><font color="#C4A000">FWMSG</font>:   System Clock : 200000000 Hz
<font color="#4E9A06">[ 9649.331506] </font><font color="#C4A000">FWMSG</font>:   Processor mode : SUPERVISOR
<font color="#4E9A06">[ 9649.331520] </font><font color="#C4A000">FWMSG</font>:   Cache architecture type : SEPARATE
<font color="#4E9A06">[ 9649.331533] </font><font color="#C4A000">FWMSG</font>:   Cache type : WRITEBACK
<font color="#4E9A06">[ 9649.331546] </font><font color="#C4A000">FWMSG</font>:   Data Cache Line : 32 [0x20]
<font color="#4E9A06">[ 9649.331558] </font><font color="#C4A000">FWMSG</font>:   Boot arguments entries : 2
<font color="#4E9A06">[ 9649.331571] </font><font color="#C4A000">FWMSG</font>:     0000: 0x00000000 0x00000000
<font color="#4E9A06">[ 9649.331633] </font><font color="#C4A000">FWMSG</font>:   Physical memory base : 0x00000000 [TLB base 0x00160000]
<font color="#4E9A06">[ 9649.331647] </font><font color="#C4A000">FWMSG</font>:   Main memory :   base : 0x00000000
[... cut ...]
<font color="#4E9A06">[ 9649.405825] </font><font color="#C4A000">FWMSG</font>: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
<font color="#4E9A06">[ 9649.406053] </font><font color="#C4A000">FWMSG</font>: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
<font color="#4E9A06">[ 9649.406371] </font><font color="#C4A000">facetimehd 0000:02:00.0</font><b>: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2</b>
[moi@platita bcwc_pcie]$ uname -a
Linux platita 4.17.2-200.fc28.x86_64 #1 SMP Mon Jun 18 20:09:31 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
</pre>`

Tlanextic, 2021-07-26:
I am running KDE Neon 5.22.3 with kernel 5.8.0-63 generic based on Ubuntu. I spend some hours and reboots to get the camera to work but now no avail. Please help a struggling n00b. :-P

iDevPro, 2024-04-26:
This issue with kernel 6.6 also :)

nthdat09, 2024-10-25:
[image]

Does anyone have solution with this problem?

Tislom58, 2025-01-10:
Same issue here, seems like the new kernel is breaking it