patjak/facetimehd #105: Issue with web video chat

gleachkr, 2016-11-29:
I'm not certain that this is an issue with bcwc_pcie, but I thought I should report it in case one of you is able to reproduce. I've installed the driver, and while it works well outside the browser, web.skype.com and Riot.im both do not function at the moment, on either firefox or chromium. Chromium throws the following error in both cases (on entering the site with skype, and on starting a video chat with riot):

`[18328:18350:1129/093004:ERROR:video_capture_device_factory_linux.cc(132)] Not implemented reached in void media::GetSupportedFormatsForV4L2BufferType(int, media::VideoCaptureFormats*)`

Thank you for your work on this awesome project. If you're unable to reproduce, I can send more details. Or if this has nothing to do with the driver, apologies for the red herring.

derjohn, 2017-04-11:
@patjak Is this issue orphaned?

I get the same error on chorme:

[13142:13198:0411/223335:ERROR:video_capture_device_factory_linux.cc(108)] Not implemented reached in void media::GetSupportedFormatsForV4L2BufferType(int, media::VideoCaptureFormats*)

Video mith e.g. https://meet.jit.si/patjak doesnt work.

patjak, 2017-04-11:
Looks like an issue with webrtc. Our driver supports step-wise sizes (fast hardware scaling) so it might get confused by that. Do you know what format the browser is asking for?

derjohn, 2017-04-11:
@patjak No really , but I tried https://test.webrtc.org/

Check resolution 320x240
Check resolution 640x480
Check resolution 1280x720
Check supported resolutions

==> All fail!

derjohn, 2017-04-11:
And I see in chromium log:

[1:30:0411/232440:ERROR:webrtcsession.cc(1226)] ConnectDataChannel called when data_channel_ is NULL.

derjohn, 2017-04-11:
I just found: https://webrtchacks.github.io/WebRTC-Camera-Resolution/

chrome 53	camera 1	4K(UHD)	16:9	3840x2160	0x0	fail: ConstraintNotSatisfiedError
chrome 53	camera 1	1080p(FHD)	16:9	1920x1080	0x0	fail: ConstraintNotSatisfiedError
chrome 53	camera 1	UXGA	4:3	1600x1200	0x0	fail: ConstraintNotSatisfiedError
chrome 53	camera 1	720p(HD)	16:9	1280x720	0x0	fail: ConstraintNotSatisfiedError
chrome 53	camera 1	SVGA	4:3	800x600	0x0	fail: ConstraintNotSatisfiedError
chrome 53	camera 1	VGA	4:3	640x480	0x0	fail: ConstraintNotSatisfiedError
chrome 53	camera 1	360p(nHD)	16:9	640x360	0x0	fail: ConstraintNotSatisfiedError
chrome 53	camera 1	CIF	4:3	352x288	0x0	fail: ConstraintNotSatisfiedError
chrome 53	camera 1	QVGA	4:3	320x240	0x0	fail: ConstraintNotSatisfiedError
chrome 53	camera 1	QCIF	4:3	176x144	0x0	fail: ConstraintNotSatisfiedError
chrome 53	camera 1	QQVGA	4:3	160x120	0x0	fail: ConstraintNotSatisfiedError

derjohn, 2017-04-12:
@patjak Can I do anything else to support you in finding the issue?

derjohn, 2017-05-03:
Another note: Google Hangout works with it.

derjohn, 2017-06-12:
Hello,

google hangouts with Chromium works here, Jitsi and appear.in began to work some days ago.

I use Linux Mint 18 with a 4.4.0-21-generic (I would tend to stay away from lowlatency variants, as it breaks at least the wifi broadcom wl driver).

Best Regards,
Andreas

ivanperez-keera, 2019-06-29:
@gleachkr @derjohn This issue is very old, but remains open. Video chat works for me on e.g., Chromium  with Hangouts.

@derjohn reports that it is working for him.

Is this still an issue and, if not, can we close this?

gleachkr, 2019-07-09:
Video chat now works for me on riot.im and jit.si, in both Chromium and Firefox.

Also noticed that the video tests on https://test.webrtc.org/ now appear to pass. I'd be OK with closing the issue.

ivanperez-keera, 2019-07-10:
Great! Well, I cannot close it myself, but I'd suggest to just close it if it now works. If the issue persists for someone else, I'm sure they'll open it or file a new one.

gleachkr, 2019-07-10:
Done!

closed 2019-07-10 by gleachkr