patjak/facetimehd #107: Installation not going through in Ubuntu 16.04...

mrinalsaurabh, 2017-01-11:
When I do make install in bcwc_pcie directory I get the following error:

> make -C /lib/modules/4.4.0-57-generic/build M=/home/calypso/bcwc_pcie modules_install
> make[1]: Entering directory '/usr/src/linux-headers-4.4.0-57-generic'
>   INSTALL /home/calypso/bcwc_pcie/facetimehd.ko
> At main.c:222:
> - SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
> - SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
> sign-file: certs/signing_key.pem: No such file or directory
>   DEPMOD  4.4.0-57-generic
> make[1]: Leaving directory '/usr/src/linux-headers-4.4.0-57-generic'

Please help!

xgnata, 2017-01-14:
The errors says that the module is not signed.
That is not an issue and it does not prevent the driver from working correctly.

Modules can be signed to guarantee that thew don't contain malicious code (kind of..). There is no need for  bcwc to be signed at this stage. it is open source and you compile it yourself. Therefore just trust it :)

mrinalsaurabh, 2017-01-14:
How do I get ahead with the installation and ignore this error, anyhow?? The camera is not working even after the installation.

js-robinson, 2017-01-17:
Try: `sudo depmod -a` `sudo modprobe facetimehd`

mrinalsaurabh, 2017-01-24:
tried it.. now it says on $ sudo depmod -a sudo modprobe facetimehd
depmod: ERROR: Bad version passed sudo

hobarrera, 2017-01-24:
It's:

    sudo depmod -a
    sudo modprobe facetimehd

There's a line jump in the middle that got formatted away.

xgnata, 2017-01-24:
You probably need to start with 
sudo modprobe -r bdc_pci

Be careful using sudo if you have no idea what you're doing with the command line : You can easily ruin your system.

mrinalsaurabh, 2017-01-26:
The commads did not give any error, but the webcam was still not detected. :( 
Is it time to give up already?

xgnata, 2017-01-26:
sudo modprobe -r bdc_pci
sudo depmod -a
sudo modprobe facetimehd

and post the result of
dmesg (it is a command. Just type dmesg after completion of sudo modprobe facetimehd)
The display of dmesg is very long. We are only interested in let's say the last page. only the lines talking about facetimehd.

mrinalsaurabh, 2017-01-26:
I did a 
dmesg | grep facetime 
and got following. Does this suffice?
[   90.065067] facetimehd: module verification failed: signature and/or required key missing - tainting kernel

Stono, 2017-02-27:
The taining kernel doesn't really matter; that's because it's an unsigned mod.
You can create an entry in `/etc/modules-load.d` to have it load at boot.

Check my gist `https://gist.github.com/Stono/990ea9f0b3c41606c292f00382d421bf`

rufusshrestha, 2019-04-09:
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-47-generic'
make -C /lib/modules/4.15.0-47-generic/build M=/home/rshrestha/src-public/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-47-generic'
  INSTALL /home/rshrestha/src-public/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: bss_file.c:175
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: bss_file.c:178
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-47-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-47-generic'

checkInstall dkms did not work too for me.

After doing:
sudo modprobe -r bdc_pci
sudo depmod -a
sudo modprobe facetimehd

I get below in dmesg

[   67.521317] audit: type=1107 audit(1554777993.014:44): pid=1164 uid=106 auid=4294967295 ses=4294967295 msg='apparmor="DENIED" operation="dbus_method_call"  bus="system" path="/" interface="org.freedesktop.DBus.ObjectManager" member="GetManagedObjects" mask="send" name="org.bluez" pid=4006 label="snap.mailspring.mailspring" peer_pid=1247 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=106 hostname=? addr=? terminal=?'
[   68.508586] audit: type=1400 audit(1554777993.997:45): apparmor="DENIED" operation="chmod" profile="snap.mailspring.mailspring" name="/home/rshrestha/snap/mailspring/common/" pid=4006 comm="mailspring" requested_mask="w" denied_mask="w" fsuid=1000 ouid=1000
[  458.529146] facetimehd: Unknown symbol vb2_queue_init (err 0)
[  458.529277] facetimehd: Unknown symbol vb2_dma_sg_memops (err 0)
[  458.529438] facetimehd: Unknown symbol video_ioctl2 (err 0)
[  458.529571] facetimehd: Unknown symbol vb2_ioctl_streamoff (err 0)
[  458.529726] facetimehd: Unknown symbol vb2_ioctl_expbuf (err 0)
[  458.529857] facetimehd: Unknown symbol vb2_ops_wait_finish (err 0)
[  458.529972] facetimehd: Unknown symbol v4l2_ctrl_handler_init_class (err 0)
[  458.530102] facetimehd: Unknown symbol vb2_ioctl_querybuf (err 0)
[  458.530222] facetimehd: Unknown symbol v4l2_fh_open (err 0)
[  458.530376] facetimehd: Unknown symbol video_devdata (err 0)
[  458.530472] facetimehd: Unknown symbol vb2_fop_release (err 0)
[  458.530592] facetimehd: Unknown symbol vb2_plane_cookie (err 0)
[  458.530692] facetimehd: Unknown symbol vb2_ioctl_dqbuf (err 0)
[  458.530849] facetimehd: Unknown symbol vb2_ioctl_create_bufs (err 0)
[  458.530973] facetimehd: Unknown symbol vb2_buffer_done (err 0)
[  458.531082] facetimehd: Unknown symbol v4l2_ctrl_subscribe_event (err 0)
[  458.531192] facetimehd: Unknown symbol video_unregister_device (err 0)
[  458.531313] facetimehd: Unknown symbol vb2_ioctl_qbuf (err 0)
[  458.531413] facetimehd: Unknown symbol vb2_fop_mmap (err 0)
[  458.531525] facetimehd: Unknown symbol video_device_alloc (err 0)
[  458.531635] facetimehd: Unknown symbol vb2_fop_read (err 0)
[  458.531724] facetimehd: Unknown symbol v4l2_device_register (err 0)
[  458.531871] facetimehd: Unknown symbol __video_register_device (err 0)
[  458.531969] facetimehd: Unknown symbol vb2_ops_wait_prepare (err 0)
[  458.532082] facetimehd: Unknown symbol vb2_ioctl_streamon (err 0)
[  458.532174] facetimehd: Unknown symbol vb2_fop_poll (err 0)
[  458.532265] facetimehd: Unknown symbol v4l2_ctrl_new_std (err 0)
[  458.532352] facetimehd: Unknown symbol v4l2_ctrl_handler_free (err 0)
[  458.532445] facetimehd: Unknown symbol v4l2_device_unregister (err 0)
[  458.532514] facetimehd: Unknown symbol vb2_wait_for_all_buffers (err 0)
[  458.532587] facetimehd: Unknown symbol video_device_release (err 0)
[  458.532674] facetimehd: Unknown symbol v4l2_event_unsubscribe (err 0)
[  458.532746] facetimehd: Unknown symbol vb2_ioctl_reqbufs (err 0)
[  750.034538] facetimehd: Unknown symbol vb2_queue_init (err 0)
[  750.034599] facetimehd: Unknown symbol vb2_dma_sg_memops (err 0)
[  750.034685] facetimehd: Unknown symbol video_ioctl2 (err 0)
[  750.034780] facetimehd: Unknown symbol vb2_ioctl_streamoff (err 0)
[  750.034845] facetimehd: Unknown symbol vb2_ioctl_expbuf (err 0)
[  750.034927] facetimehd: Unknown symbol vb2_ops_wait_finish (err 0)
[  750.034990] facetimehd: Unknown symbol v4l2_ctrl_handler_init_class (err 0)
[  750.035059] facetimehd: Unknown symbol vb2_ioctl_querybuf (err 0)
[  750.035132] facetimehd: Unknown symbol v4l2_fh_open (err 0)
[  750.035227] facetimehd: Unknown symbol video_devdata (err 0)
[  750.035287] facetimehd: Unknown symbol vb2_fop_release (err 0)
[  750.035355] facetimehd: Unknown symbol vb2_plane_cookie (err 0)
[  750.035418] facetimehd: Unknown symbol vb2_ioctl_dqbuf (err 0)
[  750.035480] facetimehd: Unknown symbol vb2_ioctl_create_bufs (err 0)
[  750.035550] facetimehd: Unknown symbol vb2_buffer_done (err 0)
[  750.035606] facetimehd: Unknown symbol v4l2_ctrl_subscribe_event (err 0)
[  750.035668] facetimehd: Unknown symbol video_unregister_device (err 0)
[  750.035728] facetimehd: Unknown symbol vb2_ioctl_qbuf (err 0)
[  750.035785] facetimehd: Unknown symbol vb2_fop_mmap (err 0)
[  750.035848] facetimehd: Unknown symbol video_device_alloc (err 0)
[  750.035914] facetimehd: Unknown symbol vb2_fop_read (err 0)
[  750.036078] facetimehd: Unknown symbol v4l2_device_register (err 0)
[  750.036273] facetimehd: Unknown symbol __video_register_device (err 0)
[  750.036356] facetimehd: Unknown symbol vb2_ops_wait_prepare (err 0)
[  750.036453] facetimehd: Unknown symbol vb2_ioctl_streamon (err 0)
[... cut ...]
[  750.036670] facetimehd: Unknown symbol v4l2_ctrl_handler_free (err 0)
[  750.036752] facetimehd: Unknown symbol v4l2_device_unregister (err 0)
[  750.036837] facetimehd: Unknown symbol vb2_wait_for_all_buffers (err 0)
[  750.036918] facetimehd: Unknown symbol video_device_release (err 0)
[  750.037016] facetimehd: Unknown symbol v4l2_event_unsubscribe (err 0)
[  750.037121] facetimehd: Unknown symbol vb2_ioctl_reqbufs (err 0)
[  838.041273] media: Linux media interface: v0.10
[  838.049572] Linux video capture interface: v2.00

rauferd, 2025-05-14:
If you want to sign your module (or just want the error message to go away), check out this discussion on StackExchange/superuser for a clean solution:
[](https://superuser.com/questions/1214116/no-openssl-sign-file-signing-key-pem-leads-to-error-while-loading-kernel-modules/1322832#1322832)

Alternatively, you might already have a signing key, but not in the proper location. On my system, I have to copy keys from `/usr/src/linux-headers-...-common` to `/usr/src/linux-headers-...-amd64` to make the error disappear.

Seems to be a duplicate of #244 and #134 , too.