patjak/facetimehd #108: GBP Error: not a valid treeish

romulorosa, 2017-01-14:
Hi,

I'm trying to install and got this error when a run gbp command:
```
romulofurtado➜/tmp/bcwc_pcie(debian)» gbp buildpackage -us -uc --git-debian-branch=debian                                        [18:39:21]
gbp:error: upstream/0.0.1_git20161107-0712f39 is not a valid treeish
```

Any ideas how to solve this problem?

Thanks!

seumasmac, 2017-01-20:
I don't know, but http://unix.stackexchange.com/questions/167533/what-does-gbperror-upstream-1-5-13-is-not-a-valid-treeish-mean might be of help!

mlopes, 2017-03-13:
Having the same issue, I'm on the debian branch, and have the `debian/` folder.  It looks like it's getting the branch from somewhere else.

bor8, 2018-04-16:
Add --git-upstream-tree=_BRANCH_ to your command. Something like

```
gbp buildpackage --git-debian-branch="ubuntu/xenial" --git-ignore-new --git-upstream-tree="ubuntu/xenial" -uc -us || exit 1
```

smoe, 2019-08-02:
I expect that error when the version stated in debian/changelog does not match with the version tagged in git. This easily happens when new releases are uploaded with "git push --all" but are then not followed by a "git push --tags".