patjak/facetimehd #111: ELEMENTARY

pierre-weiss, 2017-04-12:


derjohn, 2017-04-13:
Hello,
is this an issue or an success report? :)

The error "Direct firmware load for facetimehd/1871_01XX.dat failed with error -2" is harmless.

I just wonder, why you didn't need to sudo  before "depmod -a" ..

rgds,
j

pierre-weiss, 2017-04-13:
This is a success :). I first wrote an issue, then double checked the
threads and remarked it was in fact working after adding the depmod
commands. All was done after sudo su, so that I forgot adding it in my
thread.

I wanted to change the title or even discard the issue, but I couldn't...

My main intention was to advertise that Elementary OS could be done
following the guidelines for Ubuntu + the depmod commands...

Thanks for the hard work,
Very best regards,
Pierre Weiss

patjak, 2017-07-06:
Thanks for letting us know. Closing.

closed 2017-07-06 by patjak