patjak/facetimehd #112: MacBookPro13,1 sensorid missing

christophgysin, 2017-04-22:
I have a MacBookPro13,1 with the following camera:
```
$ sudo lspci -vnn -s 03:00.0
03:00.0 Multimedia controller [0480]: Broadcom Limited 720p FaceTime HD Camera [14e4:1570]
        Subsystem: Broadcom Limited 720p FaceTime HD Camera [14e4:1570]
        Flags: bus master, fast devsel, latency 0, IRQ 38
        Memory at 92600000 (64-bit, non-prefetchable) [size=64K]
        Memory at c0000000 (64-bit, prefetchable) [size=256M]
        Memory at 92500000 (64-bit, non-prefetchable) [size=1M]
        Capabilities: [48] Power Management version 3
        Capabilities: [58] MSI: Enable+ Count=1/1 Maskable- 64bit+
        Capabilities: [68] Vendor Specific Information: Len=44 <?>
        Capabilities: [ac] Express Endpoint, MSI 00
        Capabilities: [100] Advanced Error Reporting
        Capabilities: [13c] Device Serial Number 00-00-00-ff-ff-00-00-00
        Capabilities: [150] Power Budgeting <?>
        Capabilities: [160] Virtual Channel
        Capabilities: [1b0] Latency Tolerance Reporting
        Capabilities: [220] #15
        Kernel driver in use: facetimehd
        Kernel modules: bdc_pci
```
Loading facetimehd fails with:
```
[  185.040410] no set file for sensorid 0005 0248 found
```
Simply commenting out the check makes the camera work.

Could we add the sensor ID to the driver?

Looking at the code, those IDs seem to be mapped to facetimehd/????_????.dat files, that are used for calibration. Where can I find those files? How would I create one for my camera?

patjak, 2017-04-25:
Thanks for the report. Yes we shouldn't fatally fail if the sensor id is unknown. I've pushed a patch to the mainline branch that will fix this. I saw your patch but I prefer not to quirk on the sensor id but instead allow it to fail (like when the sensor settings file is not found). The sensor settings files live inside the driver binary (much like the firmware) but we have no tool to extract them yet.

patjak, 2019-12-10:
Fixed in https://github.com/patjak/bcwc_pcie/pull/188

closed 2019-12-10 by patjak