patjak/facetimehd #116: Crashes with recent linux

marcusramberg, 2017-05-10:
Seems like the stability of this driver has become worse since I've upgraded my laptop to 4.10.0 - Getting crashes like this now:

[  970.353969] brcmfmac 0000:04:00.0: Direct firmware load for brcm/brcmfmac43602-pcie.txt failed with error -2
[  970.377196] facetimehd 0000:05:00.0: Failed to allocate resource (size: 12, start: -118091341585152, end: -118091341585088)
[  970.377224] BUG: unable to handle kernel NULL pointer dereference at           (null)
[  970.377264] IP: isp_mem_create+0x86/0xf0 [facetimehd]
[  970.377283] PGD 0 

[  970.377299] Oops: 0000 [#1] SMP
[  970.377312] Modules linked in: facetimehd(OE) videobuf2_dma_sg videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media rfcomm xt_CHECKSUM iptable_mangle ipt_REJECT nf_reject_ipv4 xt_tcpudp ipt_MASQUERADE nf_nat_masquerade_ipv4 xfrm_user xfrm_algo iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype xt_conntrack nf_nat nf_conntrack libcrc32c br_netfilter aufs pci_stub vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) bridge stp llc ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter cmac bnep binfmt_misc nls_iso8859_1 zfs(PO) dm_crypt zunicode(PO) zavl(PO) zcommon(PO) znvpair(PO) spl(O) joydev btusb btrtl btbcm btintel snd_hda_codec_cirrus snd_hda_codec_generic snd_hda_codec_hdmi bluetooth input_leds bcm5974 snd_hda_intel snd_hda_codec snd_hda_core snd_hwdep
[  970.377574]  intel_rapl x86_pkg_temp_thermal intel_powerclamp snd_pcm coretemp kvm_intel kvm irqbypass applesmc crct10dif_pclmul input_polldev crc32_pclmul snd_seq_midi ghash_clmulni_intel pcbc snd_seq_midi_event aesni_intel aes_x86_64 crypto_simd glue_helper snd_rawmidi cryptd brcmfmac intel_cstate snd_seq snd_seq_device thunderbolt brcmutil snd_timer intel_rapl_perf intel_pch_thermal cfg80211 snd mei_me lpc_ich mei soundcore shpchp sbs acpi_als kfifo_buf sbshc industrialio apple_gmux mac_hid apple_bl parport_pc ppdev lp parport ip_tables x_tables autofs4 amdgpu hid_generic hid_apple usbhid hid uas usb_storage amdkfd amd_iommu_v2 radeon i2c_algo_bit ttm drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops ahci libahci drm fjes video [last unloaded: bdc_pci]
[  970.377821] CPU: 7 PID: 5234 Comm: AudioThread Tainted: P        W  OE   4.10.0-20-generic #22-Ubuntu
[  970.377857] Hardware name: Apple Inc. MacBookPro11,5/Mac-06F11F11946D27C5, BIOS MBP114.88Z.0172.B13.1612230252 12/23/2016
[  970.377900] task: ffff949996adad00 task.stack: ffffb9558ad1c000
[  970.377927] RIP: 0010:isp_mem_create+0x86/0xf0 [facetimehd]
[  970.377950] RSP: 0018:ffffb9558ad1fcc0 EFLAGS: 00010202
[  970.377971] RAX: ffff94975df53401 RBX: 0000000000000000 RCX: 00000000000316b9
[  970.377998] RDX: 00000000000316b8 RSI: ffff9499ff3dc5e0 RDI: ffff9499ee8037c0
[  970.378026] RBP: ffffb9558ad1fcf8 R08: 000000000001c5e0 R09: ffffffffc10380c1
[  970.378054] R10: ffffea2dc777d4c0 R11: ffffffff8724882d R12: ffff9498b788ad80
[  970.378081] R13: 000000000000000c R14: ffff949793853000 R15: 0000000000000005
[  970.378111] FS:  00007fc60403c700(0000) GS:ffff9499ff3c0000(0000) knlGS:0000000000000000
[  970.378142] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  970.378165] CR2: 0000000000000000 CR3: 000000040f4f5000 CR4: 00000000001406e0
[  970.378192] Call Trace:
[  970.378208]  fthd_isp_cmd+0x67/0x320 [facetimehd]
[  970.378228]  fthd_isp_cmd_channel_stop+0x45/0x80 [facetimehd]
[  970.378253]  fthd_stop_channel+0x12/0x70 [facetimehd]
[  970.378275]  fthd_stop_streaming+0x1d/0xc0 [facetimehd]
[  970.378298]  __vb2_queue_cancel+0x2b/0x1a0 [videobuf2_core]
[  970.378321]  vb2_core_queue_release+0x20/0x50 [videobuf2_core]
[  970.378346]  _vb2_fop_release+0x83/0xa0 [videobuf2_v4l2]
[  970.378369]  vb2_fop_release+0x2a/0x40 [videobuf2_v4l2]
[  970.378395]  v4l2_release+0x38/0x80 [videodev]
[  970.378415]  __fput+0xe7/0x220
[  970.378429]  ____fput+0xe/0x10
[  970.378445]  task_work_run+0x80/0xa0
[  970.378462]  exit_to_usermode_loop+0xaa/0xb0
[  970.378481]  syscall_return_slowpath+0x59/0x60
[  970.378501]  entry_SYSCALL_64_fastpath+0xab/0xad

missinglink, 2017-06-12:
same error here:

google hangouts has stopped detecting the camera (possibly a hangouts issue?) 

the `mplayer tv://` command works fine, can confirm the camera works there, so presumably, the driver has been loaded correctly.

```bash
Jun 12 12:45:35 peterpro kernel: [  167.033000] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
Jun 12 12:46:46 peterpro systemd-modules-load[236]: Inserted module 'facetimehd'
Jun 12 12:46:46 peterpro kernel: [    4.681947] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
Jun 12 12:46:46 peterpro kernel: [    4.682267] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
Jun 12 12:46:46 peterpro kernel: [    4.682420] facetimehd 0000:02:00.0: Setting 64bit DMA mask
Jun 12 12:46:46 peterpro kernel: [    4.688328] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
```

hobarrera, 2017-06-12:
It's working fine for me on 4.11.3. Is this the latest `master`?

marcusramberg, 2021-07-20:
Gonna close this as it's quite obsolete

closed 2021-07-20 by marcusramberg