patjak/facetimehd #118: Make error when compiling bcwc_pcie

arun-gurung, 2017-05-23:
I have successfully extracted and installed the firmware thanks to UnaiM. I`m following the instructions from the Get Started article, but when I run `make` I get errors:

[root@macbook-air bcwc_pcie]# make
make -C /lib/modules/3.10.0-514.16.1.el7.x86_64/build M=/tmp/bcwc_pcie modules
make[1]: Entering directory `/usr/src/kernels/3.10.0-514.16.1.el7.x86_64'
  CC [M]  /tmp/bcwc_pcie/fthd_v4l2.o
/tmp/bcwc_pcie/fthd_v4l2.c: In function ‘fthd_buffer_prepare’:
/tmp/bcwc_pcie/fthd_v4l2.c:207:13: warning: assignment from incompatible pointer type [enabled by default]
     sgtable = vb2_dma_sg_plane_desc(vb, i);
             ^
/tmp/bcwc_pcie/fthd_v4l2.c: At top level:
/tmp/bcwc_pcie/fthd_v4l2.c:306:2: warning: initialization from incompatible pointer type [enabled by default]
  .queue_setup            = fthd_buffer_queue_setup,
  ^
/tmp/bcwc_pcie/fthd_v4l2.c:306:2: warning: (near initialization for ‘vb2_queue_ops.queue_setup’) [enabled by default]
/tmp/bcwc_pcie/fthd_v4l2.c:310:2: warning: initialization from incompatible pointer type [enabled by default]
  .stop_streaming         = fthd_stop_streaming,
  ^
/tmp/bcwc_pcie/fthd_v4l2.c:310:2: warning: (near initialization for ‘vb2_queue_ops.stop_streaming’) [enabled by default]
/tmp/bcwc_pcie/fthd_v4l2.c: In function ‘fthd_v4l2_register’:
/tmp/bcwc_pcie/fthd_v4l2.c:697:3: error: ‘struct vb2_queue’ has no member named ‘timestamp_flags’
  q->timestamp_flags = V4L2_BUF_FLAG_TIMESTAMP_MONOTONIC;
   ^
/tmp/bcwc_pcie/fthd_v4l2.c:698:3: error: ‘struct vb2_queue’ has no member named ‘min_buffers_needed’
  q->min_buffers_needed = 1;
   ^
/tmp/bcwc_pcie/fthd_v4l2.c:723:2: error: implicit declaration of function ‘vb2_dma_sg_init_ctx’ [-Werror=implicit-function-declaration]
  dev_priv->alloc_ctx = vb2_dma_sg_init_ctx(&dev_priv->pdev->dev);
  ^
/tmp/bcwc_pcie/fthd_v4l2.c:723:22: warning: assignment makes pointer from integer without a cast [enabled by default]
  dev_priv->alloc_ctx = vb2_dma_sg_init_ctx(&dev_priv->pdev->dev);
                      ^
/tmp/bcwc_pcie/fthd_v4l2.c: In function ‘fthd_v4l2_unregister’:
/tmp/bcwc_pcie/fthd_v4l2.c:760:2: error: implicit declaration of function ‘vb2_dma_sg_cleanup_ctx’ [-Werror=implicit-function-declaration]
  vb2_dma_sg_cleanup_ctx(dev_priv->alloc_ctx);
  ^
cc1: some warnings being treated as errors
make[2]: *** [/tmp/bcwc_pcie/fthd_v4l2.o] Error 1
make[1]: *** [_module_/tmp/bcwc_pcie] Error 2
make[1]: Leaving directory `/usr/src/kernels/3.10.0-514.16.1.el7.x86_64'
make: *** [all] Error 2
[root@macbook-air bcwc_pcie]#

hobarrera, 2017-05-23:
Please use triple backticks to properly format large blocks of code:

    ```
    some
    code
    ```

That said, is it possible you're using linux kernel 3.10.0? That kernel is **extremely** old, and not supported.

arun-gurung, 2017-05-24:
You`re right, it`s kernel 3.10.0-514.16.1.el7.x86_64 and I`m running CentOS 7. It`s the stock kernel that came with the operating system. I guess I have to upgrade.

arun-gurung, 2017-05-24:
I upgraded to Linux 4.11.2 and now the camera works fine! Yay! Thanks so much to everyone who helped me.

hobarrera, 2017-05-24:
No prob. Please close since the problem is solved.

arun-gurung, 2017-05-24:
Problem solved, case closed.

closed 2017-05-24 by arun-gurung