patjak/facetimehd #119: Not building as dynamic kernel module

omangin, 2017-05-24:
The last commit (4a69e46f) seems to break the module build (at least through DKMS). I'm using the archlinux `PKGBUILD` from [github.com/aur-packages/bcwc-pcie-dkms](https://github.com/aur-packages/bcwc-pcie-dkms). It works with 7f1cd87 though.

Here is the output from the package manager:
```
:: Processing package changes...
(1/1) upgrading bcwc-pcie-dkms                                                                                [#################################################################] 100%

Creating symlink /var/lib/dkms/bcwc-pcie/0r251.4a69e46/source ->
                 /usr/src/bcwc-pcie-0r251.4a69e46

DKMS: add completed.

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.11.2-1-ARCH -C /usr/lib/modules/4.11.2-1-ARCH/build M=/var/lib/dkms/bcwc-pcie/0r251.4a69e46/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.11.2-1-ARCH (x86_64)
Consult /var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/make.log for more information.
modprobe: FATAL: Module facetimehd not found in directory /lib/modules/4.11.2-1-ARCH
:: Running post-transaction hooks...
(1/2) Install DKMS modules
==> dkms install bcwc-pcie/0r251.4a69e46 -k 4.11.2-1-ARCH
Error! Bad return status for module build on kernel: 4.11.2-1-ARCH (x86_64)
Consult /var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/make.log for more information.
```
And the file `/var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/make.log`:

```
DKMS make.log for bcwc-pcie-0r251.4a69e46 for kernel 4.11.2-1-ARCH (x86_64)
Wed May 24 11:06:55 EDT 2017
make: Entering directory '/usr/lib/modules/4.11.2-1-ARCH/build'
  LD      /var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/built-in.o
  CC [M]  /var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/fthd_hw.o
  CC [M]  /var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/fthd_drv.o
  CC [M]  /var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/fthd_ringbuf.o
  CC [M]  /var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/fthd_isp.o
In file included from /var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/fthd_isp.c:27:0:
/var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/fthd_isp.c: In function ‘fthd_isp_cmd_set_loadfile’:
/var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/fthd_isp.c:585:22: error: ‘file’ undeclared (first use in this function)
  FTHD_S2_MEMCPY_TOIO(file->offset, fw->data, fw->size);
                      ^
/var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/fthd_hw.h:37:86: note: in definition of macro ‘FTHD_S2_MEMCPY_TOIO’
 #define FTHD_S2_MEMCPY_TOIO(offset, buf, len) _FTHD_S2_MEMCPY_TOIO(dev_priv, (buf), (offset), (len))
                                                                                      ^~~~~~
/var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/fthd_isp.c:585:22: note: each undeclared identifier is reported only once for each function it appears in
  FTHD_S2_MEMCPY_TOIO(file->offset, fw->data, fw->size);
                      ^
/var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/fthd_hw.h:37:86: note: in definition of macro ‘FTHD_S2_MEMCPY_TOIO’
 #define FTHD_S2_MEMCPY_TOIO(offset, buf, len) _FTHD_S2_MEMCPY_TOIO(dev_priv, (buf), (offset), (len))
                                                                                      ^~~~~~
make[1]: *** [scripts/Makefile.build:294: /var/lib/dkms/bcwc-pcie/0r251.4a69e46/build/fthd_isp.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1492: _module_/var/lib/dkms/bcwc-pcie/0r251.4a69e46/build] Error 2
make: Leaving directory '/usr/lib/modules/4.11.2-1-ARCH/build'
```

patjak, 2017-05-24:
Thanks for spotting. I've pushed a fix fcdcc9aeb1c756ec3f91642f0221b79d8ee302f6

omangin, 2017-05-24:
That works! Thanks.

closed 2017-05-24 by omangin