patjak/facetimehd #121: Which branch shall I use ?

jbfavre, 2017-06-10:
Hello @patjak and thanks for your work !
I'm building the Debian package of `bcwc_pcie`.

It used to be an `Intent to package` work, but has changed to a `Request for packaging`, meaning other people would like to see it included in Debian.

My question is: which branch am I supposed to use ? `mainline` or `master` ?
It seems `mainline` is the most active one. but is it considered as stable, or is it a sort of dev branch ?

Waiting for your answer,
Cheers

hobarrera, 2017-06-11:
FWIW, in Arch we use `mainline`, **but** we cherry-pick two commits[1], since without these, the package won't work (no `dkms.conf`).

AFAIK, the work on `mainline` is targeted at getting this into the mainline kernel, but it *does* have some small fixes over `master`.

[1]: 76945312a9d2e786bebc9bb5711f5962365c5c20 and 4901c363e8688a46b83bfcbee92c1588dca6054c

jbfavre, 2017-06-12:
Thanks for the answer @hobarrera 
Will have a look on mainline then :)

Cheers

closed 2017-07-06 by patjak