patjak/facetimehd #123: Apple Facetime HD no longer works in Google Chrome v59.

rufusshrestha, 2017-06-16:
Apple Facetime HD no longer works in chrome version 59.0.3071.104  channel: stable on Distributor ID:	Ubuntu Description:	Ubuntu 16.04.2 LTS Release:	16.04 Codename:	xenial

Steps to reproduce the problem:
1. Upgrade to chrome v 59 in Macbook Pro Ubuntu 16.04
2. Go to hangout to start a video chat

What is the expected behavior?
Hangout should be able to use drivers compiled from https://github.com/patjak/bcwc_pcie/wiki

What went wrong?
Chrome cannot access video drivers compiled from this repo in v59+

Did this work before? Yes chrome64_58.0.3029.96.deb

Works in Cheese and other apps and drivers in Chrome works for v58 without issues

Some outputs:

lsmod | grep facetime

facetimehd             98304  0
videobuf2_dma_sg       20480  1 facetimehd
videobuf2_v4l2         28672  1 facetimehd
videobuf2_core         36864  2 facetimehd,videobuf2_v4l2
videodev              176128  4 v4l2_common,facetimehd,videobuf2_core,videobuf2_v4l2

dmesg | grep facetime

[    4.702915] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[    4.703270] facetimehd 0000:03:00.0: Found FaceTime HD camera with device id: 1570
[    4.703460] facetimehd 0000:03:00.0: Setting 64bit DMA mask
[    4.709236] facetimehd 0000:03:00.0: S2 PCIe link init succeeded
[    4.709268] facetimehd 0000:03:00.0: Refclk: 25MHz (0xa)
[    4.719222] facetimehd 0000:03:00.0: PLL reset finished
[    4.719235] facetimehd 0000:03:00.0: Waiting for S2 PLL to lock at 450 MHz
[    5.023688] facetimehd 0000:03:00.0: Failed to lock S2 PLL: 0xc902c902
[    5.043642] facetimehd 0000:03:00.0: DDR40 PHY PLL locked on safe settings
[    5.043656] facetimehd 0000:03:00.0: STRAP valid
[    5.043657] facetimehd 0000:03:00.0: Configuring DDR PLLs for 450 MHz
[    5.043665] facetimehd 0000:03:00.0: DDR40 PLL is locked after 0 us
[    5.043675] facetimehd 0000:03:00.0: First DDR40 VDL calibration completed after 2 us
[    5.043687] facetimehd 0000:03:00.0: Second DDR40 VDL calibration completed after 2 us
[    5.043688] facetimehd 0000:03:00.0: Using step size 157
[    5.043693] facetimehd 0000:03:00.0: VDL set to: coarse=0x10008, fine=0x1011c
[    5.043698] facetimehd 0000:03:00.0: Virtual VTT enabled
[    5.064112] facetimehd 0000:03:00.0: S2 DRAM memory address: 0x22159559
[    5.064126] facetimehd 0000:03:00.0: Rewrite DDR mode registers succeeded
[    5.064326] facetimehd 0000:03:00.0: Full memory verification succeeded! (0)
[    5.526288] facetimehd 0000:03:00.0: Loaded firmware, size: 1392kb
[    5.566281] facetimehd 0000:03:00.0: ISP woke up after 0ms
[    5.566287] facetimehd 0000:03:00.0: Number of IPC channels: 7, queue size: 44865
[    5.566324] facetimehd 0000:03:00.0: Firmware requested heap size: 3072kb
[    5.576332] facetimehd 0000:03:00.0: ISP second int after 0ms
[    5.576334] facetimehd 0000:03:00.0: Channel description table at 00800000
[    5.586439] facetimehd 0000:03:00.0: magic value: 00000000 after 0 ms
[    5.586441] facetimehd 0000:03:00.0: Enabling interrupts
[    5.661587] facetimehd 0000:03:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2

hobarrera, 2017-06-16:
Having the exact same issue on ArchLinux, with kernel `4.11.5-1-ARCH`. Hangouts (I only use chromium for hangouts) won't work, any other app does.

For the record, @rufusshrestha what kernel are you running? Ubuntu tends to run older kernels, is that the case here? The would pretty much discard some kernel-related issue.

rufusshrestha, 2017-06-16:
I'm on Linux rufus 4.4.0-79-generic x64

VKen, 2017-06-16:
Confirmed same issue and behavior, on macbook pro retina, running ArchLinux, since 2 days ago.

Doesn't work on Chromium (Version 59.0.3071.86 (Developer Build) (64-bit))
Doesn't work in Chrome Beta Version 60.0.3112.32 (Official Build) beta (64-bit)

I'm on Kernel 4.11.5-1-ARCH #1 SMP PREEMPT Wed Jun 14 16:19:27 CEST 2017 x86_64 GNU/Linux

For the errors below in `dmesg`
```
[    2.029196] facetimehd: loading out-of-tree module taints kernel.
[    4.266706] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
```
Seems like there are other issue threads saying they are harmless.

Cheese works.

Re-Installation has some weird errors which keeps repeating `modprobe: FATAL: Module facetimehd not found`

```
◯  pacaur -Sa bcwc-pcie-git
:: resolving dependencies...
:: looking for inter-conflicts...
:: bcwc-pcie-git latest revision -- fetching

AUR Packages  (1) bcwc-pcie-git-latest  

... snip ...

Packages (1) bcwc-pcie-git-r252.fcdcc9a-1

Total Installed Size:  0.17 MiB
Net Upgrade Size:      0.00 MiB

:: Proceed with installation? [Y/n] 
(1/1) checking keys in keyring                                                                                                 [#############################################################################] 100%
(1/1) checking package integrity                                                                                               [#############################################################################] 100%
(1/1) loading package files                                                                                                    [#############################################################################] 100%
(1/1) checking for file conflicts                                                                                              [#############################################################################] 100%
(1/1) checking available disk space                                                                                            [#############################################################################] 100%
:: Running pre-transaction hooks...
(1/1) Remove DKMS modules
==> dkms remove bcwc-pcie-git/r252.fcdcc9a -k 4.11.5-1-ARCH
:: Processing package changes...
(1/1) reinstalling bcwc-pcie-git                                                                                               [#############################################################################] 100%
Error! Could not find module source directory.
Directory: /usr/src/bcwc-pcie-r252.fcdcc9a does not exist.
modprobe: FATAL: Module facetimehd not found in directory /lib/modules/4.11.5-1-ARCH
:: Running post-transaction hooks...
(1/2) Install DKMS modules
==> dkms install bcwc-pcie-git/r252.fcdcc9a -k 4.11.5-1-ARCH
(2/2) Arming ConditionNeedsUpdate...
```

hobarrera, 2017-06-17:
Has this been reported on the chromium side, since it's updating chromium that breaks things?

rufusshrestha, 2017-06-17:
Yes, have raised a bug in chromium project bug tracker

https://bugs.chromium.org/p/chromium/issues/detail?id=733859

VKen, 2017-06-20:
Hi All,

Just checking back, 

Previously, I have removed the permission for camera access by hangouts.

I have downgraded my package to Chromium Version 58.0.3029.96 (64-bit), and it works, asking for my camera permission.

Upgrade to Chromium Version 59.0.3071.104 (Developer Build) (64-bit), it stopped working.
Doesn't ask for camera permission.

I'll update it in the chromium project bug tracker too.

slashrsm, 2017-06-29:
Same here: Arch Linux, 4.11.6-3-ARCH kernel, Chromium 59.0.3071.115 64bit.

thecosmicfrog, 2017-06-29:
Same issue here. Ubuntu 16.04 LTS, Google Chrome 59.

hobarrera, 2017-06-30:
The issue has been confirmed as a regression on Chromium's issue tracker, so no need for more "me too". Looks like it's all on their side.

JonLundy, 2017-07-19:
i am having a similar issue to this.. but with cheese it receives a single frame and then gives an error that the device is busy. 

```
(cheese:11573): cheese-WARNING **: Device '/dev/video0' is busy: gstv4l2object.c(3592): gst_v4l2_object_set_format_full (): /GstCameraBin:camerabin/GstWrapperCameraBinSrc:camera_source/GstBin:bin35/GstV4l2Src:v4l2src1:
Call to S_FMT failed for YUYV @ 1280x720: Device or resource busy
```

JonLundy, 2017-07-19:
when i load the module:

```
Jul 19 10:48:33 hostname kernel: Linux video capture interface: v2.00
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: Setting 64bit DMA mask
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: S2 PCIe link init succeeded
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: PLL reset finished
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: Failed to lock S2 PLL: 0xc902c902
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: STRAP valid
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 3 us
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 1 us
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: Using step size 147
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x10119
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: Virtual VTT enabled
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
Jul 19 10:48:33 hostname kernel: facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
Jul 19 10:48:34 hostname kernel: facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
Jul 19 10:48:34 hostname kernel: facetimehd 0000:02:00.0: ISP woke up after 0ms
Jul 19 10:48:34 hostname kernel: facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
Jul 19 10:48:34 hostname kernel: facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
Jul 19 10:48:34 hostname kernel: facetimehd 0000:02:00.0: ISP second int after 0ms
Jul 19 10:48:34 hostname kernel: facetimehd 0000:02:00.0: Channel description table at 00800000
Jul 19 10:48:34 hostname kernel: facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
Jul 19 10:48:34 hostname kernel: facetimehd 0000:02:00.0: Enabling interrupts
Jul 19 10:48:34 hostname kernel: FWMSG: 
Jul 19 10:48:34 hostname kernel: FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
Jul 19 10:48:34 hostname kernel: FWMSG: 
Jul 19 10:48:34 hostname kernel: FWMSG:   Restart count : 0
Jul 19 10:48:34 hostname kernel: FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
Jul 19 10:48:34 hostname kernel: FWMSG:   System Clock : 200000000 Hz
Jul 19 10:48:34 hostname kernel: FWMSG:   Processor mode : SUPERVISOR
Jul 19 10:48:34 hostname kernel: FWMSG:   Cache architecture type : SEPARATE
Jul 19 10:48:34 hostname kernel: FWMSG:   Cache type : WRITEBACK
Jul 19 10:48:34 hostname kernel: FWMSG:   Data Cache Line : 32 [0x20]
Jul 19 10:48:34 hostname kernel: FWMSG:   Boot arguments entries : 2
Jul 19 10:48:34 hostname kernel: FWMSG:     0000: 0x00000000 0x00000000
Jul 19 10:48:34 hostname kernel: FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
Jul 19 10:48:34 hostname kernel: FWMSG:   Main memory :   base : 0x00000000
Jul 19 10:48:34 hostname kernel: FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
Jul 19 10:48:34 hostname kernel: FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
Jul 19 10:48:34 hostname kernel: FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
Jul 19 10:48:34 hostname kernel: FWMSG:   Shared window : base : 0x00800000 [static wiring]
Jul 19 10:48:34 hostname kernel: FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
Jul 19 10:48:34 hostname kernel: FWMSG:   Shared memory : base : 0x00800000
Jul 19 10:48:34 hostname kernel: FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
Jul 19 10:48:34 hostname kernel: FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
Jul 19 10:48:34 hostname kernel: FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
Jul 19 10:48:34 hostname kernel: FWMSG:   Heap free space : 10002626 [0x0098a0c2]
Jul 19 10:48:34 hostname kernel: FWMSG:   Heap allocated space : 94656 [0x000171c0]
Jul 19 10:48:34 hostname kernel: FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
Jul 19 10:48:34 hostname kernel: FWMSG:   Application : adc [release]
Jul 19 10:48:34 hostname kernel: FWMSG:   Linked on : Jul 25 2015 - 08:48:55
Jul 19 10:48:34 hostname kernel: FWMSG:   Release : S2ISP-01.43.00
Jul 19 10:48:34 hostname kernel: FWMSG:   H4ISPAPPLE : 11536
Jul 19 10:48:34 hostname kernel: FWMSG:   H4ISPCD : 4081
Jul 19 10:48:34 hostname kernel: FWMSG:   ffw : 4143
Jul 19 10:48:34 hostname kernel: FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
Jul 19 10:48:34 hostname kernel: FWMSG: 
Jul 19 10:48:34 hostname kernel: FWMSG: -------------------------------------------------------------------------------
Jul 19 10:48:34 hostname kernel: FWMSG: 
Jul 19 10:48:34 hostname kernel: FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[... cut ...]
Jul 19 10:48:34 hostname kernel: facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
```

lsmod
```
facetimehd            102400  0
videobuf2_v4l2         24576  1 facetimehd
videobuf2_core         40960  2 facetimehd,videobuf2_v4l2
videodev              172032  3 facetimehd,videobuf2_core,videobuf2_v4l2
videobuf2_dma_sg       16384  1 facetimehd
```
uname -a
```
Linux hostname 4.11.8-300.fc26.x86_64 #1 SMP Thu Jun 29 20:09:48 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
```

derjohn, 2017-07-22:
+1 here, I think it might be related to '[7743:7781:0722/094734.525112:ERROR:video_capture_device_factory_linux.cc(133)] Not implemented reached in void media::GetSupportedFormatsForV4L2BufferType(int, media::VideoCaptureFormats*)'

´´´
chromium-browser --enable-logging --vmodule=*/webrtc/*=2,*/libjingle/*=2,*=-2 --no-sandbox   
Using PPAPI flash.
--disable-new-tab-first-run --enable-user-scripts --ppapi-flash-path=/usr/lib/adobe-flashplugin/libpepflashplayer.so --ppapi-flash-version=
[7743:7778:0722/094727.195999:ERROR:nss_util.cc(802)] After loading Root Certs, loaded==false: NSS error code: -8018
[7743:7743:0722/094727.405539:ERROR:account_tracker.cc(356)] OnGetTokenFailure: Invalid credentials.
[7743:7743:0722/094727.405581:ERROR:account_tracker.cc(356)] OnGetTokenFailure: Invalid credentials.
[7743:7743:0722/094727.427683:ERROR:display_info_provider_aura.cc(31)] Not implemented reached in virtual void extensions::DisplayInfoProviderAura::UpdateDisplayUnitInfoForPlatform(const display::Display&, extensions::api::system_display::DisplayUnitInfo*)
[7743:7781:0722/094734.525112:ERROR:video_capture_device_factory_linux.cc(133)] Not implemented reached in void media::GetSupportedFormatsForV4L2BufferType(int, media::VideoCaptureFormats*)
[7743:7778:0722/094755.062432:ERROR:service_manager.cc(158)] Connection InterfaceProviderSpec prevented service: content_plugin from binding interface: memory_instrumentation::mojom::Coordinator exposed by: content_browser
´´´

hobarrera, 2017-07-22:
@derjohn 

Yeah, that's pretty much what's been reported upstream: https://bugs.chromium.org/p/chromium/issues/detail?id=733859

hobarrera, 2017-07-22:
Can we maybe add some label to this issue identifying that it's a third-party issue?

rufusshrestha, 2017-08-16:
> As per the latest comments on the chromium issue tracker, it seems that
> it's somehow triggered by the camera reporting 0.00 fps and 0x0
> resolution: https://bugs.chromium.org/p/chromium/issues/detail?id=
> 733859#c23
>
> Might it be possible to work around that on the driver side?

Commit at
https://chromium.googlesource.com/chromium/src.git/+/9c77470ff34bac937ceb765a27cee1703f0f2426
will likely fix this, however, the underlying issue seems to be the driver
reporting incorrect framerate and res. Could the Facetime HD driver be
improved to report fps/res correctly?

rufusshrestha, 2017-09-09:
This has been confirmed fixed in Chromium Dev Channel
https://bugs.chromium.org/p/chromium/issues/detail?id=733859

However, it may be worthwhile to look at the driver aspect to prevent 0x0 resolution reported by the driver. Thanks again guys.

thecosmicfrog, 2017-09-14:
I have tested this in the latest Chrome dev channel release. The webcam works, but the image is very stretched horizontally. The best way I can describe it is like a 4:3 image stretched to fill up a 16:9 screen.

I've seen this behaviour in Google Hangouts and some other web-based camera testers.

derjohn, 2017-10-15:
If this is really a chromium bug, why doesnt it work with FF then?

rufusshrestha, 2017-10-26:
v62 google chrome stable works with facetimehd but image is stretched horizontally as its using fallback video modes since the driver is not reporting correctly. This has also to be a bug in this driver. Anyway this can be investigated?

ericfreese, 2017-11-19:
I really don't have any experience with this stuff, but I did try to dig into this a little bit a few weeks ago. The underlying problem seems to be that the [driver reports its frame size type as `V4L2_FRMSIZE_TYPE_STEPWISE`](https://github.com/patjak/bcwc_pcie/blob/master/fthd_v4l2.c#L549) and [Chromium doesn't yet support that frame size type](https://cs.chromium.org/chromium/src/media/capture/video/linux/video_capture_device_factory_linux.cc?l=138) so the struct values remain unset (aka `0`).

There is a comment in the Chromium code that points to https://bugs.chromium.org/p/chromium/issues/detail?id=249953 as an open bug.

If the driver were to be modified to solve this problem, I think it would have to be switched back to use `V4L2_FRMSIZE_TYPE_DISCRETE` as it did before commit [`5c8ecae`](https://github.com/patjak/bcwc_pcie/commit/5c8ecaef7f7227f7c7726eb425b05b5cccaf5769#diff-701ca18d503314a7ac0faf0f951bbe43L551).

estathop, 2018-04-07:
New to Ubuntu 17.10.1 x64 on a Macbook Air 2015, After so much difficulty in installing firmware and drivers etc, I come to see that Skype and Firefox via Facebook Messenger recognizes there is a camera in system but there isn't a video transmission. When I call on Messenger no video is transmitting although it lets me perform a video call, when visiting settings on Skype the system finds  Apple facetime HD but there is only a black rectangle and no video transmitted. On the other hand when I try the mplayer tv:// on terminal the video is ok

NoMod-Programming, 2018-11-01:
@ericfreese This might be of some use, but my machine reports the following:

    FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 2312: FlowIC00: Meta data buffer unavailable

Then a little later:

    facetimehd 0000:02:00.0: IO: timeout

A naive reading of this is telling me that it can't set any kind of metadata, which means that the fps and resolution would have values of 0.

defenestration, 2019-02-12:
Same issue where it works in chrome for a while but starts failing after a while.

Mostly tehse show in dmesg:
```
[ +13.389641] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x20a!
[  +9.142230] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x28a!
```

Some more interesting segments in dmesg. not sure if related.
```
[  +9.242356] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x20a!
[  +9.142230] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x28a!
[  +2.733114] brcmfmac: brcmf_msgbuf_get_pktid: Invalid packet id 695 (not in use)
...
[  +1.268010] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x20a!
[  +0.033277] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x20a!
[  +0.044162] usb 1-2.3: reset full-speed USB device number 10 using 
```
Suddenly starts spamming these messages instead
```
[  +0.004498] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x804!
[  +0.028878] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 2312: FlowIC00: Meta data buffer unavailable!
```

then camera died around this time.
```
[  +0.004404] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x804!
[  +0.007165] FWMSG: [ISP] CH = 0 CMD = 0x0101 [CISP_CMD_CH_STOP]
[  +0.026200] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x804!
[  +1.983987] facetimehd 0000:02:00.0: IO: timeout
```

Hopefully thats helpful. Running fedora 29.  
Version: MacBookPro12,1
```
$uname -r
4.20.4-200.fc29.x86_64
```

maximilianduell, 2019-03-08:
I am also running into this issue time and again.
Could this be solved by a (cheap) workaround on the level of the driver?

If I understand https://bugs.chromium.org/p/chromium/issues/detail?id=249953 correctly, the problem is that the facetime webcam basically supports a continuum of resolutions. 

It would be nice to have a "chromium/skype compatibility mode" which restricts this to the typical webcam resolution capabilities, say by picking some special resolutions like full, half, quarter. It would be nice because people here would not have to suffer and keep using skype, etc., without compiling chromium-dev, at least until these google people get their stuff together.

maximilianduell, 2019-03-12:
I tried out the workaround described in my last post, and it seems to be working well. Does it make sense to implement this properly, perhaps with a compile-time switch or even better with a module parameter, perhaps "chrome_workaround=1"?

drinkingcoffee, 2019-04-15:
just checking in to see if anyone has a workaround to get the camera working for skype?
At the moment the camera works in chrome, but not in skype (and it even doesn't work trying to use the webapp version of skype in the same chrome browser where hangouts works).

maximilianduell, 2019-05-05:
I put a version with the fix on my github page https://github.com/maximilianduell/bcwc_pcie

ivanperez-keera, 2019-06-29:
The last two responses seem inconsistent:

> @drinkingcoffee: At the moment the camera works in chrome, but not in skype.

>  @maximilianduell: I put an adaption of the current version with the fix described in my post above on my github page 

(which is for chrome *and* skype)

Currently, it works just "fine" for me on Chromium.

Does this currently work on Chrome then?

patjak, 2020-05-22:
Should be fixed with 545cb18e2af1dc6853e9ade644f8d0fb54d257b1
Please test and repoen if problem still exists

closed 2020-05-22 by patjak

drinkingcoffee, 2020-05-22:
just tried it out and it looks good. Thanks!

gabriel-lucas, 2020-05-22:
I also tested via this [script](https://www.robert-oswald.de/blog/wp-content/uploads/2020/01/elementary-OS-5.1-Install-Guide.pdf) and works :)
many thanks @patjak