patjak/facetimehd #124: The driver hangs Open Broadcaster Software and the module gets subsequently frozen

arun-gurung, 2017-06-22:
After an hour or so, OBS will hang and the next time I start it, the module has frozen and no webcam software will recognise the facetimehd. I cannot unload the module.

patjak, 2017-06-23:
Can you see anything suspicious in dmesg when this happen?

arun-gurung, 2017-06-23:
The only abnormality in dmesg is this and I don't know if it tells anything:

Jun 20 20:15:23 macbook-air kernel: facetimehd 0000:02:00.0: IO: timeout

arun-gurung, 2017-06-23:
I'm sorry, but now OBS hangs also when only using my Logitech C920 and not only the facetimehd camera. But I don't get if this is an internal fault in OBS, because the facetimehd module hangs and not the Logitech.

arun-gurung, 2017-06-24:
This is very weird: Now my facetimehd suddenly works in OBS without reloading the module, but the Logitech C920 has blacked out in OBS. But the C920 works in the "guvcview" webcam program.

arun-gurung, 2017-06-29:
Now OBS hanged after thirty minutes. I "xkill" it and restarted as soon as possible and the two cameras were still available to resume the stream.

patjak, 2017-06-29:
It sounds like a bug in OBS to me. There still might be a problem in facetimehd for handling misbehaving v4l2 clients. I'll experiment some with OBS on my own and see if I run into the same issue.

arun-gurung, 2017-06-29:
Yes, it looke like a bug in OBS. Mind you, the facetimehd stopped in OBS and came back the second time I started OBS.

arun-gurung, 2017-06-30:
It must definitely be a problem with OBS. I use Guvcview webcam software and both cameras now work here without me rebooting the system.

arun-gurung, 2017-07-01:
Yes, there must be an error with OBS, because I now restarted OBS and both facetimehd and Logitech C920 still works.

After 30 minutes to only an hour, the OBS will freeze, but the video still is streaming but I can´t do any changes in OBS after this.

arun-gurung, 2017-07-07:
Even if OBS has a bug, there still the frozen facetimehd module it leaves behind. So there must also be an error in facetimehd since I am unable to unload/reload it.

arun-gurung, 2019-10-01:
I now have Linux 4.18 and OBS 24.0.1 and the problem is gone.

closed 2019-10-01 by arun-gurung

patjak, 2019-10-01:
Perhaps this got solved by 9190f7d3b8e26179149a90b3d0138e6bc179afab
Thanks for testing