patjak/facetimehd #125: make command gives error

tareqalam, 2017-06-27:
in Get started on Debian on step $ make it fails with an error like this in my laptop:
```
make -C /lib/modules/4.4.0-81-generic/build M=/home/tareq/DATA/software/bcwc_pcie modules
make[1]: Entering directory `/usr/src/linux-headers-4.4.0-81-generic'
Makefile:633: arch//Makefile: No such file or directory
Makefile:1445: *** target pattern contains no `%'.  Stop.
make[1]: Leaving directory `/usr/src/linux-headers-4.4.0-81-generic'
make: *** [all] Error 2
```
I have done the Firmware extraction step 
Using ubunt 16.04 here (recently upgraded from 14.04)
Thanks in advance

seumasmac, 2017-06-29:
@tareqalam That error is because you are running Linux kernel `4.4.0-81-generic` but you don't have `linux-headers-4.4.0-81-generic` installed, so the directory doesn't exist.

Try:
`apt-get install linux-headers-4.4.0-81-generic`
and see if that fixes the problem.

patjak, 2017-07-06:
@seumasmac Lets hope that solved the problem. Closing

closed 2017-07-06 by patjak