patjak/facetimehd #131: Gentoo on MBP Retina 2015 get no /dev/video0. dmesg : no symbol version for module_layout

wminyu, 2017-07-24:
I haven't select bdc_pci in my kernel, but get no /dev/video0

gentoo-mbp ~ # lsmod |grep facetimehd
facetimehd             39964  0
videobuf2_dma_sg        6361  1 facetimehd
gentoo-mbp ~ # lsmod |grep bdc_pci
gentoo-mbp ~ # lsmod |grep facetimehd
facetimehd             39964  0
videobuf2_dma_sg        6361  1 facetimehd
gentoo-mbp ~ # dmesg  |grep facetimehd
[    5.613643] facetimehd: no symbol version for module_layout
[    5.613644] facetimehd: loading out-of-tree module taints kernel.
[    7.979281] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
gentoo-mbp ~ # uname -r
4.9.34-gentoo
gentoo-mbp ~ # ls /dev/ |grep video

blsz, 2017-10-14:
I am running gentoo 4.9.34 on a mid-2011 iMac 27".  The FaceTime HD camera appears to be detected but /dev/video* is not created and /sys/class/video4linux/ is empty.

kernel config https://gist.github.com/anonymous/56f103b8c45e4980a2fbb359c745b761

```
# lsmod | grep bdc_pci

# lsmod | grep facetimehd
facetimehd             42626  0
videobuf2_dma_sg        6929  1 facetimehd
videobuf2_v4l2         14283  1 facetimehd
videodev              124627  2 facetimehd,videobuf2_v4l2
videobuf2_core         26394  2 facetimehd,videobuf2_v4l2

# dmesg
...
[    4.197533] usb 1-2: New USB device found, idVendor=05ac, idProduct=850b
[    4.199295] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.199296] usb 1-2: Product: FaceTime HD Camera (Built-in)
[    4.199297] usb 1-2: Manufacturer: Apple Inc.
[    4.199298] usb 1-2: SerialNumber: CC2B3V04X4DGGFLP
...
[    7.156624] Linux video capture interface: v2.00
[    7.159056] facetimehd: loading out-of-tree module taints kernel.

# journalctl -b | grep -i facetime
Oct 14 11:44:54 bigmac kernel: usb 1-2: Product: FaceTime HD Camera (Built-in)
Oct 14 11:44:54 bigmac kernel: facetimehd: loading out-of-tree module taints kernel.
Oct 14 11:44:54 bigmac systemd-modules-load[3440]: Inserted module 'facetimehd'

# uname -r
4.9.34-gentoo

# sudo dmidecode -s system-product-name
iMac12,2

# lsusb | grep Apple
Bus 002 Device 004: ID 05ac:8242 Apple, Inc. Built-in IR Receiver
Bus 002 Device 003: ID 05ac:8403 Apple, Inc. Internal Memory Card Reader
Bus 001 Device 003: ID 05ac:850b Apple, Inc.
Bus 001 Device 008: ID 05ac:820b Apple, Inc. Bluetooth HID Mouse
Bus 001 Device 007: ID 05ac:820a Apple, Inc. Bluetooth HID Keyboard
Bus 001 Device 006: ID 05ac:8215 Apple, Inc. Built-in Bluetooth 2.0+EDR HCI

# cat /etc/modules-load.d/facetimehd.conf
videobuf2-core
videobuf2_v4l2
videobuf2-dma-sg
facetimehd

#  ls /sys/class/video4linux/

# ls -al  /dev/ | grep video
crw-rw----   1 root  video      29,   0 Oct 14 11:44 fb0
crw-rw----   1 root  video      29,   1 Oct 14 11:44 fb1
```

blsz, 2017-10-19:
I'm also seeing "facetimehd: Unknown symbol" messages in dmesg.
```
[    7.159056] facetimehd: loading out-of-tree module taints kernel.
...
[ 2012.355719] facetimehd: Unknown symbol vb2_queue_init (err 0)
[ 2012.355738] facetimehd: Unknown symbol vb2_dma_sg_memops (err 0)
[ 2012.355757] facetimehd: Unknown symbol video_ioctl2 (err 0)
[ 2012.355776] facetimehd: Unknown symbol vb2_ioctl_streamoff (err 0)
[ 2012.355791] facetimehd: Unknown symbol vb2_ioctl_expbuf (err 0)
[ 2012.355810] facetimehd: Unknown symbol vb2_ops_wait_finish (err 0)
[ 2012.355825] facetimehd: Unknown symbol v4l2_ctrl_handler_init_class (err 0)
[ 2012.355842] facetimehd: Unknown symbol vb2_ioctl_querybuf (err 0)
[ 2012.355859] facetimehd: Unknown symbol v4l2_fh_open (err 0)
[ 2012.355880] facetimehd: Unknown symbol video_devdata (err 0)
[ 2012.355895] facetimehd: Unknown symbol vb2_fop_release (err 0)
[ 2012.355911] facetimehd: Unknown symbol vb2_plane_cookie (err 0)
[ 2012.355926] facetimehd: Unknown symbol vb2_ioctl_dqbuf (err 0)
[ 2012.355941] facetimehd: Unknown symbol vb2_ioctl_create_bufs (err 0)
[ 2012.355959] facetimehd: Unknown symbol vb2_buffer_done (err 0)
[ 2012.355973] facetimehd: Unknown symbol v4l2_ctrl_subscribe_event (err 0)
[ 2012.355988] facetimehd: Unknown symbol video_unregister_device (err 0)
[ 2012.356004] facetimehd: Unknown symbol vb2_ioctl_qbuf (err 0)
[ 2012.356018] facetimehd: Unknown symbol vb2_fop_mmap (err 0)
[ 2012.356032] facetimehd: Unknown symbol video_device_alloc (err 0)
[ 2012.356049] facetimehd: Unknown symbol vb2_fop_read (err 0)
[ 2012.356063] facetimehd: Unknown symbol v4l2_device_register (err 0)
[ 2012.356091] facetimehd: Unknown symbol __video_register_device (err 0)
[ 2012.356107] facetimehd: Unknown symbol vb2_ops_wait_prepare (err 0)
[ 2012.356124] facetimehd: Unknown symbol vb2_ioctl_streamon (err 0)
[ 2012.356138] facetimehd: Unknown symbol vb2_fop_poll (err 0)
[ 2012.356152] facetimehd: Unknown symbol v4l2_ctrl_new_std (err 0)
[ 2012.356165] facetimehd: Unknown symbol v4l2_ctrl_handler_free (err 0)
[ 2012.356179] facetimehd: Unknown symbol v4l2_device_unregister (err 0)
[ 2012.356194] facetimehd: Unknown symbol vb2_wait_for_all_buffers (err 0)
[ 2012.356207] facetimehd: Unknown symbol video_device_release (err 0)
[ 2012.356224] facetimehd: Unknown symbol v4l2_event_unsubscribe (err 0)
[ 2012.356239] facetimehd: Unknown symbol vb2_ioctl_reqbufs (err 0)
[ 2062.093039] Linux video capture interface: v2.00
[ 2106.827499] facetimehd: Unknown symbol vb2_queue_init (err 0)
[ 2106.827517] facetimehd: Unknown symbol vb2_dma_sg_memops (err 0)
[ 2106.827536] facetimehd: Unknown symbol video_ioctl2 (err 0)
[ 2106.827554] facetimehd: Unknown symbol vb2_ioctl_streamoff (err 0)
[ 2106.827569] facetimehd: Unknown symbol vb2_ioctl_expbuf (err 0)
[ 2106.827587] facetimehd: Unknown symbol vb2_ops_wait_finish (err 0)
[ 2106.827602] facetimehd: Unknown symbol v4l2_ctrl_handler_init_class (err 0)
[ 2106.827619] facetimehd: Unknown symbol vb2_ioctl_querybuf (err 0)
[ 2106.827635] facetimehd: Unknown symbol v4l2_fh_open (err 0)
[ 2106.827655] facetimehd: Unknown symbol video_devdata (err 0)
[ 2106.827670] facetimehd: Unknown symbol vb2_fop_release (err 0)
[ 2106.827685] facetimehd: Unknown symbol vb2_plane_cookie (err 0)
[ 2106.827699] facetimehd: Unknown symbol vb2_ioctl_dqbuf (err 0)
[ 2106.827714] facetimehd: Unknown symbol vb2_ioctl_create_bufs (err 0)
[ 2106.827732] facetimehd: Unknown symbol vb2_buffer_done (err 0)
[ 2106.827745] facetimehd: Unknown symbol v4l2_ctrl_subscribe_event (err 0)
[ 2106.827760] facetimehd: Unknown symbol video_unregister_device (err 0)
[ 2106.827776] facetimehd: Unknown symbol vb2_ioctl_qbuf (err 0)
[ 2106.827790] facetimehd: Unknown symbol vb2_fop_mmap (err 0)
[ 2106.827804] facetimehd: Unknown symbol video_device_alloc (err 0)
[ 2106.827820] facetimehd: Unknown symbol vb2_fop_read (err 0)
[ 2106.827833] facetimehd: Unknown symbol v4l2_device_register (err 0)
[ 2106.827860] facetimehd: Unknown symbol __video_register_device (err 0)
[ 2106.827875] facetimehd: Unknown symbol vb2_ops_wait_prepare (err 0)
[ 2106.827892] facetimehd: Unknown symbol vb2_ioctl_streamon (err 0)
[ 2106.827905] facetimehd: Unknown symbol vb2_fop_poll (err 0)
[ 2106.827919] facetimehd: Unknown symbol v4l2_ctrl_new_std (err 0)
[ 2106.827932] facetimehd: Unknown symbol v4l2_ctrl_handler_free (err 0)
[ 2106.827946] facetimehd: Unknown symbol v4l2_device_unregister (err 0)
[ 2106.827960] facetimehd: Unknown symbol vb2_wait_for_all_buffers (err 0)
[ 2106.827973] facetimehd: Unknown symbol video_device_release (err 0)
[ 2106.827990] facetimehd: Unknown symbol v4l2_event_unsubscribe (err 0)
[ 2106.828004] facetimehd: Unknown symbol vb2_ioctl_reqbufs (err 0)
[ 2112.924329] Linux video capture interface: v2.00
[ 2158.823972] Linux video capture interface: v2.00
```

vastbinderj, 2018-02-17:
I have the same issue here. Everything loads, I only see the direct firmware error we've been told to ignore.

 Has anyone tried a later kernel to see if the issue is resolved?

```
dmesg | grep facetime
[    9.588221] facetimehd: loading out-of-tree module taints kernel.
[    9.590887] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[    9.591108] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[    9.597934] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[    9.597962] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[    9.607997] facetimehd 0000:02:00.0: PLL reset finished
[    9.612290] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[    9.907573] facetimehd 0000:02:00.0: Failed to lock S2 PLL: 0xc902c902
[    9.927690] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[    9.927705] facetimehd 0000:02:00.0: STRAP valid
[    9.927706] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[    9.927713] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[    9.927734] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[    9.927757] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 1 us
[    9.927758] facetimehd 0000:02:00.0: Using step size 140
[    9.927763] facetimehd 0000:02:00.0: VDL set to: coarse=0x10007, fine=0x10118
[    9.927768] facetimehd 0000:02:00.0: Virtual VTT enabled
[    9.948406] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[    9.948434] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[    9.948717] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[   10.449536] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[   10.601368] facetimehd 0000:02:00.0: ISP woke up after 0ms
[   10.601373] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[   10.601415] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[   10.611444] facetimehd 0000:02:00.0: ISP second int after 0ms
[   10.611446] facetimehd 0000:02:00.0: Channel description table at 00800000
[   10.621710] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[   10.621710] facetimehd 0000:02:00.0: Enabling interrupts
[   10.698745] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
[   10.709030] facetimehd 0000:02:00.0: fthd_hw_deinit
```

vastbinderj, 2018-02-17:
I think the issue is the driver is unloading itself given the final message: 

> fthd_hw_deinit

The question is why....

blsz, 2018-02-21:
In my case, I had neglected to enable USB Video Class (UVC) support in the kernel.  Once I enabled it, /dev/video0 appeared and the facetime camera started working.
```
Device Drivers --->
  <M> Multimedia support  --->
   [*]   Cameras/video grabbers support
   [*]   Media USB Adapters  --->
     <M>   USB Video Class (UVC)  
     [*]     UVC input events device support (NEW)
```
```
# journalctl -b | grep -i facetime
Feb 20 16:52:58 bigmac kernel: usb 1-2: Product: FaceTime HD Camera (Built-in)
Feb 20 16:52:58 bigmac kernel: facetimehd: loading out-of-tree module taints kernel.
Feb 20 16:52:58 bigmac systemd-modules-load[2018]: Inserted module 'facetimehd'
Feb 20 16:52:58 bigmac kernel: uvcvideo: Found UVC 1.00 device FaceTime HD Camera (Built-in) (05ac:850b)
Feb 20 16:52:58 bigmac kernel: input: FaceTime HD Camera (Built-in) as /devices/pci0000:00/0000:00:1a.7/usb1/1-2/1-2:1.0/input/input10
```

vastbinderj, 2018-02-21:
well - I think it has something to do with the current latest stable kernel Gentoo. I was on 4.9.76 and upgraded to 4.15.4. 

After moving to the new kernel, the error message `fthd_hw_deinit` went away and I have the camera working well now.

vastbinderj, 2018-02-21:
Off topic....

@blsz - now that I have my macbook pro 12, 1 working well on Gentoo I'm about to embark on loading Gentoo on my iMac. Do you have any pointers or did you record your experience in a blog post?

blsz, 2018-02-21:
@vastbinderj - I did not post it.  But here's my updated kernel config (I'm still on 4.9.34) and a few notes.
https://gist.github.com/anonymous/691bf06d7fc716b8e60fd31a19ea9558

I stopped using grub and switched to rEFInd - http://www.rodsbooks.com/refind/

Backlight control doesn't work.  This may relate to GPU power management and `ACPI Error: Evaluating _BCM failed (20160831/video-354)`.   I gave up on this and manage backlight with a script. 

Graphics is working despite radeon "ring 0 stalled" and "GPU lockup" messages on boot.
```
radeon 0000:01:00.0: ring 0 stalled for more than 10190msec
radeon 0000:01:00.0: GPU lockup (current fence id 0x0000000000000002 last fence id 0x0000000000000003 on ring 0)
..
 radeon 0000:01:00.0: GPU reset succeeded, trying to resume
```

dmesg: https://gist.github.com/blsz/07f9733ca8b51f93a476f28ac04f90b3