patjak/facetimehd #133: Missing firmware

bhundven, 2017-08-02:
Seeing this:

```[22300.268153] media: Linux media interface: v0.10
[22300.272135] Linux video capture interface: v2.00
[22300.276480] facetimehd: loading out-of-tree module taints kernel.
[22300.276529] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[22379.352954] media: Linux media interface: v0.10
[22379.356605] Linux video capture interface: v2.00
[22379.361685] facetimehd 0000:04:00.0: Found FaceTime HD camera with device id: 1570
[22379.361836] facetimehd 0000:04:00.0: Setting 64bit DMA mask
[22379.367100] facetimehd 0000:04:00.0: S2 PCIe link init succeeded
[22379.367127] facetimehd 0000:04:00.0: Refclk: 25MHz (0xa)
[22379.377079] facetimehd 0000:04:00.0: PLL reset finished
[22379.377081] facetimehd 0000:04:00.0: Waiting for S2 PLL to lock at 450 MHz
[22379.377093] facetimehd 0000:04:00.0: S2 PLL is locked after 10 us
[22379.387060] facetimehd 0000:04:00.0: S2 PLL is in bypass mode
[22379.406995] facetimehd 0000:04:00.0: DDR40 PHY PLL locked on safe settings
[22379.407009] facetimehd 0000:04:00.0: STRAP valid
[22379.407010] facetimehd 0000:04:00.0: Configuring DDR PLLs for 450 MHz
[22379.407017] facetimehd 0000:04:00.0: DDR40 PLL is locked after 0 us
[22379.407028] facetimehd 0000:04:00.0: First DDR40 VDL calibration completed after 2 us
[22379.407039] facetimehd 0000:04:00.0: Second DDR40 VDL calibration completed after 2 us
[22379.407040] facetimehd 0000:04:00.0: Using step size 159
[22379.407045] facetimehd 0000:04:00.0: VDL set to: coarse=0x10008, fine=0x1011c
[22379.407049] facetimehd 0000:04:00.0: Virtual VTT enabled
[22379.427467] facetimehd 0000:04:00.0: S2 DRAM memory address: 0x22159559
[22379.427482] facetimehd 0000:04:00.0: Rewrite DDR mode registers succeeded
[22379.427695] facetimehd 0000:04:00.0: Full memory verification succeeded! (0)
[22379.888962] facetimehd 0000:04:00.0: Loaded firmware, size: 1392kb
[22380.034014] facetimehd 0000:04:00.0: ISP woke up after 0ms
[22380.034029] facetimehd 0000:04:00.0: Number of IPC channels: 7, queue size: 44865
[22380.034034] facetimehd 0000:04:00.0: Firmware requested heap size: 3072kb
[22380.044005] facetimehd 0000:04:00.0: ISP second int after 0ms
[22380.044030] facetimehd 0000:04:00.0: Channel description table at 00800000
[22380.054152] facetimehd 0000:04:00.0: magic value: 00000000 after 0 ms
[22380.054153] facetimehd 0000:04:00.0: Enabling interrupts
[22380.054342] FWMSG: 
[22380.054364] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[22380.054374] FWMSG: 
[22380.054387] FWMSG:   Restart count : 0
[22380.054408] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[22380.054421] FWMSG:   System Clock : 200000000 Hz
[22380.054434] FWMSG:   Processor mode : SUPERVISOR
[22380.054471] FWMSG:   Cache architecture type : SEPARATE
[22380.054487] FWMSG:   Cache type : WRITEBACK
[22380.054502] FWMSG:   Data Cache Line : 32 [0x20]
[22380.054516] FWMSG:   Boot arguments entries : 2
[22380.054657] FWMSG:     0000: 0x00000000 0x00000000
[22380.054677] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[22380.054692] FWMSG:   Main memory :   base : 0x00000000
[22380.054708] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[22380.054726] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[22380.054744] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[22380.054761] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[22380.054893] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[22380.054908] FWMSG:   Shared memory : base : 0x00800000
[22380.054927] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[22380.054947] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[22380.055102] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[22380.055118] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[22380.055164] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[22380.055183] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[22380.055198] FWMSG:   Application : adc [release]
[22380.055213] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[22380.055227] FWMSG:   Release : S2ISP-01.43.00
[22380.055240] FWMSG:   H4ISPAPPLE : 11536
[22380.055254] FWMSG:   H4ISPCD : 4081
[22380.055265] FWMSG:   ffw : 4143
[22380.055283] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[22380.055294] FWMSG: 
[22380.055316] FWMSG: -------------------------------------------------------------------------------
[22380.055326] FWMSG: 
[22380.128416] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[22380.128526] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[22380.128769] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[22380.129103] facetimehd 0000:04:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2```

Seem to be missing `facetimehd/1871_01XX.dat`. Not sure if that is required or not?

wvengen, 2017-08-03:
see #104

ivanperez-keera, 2019-06-29:
This issue is old. @bhundven did the solution to #104 address your problem as well?

closed 2019-12-09 by patjak