patjak/facetimehd #135: Enabling facetimehd breaks wl

rberaldo, 2017-09-04:
Hi! Amazing work you guys are doing here. I can confirm the webcam works on my MacBook Air 2017.

However, it breaks the Broadcom `wl` driver in Manjaro (and, presumably, Arch Linux). This laptop comes with the Broadcom BCM4360 wireless card, which requires the `wl` driver to work. Upon installing `bcwc_pcie` and loading the `facetimehd` during boot, the `wl` driver breaks and wifi is not available. `rmmod`ing `facetimehd` doesn't fix the issue. However, if I backlist `facetimehd` and then `modprobe` it _after_ the boot process is done, both drivers will live happily together.

Here are two excerpts from my `journalctl` which I think will prove informative:

```
Sep 04 01:53:12 volare kernel: facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2                                                     
Sep 04 01:53:12 volare kernel: shpchp: Standard Hot Plug PCI Controller Driver version: 0.4                                                                                        
Sep 04 01:53:12 volare systemd-modules-load[193]: Inserted module 'facetimehd'
```

and

```
Sep 04 01:53:13 volare kernel: wl 0000:03:00.0: enabling device (0000 -> 0002)                                                                                                     
Sep 04 01:53:13 volare kernel: wl driver 6.30.223.271 (r587334) failed with code 1                                                                                                 
Sep 04 01:53:13 volare kernel: ERROR @wl_cfg80211_detach :                                                                                                                         
Sep 04 01:53:13 volare kernel: NULL ndev->ieee80211ptr, unable to deref wl  
```

For people who value context, the whole thing is here: https://paste.ee/p/wVhyE

it took me a while to realize it was `facetimehd` breaking my wifi, so I'd be happy to provide any help and info so we can, together, corner this bug and then squash it for once and for all.

hobarrera, 2017-09-07:
I ran `wl` and `facetimehd` for quite some time on a 2013 MBA, on Archlinux, so it doesn't seem to affect all scenarios.

Not sure if it's the distro, slightly older kernel (haven't had that laptop in like 6 months), or your specific wireless card.

rberaldo, 2017-09-07:
I could be the kernel. I'm on 4.9.47, which is the current LTS. I'll investigate and report back if anything changes.

rberaldo, 2017-09-07:
Kernel 4.13.0 does not *always* have this problem: of the four times I booted up the computer, only once did `wl` fail to load. `modprob wl` doesn't do any good; I have to reboot the machine.

xgnata, 2017-09-13:
I got the same behavior a few times.
"However, if I backlist  facetimehd  and then  modprobe  it after the boot process is done, both drivers will live happily together."
Yes maybe there is something not fully correct in the driver implementation which disturbs the wifi card. It will be hard to know without the help of a real expert.
As a workaround, one can tell the system to load facetimehd *after* completion of the wl loading process? Of course one can do that with a systemd timer which loads facetimehd let’s say 10s after the boot. I think it is even possible to check is wl is already loaded in the system timer system. Ugly but that should work...but is there a simple way to do that in /etc/modules/… ??

bhcopeland, 2018-01-29:
Not able to use this driver anymore due to this bug. :(

arun-gurung, 2018-03-29:
I have Linux 4.15 on Macbook Air and both facetimehd and wl are working together with no problems. Perhaps you should update your kernel and reinstall facetimehd and wl.

glutamatt, 2018-04-07:
_Context_ 
 - kernel **4.14.31**-1-MANJARO
- **brcmfmac module**  loaded for the wifi card Broadcom Limited BCM43602

----

> ### 😒 I **had to remove** facetimehd firmware and the bcwc_pcie driver to get my wifi card back to work

xgnata, 2018-04-08:
What if you make sure facetimehd is loaded after wl?
There must be something wrong in the facetimehd initialization whcih
disturb the wireless card. We need someone who knows something about pcie
drivers...

or mute
the thread

glutamatt, 2018-04-08:
> What if you make sure facetimehd is loaded after wl?

I don't have any `wl` module : my wifi card is drived by the `brcmfmac` module (the issue title is not the exact context of my problem, but the root cause could probably be the same)

facetimehd and wifi used to work properly together until recently, I can't tell why it broke but it doesn't seem to be a kernel update : the only thing i'm sure is that _I had to remove facetimehd firmware and the bcwc_pcie driver to get my wifi card back to work_

xgnata, 2018-04-08:
Even if you make sure that your wifi driver is fully loaded before
facetimehd? Don't load it during the boot process and load it manually
using modprobe. At least on my mac it used to work.

bbgun7, 2018-04-10:
Had the same issue with Linux 4.16.1-1-MANJARO and brcmfmac on my 2015 macbook pro. How would you go about making sure the wifi driver is fully loaded before facetimehd?

xgnata, 2018-04-11:
Don't put anything facetimehd related in /etc/modules or
/etc/modules-load.d or whatever the name is in your distrib.
Reboot. The facetimehd module should not be loaded automatically and your
wifi should work. Once you have your wifi up and running, you can try :
modprobe facetimehd

It is not an automated method but, if it work, there are ways to
automatise.

glutamatt, 2018-04-25:
so it broke down on its own, and suddenly, it "repaired up" on its own ✨

arch package | version
------------ | -------------
bcwc-pcie-git version |  r254.a22f248-2
facetimehd-firmware |  1.43.0-1
linux kernel | 4.14.34-1

👌