patjak/facetimehd #136: Macbook 2015 not showing video

denverslair, 2017-09-10:
Hi, 
I feel this nearly works, but all I get is a green screen. 
I load the v4l modules, then facetimehd module and run qv4l2, the green LED comes on, but all i get is a green screen. 
Kernel : 4.11, macbook 8,1 (2015) Manjaro
Dmesg gives many reports of `ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x804!` then finally  `facetimehd 0000:02:00.0: IO: timeout`
lspci -vv  & dmesg attached.
Thanks for your help!
[facetime_dmesg.txt](https://github.com/patjak/bcwc_pcie/files/1290926/facetime_dmesg.txt)
[facetime_lspci.txt](https://github.com/patjak/bcwc_pcie/files/1290927/facetime_lspci.txt)

jhanschoo, 2018-06-14:
I'd like to just add that this also isn't working in Macbook 9,1 (Early 2016 Macbook 12")

omidn, 2020-03-18:
I had the same issue on Kernel 4.*, just upgraded to 5.4 LTS on an arch Linux and it started working flawlessly.

francescoblefari, 2020-04-24:
> I had the same issue on the kernel 4.* on an arch Linux and a Macbook Pro 2015, just upgraded the kernel to 5.4 LTS and it started working flawlessly.

have you seen this problem?

modprobe: FATAL: Module facetimehd not found in directory /lib/modules/5.4.23-1-MANJARO
done