patjak/facetimehd #137: Automatically create DEB via travis/launchpad

ivanperez-keera, 2017-09-12:
Most people use debian and Ubuntu.

To make this driver used more widely, it would be great to have a DEB. This could be created automatically via a combination of travis and launchpad PPAs.

I can't contribute time, but I can try it and I can contribute to a bounty.

ivanperez-keera, 2017-09-12:
A minot technical detail that may be obvious to some but some beginners who want to give this a try might have more trouble with that:

There soft of is something already: there's a `deb` rule in `firmware/Makefile`, but that can only be executed in the user's machine because it needs to grab the firmware from the OSX installation (for legal reasons, I guess?). So that part would probably have to be re-run by the deb upon installation (and, also, every time the firmware is upgraded in OSX).

ivanperez-keera, 2017-09-12:
This issue is related:

https://github.com/patjak/bcwc_pcie/issues/107

This gist is a really good starting point:

https://gist.github.com/Stono/990ea9f0b3c41606c292f00382d421bf

wvengen, 2017-09-12:
There already is a `dkms.conf`, so running `dkms add .` in the source tree should already do everything necessary (when [dkms](https://en.wikipedia.org/wiki/Dynamic_Kernel_Module_Support) is installed), including rebuilding when a new kernel is installed.

Looking at existing dkms packages like [ddcci-dkms](https://sources.debian.net/src/ddcci-driver-linux/0.3.1-1/) and [dm-writeboost-dkms](https://sources.debian.net/src/dm-writeboost/2.2.7-1/) should give a good starting point for making a debian package for the driver.

For the firmware, there could be package that downloads and extracts the firmware file from the known location, just like the flash plugin or msttcorefonts do. Maybe there is a user agreement / license that needs to be agreed to, in which case that should be shown first.