patjak/facetimehd #138: Module refuses to load on latest Arch kernel (4.12.12-1-ARCH)

slashrsm, 2017-09-15:
When I try to load the module I get the following message:

`
➜  sudo modprobe facetimehd
modprobe: ERROR: could not insert 'facetimehd': Exec format error
`

There is also an error in dmesg:

`
[  577.828988] facetimehd: disagrees about version of symbol module_layout
`

Any idea how to fix this?

hobarrera, 2017-09-15:
Did you install it via DKMS? If you didn't you probably need to manually rebuild the module.

slashrsm, 2017-09-15:
I did. Actually, I tried DKMS and manually rebuilding. Same result.

hobarrera, 2017-09-15:
That's odd. Arch user (and the downstream package maintainer too) here, and the module has worked on recent arch kernels (currently 4.13.2-1).

Can you check if 4.13.2-1 give the same error?

slashrsm, 2017-09-28:
Today my system updated to 4.13.3-1 and auto-resolved the problem. Thanks!

closed 2017-09-28 by slashrsm