patjak/facetimehd #139: Facetime HD not working on skypefolinux MBP 11.3

gianlucapettinello, 2017-10-14:
I installed skypeforlinux-bin (Archlinux) and the camera doesn't work. In settings it gives a black rectangle. The camera works in Cheese

Tried to preload V4l2 compatibility but no success.

Thanks for your help

Gianluca

WhyNotHugo, 2017-10-14:
This is the same as #123, which is an upstream chromium bug (Skype internally uses chromium).

The upcoming chromium release fixes this, so it depends on when this fix trickles down to Skype.

robertoswald, 2020-06-08:
Since todays update it is working on my MacbookPro 13" 2013.

rrthomas, 2020-06-08:
Still not working for me on MBP 11.1 (2015), sadly!

robertoswald, 2020-06-09:
- ubuntu 20.04 (5.4.0-33-generic)
- Skype version 8.61.0.77
- facetimhd driver imstalled via this script

**/usr/local/sbin/enable_facetimehd.sh**
```
#!/bin/bash

KERNEL=`uname -r`
MODULE=/lib/modules/$KERNEL/extra/facetimehd.ko

check_it()
{
  if [ ! $? = 0 ]; then
    echo "Something get wrong!"
    exit 2
  fi
}

new_branch_and_install()
{
  git checkout master; git pull
  git checkout -b $KERNEL
  make
  check_it
  make install
  git add -A; git commit -m "`date`"
}     

load_module()
{
  depmod
  modprobe -r bdc_pci
  modprobe facetimehd
}

do_install()
{
  cd /usr/local/src
  
  if [ ! -d facetimehd-firmware ]; then
    git clone https://github.com/patjak/facetimehd-firmware.git
    check_it
  fi

  if [ ! -d bcwc_pcie ]; then
    git clone https://github.com/patjak/bcwc_pcie.git
    check_it
  fi
  
  cd facetimehd-firmware
  new_branch_and_install
  
  cd ../bcwc_pcie
  new_branch_and_install
}

if [ ! -f $MODULE ]; then
  do_install
fi

load_module
```

patjak, 2020-06-09:
Great. Thanks for testing.

closed 2020-06-09 by patjak