patjak/facetimehd #141: Webcam not working in Ubuntu 17.10 on a Macbook air 7.1

luciomarinelli, 2017-11-01:
I decided to replace MacOs with a real OS on my wife's Macbook air, so I followed Ubuntu tutorial and successfully installed Ubuntu 17.10 on her Macbook air 7.1 (dual boot so far).
Everything seems to work out of the box apart from the webcam. I reached patjak github page and followed instructions to install the facetimehd driver, however it still does not work. Apparently the /dev/video ore /dev/video0 was not created. I also tried to blacklist modprobe bdc_pci and rebooted but still cheese or mplayer tv:// do not work.

Here is the dmesg output after freshly loading the facetimehd module after a reboot:

[  239.595147] show_signal_msg: 12 callbacks suppressed
[  239.595152] nautilus[1378]: segfault at fffffffffffffe80 ip 00007f77b7c60367 sp 00007ffe67f45198 error 5 in libgtk-3.so.0.2200.24[7f77b7982000+6fc000]
[  940.649699] media: Linux media interface: v0.10
[  940.656803] Linux video capture interface: v2.00
[  940.665093] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[  940.665301] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[  940.671032] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[  940.671064] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[  940.681011] facetimehd 0000:02:00.0: PLL reset finished
[  940.681012] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[  940.681024] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[  940.690979] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[  940.710897] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[  940.710911] facetimehd 0000:02:00.0: STRAP valid
[  940.710912] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[  940.710918] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[  940.710927] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[  940.710938] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[  940.710938] facetimehd 0000:02:00.0: Using step size 151
[  940.710943] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[  940.710947] facetimehd 0000:02:00.0: Virtual VTT enabled
[  940.731358] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[  940.731387] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[  940.731598] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[  941.199765] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[  941.358488] facetimehd 0000:02:00.0: ISP woke up after 0ms
[  941.358494] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[  941.358497] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[  941.368447] facetimehd 0000:02:00.0: ISP second int after 0ms
[  941.368449] facetimehd 0000:02:00.0: Channel description table at 00800000
[  941.378563] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[  941.378565] facetimehd 0000:02:00.0: Enabling interrupts
[  941.378768] FWMSG: 
[  941.378789] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[  941.378798] FWMSG: 
[  941.378810] FWMSG:   Restart count : 0
[  941.378830] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[  941.378951] FWMSG:   System Clock : 200000000 Hz
[  941.378964] FWMSG:   Processor mode : SUPERVISOR
[  941.378978] FWMSG:   Cache architecture type : SEPARATE
[  941.378991] FWMSG:   Cache type : WRITEBACK
[  941.379004] FWMSG:   Data Cache Line : 32 [0x20]
[  941.379017] FWMSG:   Boot arguments entries : 2
[  941.379030] FWMSG:     0000: 0x00000000 0x00000000
[  941.379047] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[  941.379061] FWMSG:   Main memory :   base : 0x00000000
[  941.379192] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[  941.379209] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[  941.379226] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[  941.379242] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[  941.379259] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[  941.379273] FWMSG:   Shared memory : base : 0x00800000
[  941.379416] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[  941.379435] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[  941.379455] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[  941.379516] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[  941.379531] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[  941.379548] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[  941.379580] FWMSG:   Application : adc [release]
[  941.379594] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[  941.379607] FWMSG:   Release : S2ISP-01.43.00
[  941.379642] FWMSG:   H4ISPAPPLE : 11536
[  941.379654] FWMSG:   H4ISPCD : 4081
[  941.379665] FWMSG:   ffw : 4143
[  941.379734] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[  941.379744] FWMSG: 
[  941.379764] FWMSG: -------------------------------------------------------------------------------
[  941.379773] FWMSG: 
[  941.453108] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[  941.453273] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[  941.453491] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[  941.453734] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2

It would be great if also the webcam would work!
Thank you

trendsetter37, 2018-10-18:
Getting the same thing for `Ubuntu 18.04` as well.

heewa, 2019-06-14:
Same for me on Ubuntu 19.04, Macbook Air 2013. I don't mind doing some debug work, if anyone can guide me on where to look, including sources (though I'm not familiar with this stuff specifically).

diogoeichert, 2022-03-22:
Still not working on MacBook Air 7,1 running Ubuntu 20.04.4 LTS.

patjak, 2022-03-23:
There are no visible errors in the logs above. Are you getting the same dmesg output? Are you also missing the /dev/video0 device? What does "ls -la /dev/video*" show you?

diogoeichert, 2022-03-24:
Output from `ls -la /dev/video*`:
```
ls: cannot access '/dev/video*': No such file or directory
```

Output from `dmesg` contains no "facetimehd" entries.

Output from `modprobe facetimehd`:
```
modprobe: FATAL: Module facetimehd not found in directory /lib/modules/5.13.0-37-generic
```

Also:
```
$ ls -ld /lib/modules/5.13.0-37-generic/extra/facetimehd.ko 
-rw-r--r-- 1 root root 3729824 Mär 24 07:34 /lib/modules/5.13.0-37-generic/extra/facetimehd.ko
$ ls -ld /usr/lib/modules/5.13.0-37-generic/extra/facetimehd.ko 
-rw-r--r-- 1 root root 3729824 Mär 24 07:34 /usr/lib/modules/5.13.0-37-generic/extra/facetimehd.ko
```

diogoeichert, 2022-03-24:
In my case, I was missing `depmod`, but now I've found the correct docs for the driver, thanks.

luciomarinelli, 2022-03-25:
Following my original issue, I just wanted to say that I succeeded in having the webcam working on the MacBook Air 7.1 running Ubuntu 18.04 and following the instructions provided in [patjak/facetimehd](https://github.com/patjak/facetimehd/wiki/Installation).
The only thing that is not working for me is the "auto-compile on kernal update" procedure, so I manually compile the package after every kernel update.

diogoeichert, 2022-04-02:
What must be done for this driver to get included in the regular distribution, e.g. in Ubuntu (even if it must go in the optional drivers pack)?