patjak/facetimehd #142: Suggested Clarification in DKIM Installation Section

robertpeteuil, 2017-11-19:
Thank you for working and documenting this, it's awesome.  It's working on my 2013 Mac Air with Ubuntu 16.04.03 & kernel 4.10.0-38.

However, during the DKIM process I ran into a small quirk - the command below copied the deb file to my user's home instead of `/root`.  

> Copy deb locally: `# cp /var/lib/dkms/facetimehd/0.1/deb/facetimehd-dkms_0.1_all.deb ~`

I believe this occurred because I assumed root with `sudo -s`, which inherits environment variables from the original user environment.  Since there are many ways to perform tasks as root, and many do not reassign `~`, I suggest explicitly stating `/root` as the destination.

wvengen, 2017-12-28:
Good idea, I've changed it.

robertpeteuil, 2018-01-30:
I just realize I never thanked you and closed this.

Thanks for the quick change!

closed 2018-01-30 by robertpeteuil