patjak/facetimehd #143: Probe failed with error -22 on MacBookPro14,1

gohai, 2017-11-20:
I am trying this driver with Fedora Rawhide (4.15-based) on a new MacBook Pro from this summer:

```
[   61.909210] facetimehd 0000:03:00.0: Found FaceTime HD camera with device id: 1570
[   61.909628] facetimehd 0000:03:00.0: Setting 64bit DMA mask
[   61.915015] facetimehd 0000:03:00.0: S2 PCIe link init succeeded
[   61.915044] facetimehd 0000:03:00.0: Refclk: 24MHz (0x2
[   61.925070] facetimehd 0000:03:00.0: PLL reset finished
[   61.925072] facetimehd 0000:03:00.0: Waiting for S2 PLL to lock at 450 MHz
[   61.925085] facetimehd 0000:03:00.0: S2 PLL is locked after 10 us
[   61.935119] facetimehd 0000:03:00.0: S2 PLL is in bypass mode
[   61.955197] facetimehd 0000:03:00.0: DDR40 PHY PLL locked on safe settings
[   61.955211] facetimehd 0000:03:00.0: STRAP valid
[   61.955213] facetimehd 0000:03:00.0: Configuring DDR PLLs for 450 MHz
[   61.955221] facetimehd 0000:03:00.0: DDR40 PLL is locked after 0 us
[   61.955230] facetimehd 0000:03:00.0: First DDR40 VDL calibration completed after 2 us
[   61.955241] facetimehd 0000:03:00.0: Second DDR40 VDL calibration completed after 2 us
[   61.955243] facetimehd 0000:03:00.0: Using step size 154
[   61.955248] facetimehd 0000:03:00.0: VDL set to: coarse=0x10008, fine=0x1011c
[   61.955253] facetimehd 0000:03:00.0: Virtual VTT enabled
[   61.975815] facetimehd 0000:03:00.0: S2 DRAM memory address: 0x22159559
[   61.975829] facetimehd 0000:03:00.0: Rewrite DDR mode registers succeeded
[   61.976032] facetimehd 0000:03:00.0: Full memory verification succeeded! (0)
[   62.422493] facetimehd 0000:03:00.0: Loaded firmware, size: 1392kb
[   62.573660] facetimehd 0000:03:00.0: ISP woke up after 0ms
[   62.573668] facetimehd 0000:03:00.0: Number of IPC channels: 7, queue size: 44865
[   62.573743] facetimehd 0000:03:00.0: Firmware requested heap size: 3072kb
[   62.583778] facetimehd 0000:03:00.0: ISP second int after 0ms
[   62.583781] facetimehd 0000:03:00.0: Channel description table at 00800000
[   62.594070] facetimehd 0000:03:00.0: magic value: 00000000 after 0 ms
[   62.594073] facetimehd 0000:03:00.0: Enabling interrupts
[   62.729240] facetimehd 0000:03:00.0: fthd_hw_deinit
[   62.729904] facetimehd: probe of 0000:03:00.0 failed with error -22th device id: 1570
[   61.909628] facetimehd 0000:03:00.0: Setting 64bit DMA mask
[   61.915015] facetimehd 0000:03:00.0: S2 PCIe link init succeeded
[   61.915044] facetimehd 0000:03:00.0: Refclk: 24MHz (0x2
[   61.925070] facetimehd 0000:03:00.0: PLL reset finished
[   61.925072] facetimehd 0000:03:00.0: Waiting for S2 PLL to lock at 450 MHz
[   61.925085] facetimehd 0000:03:00.0: S2 PLL is locked after 10 us
[   61.935119] facetimehd 0000:03:00.0: S2 PLL is in bypass mode
[   61.955197] facetimehd 0000:03:00.0: DDR40 PHY PLL locked on safe settings
[   61.955211] facetimehd 0000:03:00.0: STRAP valid
[   61.955213] facetimehd 0000:03:00.0: Configuring DDR PLLs for 450 MHz
[   61.955221] facetimehd 0000:03:00.0: DDR40 PLL is locked after 0 us
[   61.955230] facetimehd 0000:03:00.0: First DDR40 VDL calibration completed after 2 us
[   61.955241] facetimehd 0000:03:00.0: Second DDR40 VDL calibration completed after 2 us
[   61.955243] facetimehd 0000:03:00.0: Using step size 154
[   61.955248] facetimehd 0000:03:00.0: VDL set to: coarse=0x10008, fine=0x1011c
[   61.955253] facetimehd 0000:03:00.0: Virtual VTT enabled
[   61.975815] facetimehd 0000:03:00.0: S2 DRAM memory address: 0x22159559
[   61.975829] facetimehd 0000:03:00.0: Rewrite DDR mode registers succeeded
[   61.976032] facetimehd 0000:03:00.0: Full memory verification succeeded! (0)
[   62.422493] facetimehd 0000:03:00.0: Loaded firmware, size: 1392kb
[   62.573660] facetimehd 0000:03:00.0: ISP woke up after 0ms
[   62.573668] facetimehd 0000:03:00.0: Number of IPC channels: 7, queue size: 44865
[   62.573743] facetimehd 0000:03:00.0: Firmware requested heap size: 3072kb
[   62.583778] facetimehd 0000:03:00.0: ISP second int after 0ms
[   62.583781] facetimehd 0000:03:00.0: Channel description table at 00800000
[   62.594070] facetimehd 0000:03:00.0: magic value: 00000000 after 0 ms
[   62.594073] facetimehd 0000:03:00.0: Enabling interrupts
[   62.729240] facetimehd 0000:03:00.0: fthd_hw_deinit
[   62.729904] facetimehd: probe of 0000:03:00.0 failed with error -22
```
I don't have the `bdc_pci` driver installed.

aijanai, 2018-01-02:
Same here:
MacbookPro 14,1
Ubuntu 16.04.3 LTS, 
4.10.0-42-generic #46~16.04.1-Ubuntu SMP
```
[ 9695.712244] FWMSG: 
[ 9695.712265] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[ 9695.712275] FWMSG: 
[ 9695.712287] FWMSG:   Restart count : 0
[ 9695.712308] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[ 9695.712321] FWMSG:   System Clock : 200000000 Hz
[ 9695.712335] FWMSG:   Processor mode : SUPERVISOR
[ 9695.712349] FWMSG:   Cache architecture type : SEPARATE
[ 9695.712363] FWMSG:   Cache type : WRITEBACK
[ 9695.712376] FWMSG:   Data Cache Line : 32 [0x20]
[ 9695.712389] FWMSG:   Boot arguments entries : 2
[ 9695.712403] FWMSG:     0000: 0x00000000 0x00000000
[ 9695.712456] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[ 9695.712471] FWMSG:   Main memory :   base : 0x00000000
[ 9695.712588] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[ 9695.712606] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[ 9695.712623] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[ 9695.712640] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[ 9695.712659] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[ 9695.712673] FWMSG:   Shared memory : base : 0x00800000
[ 9695.712791] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[ 9695.712811] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[ 9695.712951] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[ 9695.712967] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[ 9695.712983] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[ 9695.713001] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[ 9695.713014] FWMSG:   Application : adc [release]
[ 9695.713029] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[ 9695.713042] FWMSG:   Release : S2ISP-01.43.00
[ 9695.713054] FWMSG:   H4ISPAPPLE : 11536
[ 9695.713066] FWMSG:   H4ISPCD : 4081
[ 9695.713077] FWMSG:   ffw : 4143
[ 9695.713094] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[ 9695.713104] FWMSG: 
[ 9695.713348] FWMSG: -------------------------------------------------------------------------------
[ 9695.713357] FWMSG: 
[ 9695.815293] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 2667: Sensor is null after hNVStorage Validate

[ 9695.845393] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[ 9695.845601] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[ 9695.845872] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[ 9695.846209] no set file for sensorid 0005 0248 found
[ 9695.846216] facetimehd 0000:03:00.0: fthd_hw_deinit
[ 9695.846331] facetimehd: probe of 0000:03:00.0 failed with error -22

```

patjak, 2019-12-10:
This is fixed in: https://github.com/patjak/bcwc_pcie/pull/188

closed 2019-12-10 by patjak