patjak/facetimehd #145: Facetime HD not working in Skype Preview

arun-gurung, 2017-11-29:
Report:

Facetime HD works in Cheese and OBS, but not in Skype Preview version 8.11.76.8. 

My Logitech HD Pro Webcam C920 works fine in Skype.

Linux kernel 4.11.

wvengen, 2017-12-28:
This is probably a duplicate of #139 (the Chrome bug), right?

estathop, 2018-04-07:
it is probably the chromium bug, I can confirm the same problem with Skype and Facebook Messenger on FireFox, Ubuntu 17.10 Macbook Air 2015

WhyNotHugo, 2018-04-10:
It's the same bug as older chrome versions. Skype preview is basically a webapp with an older chromium embedded.

trackermb, 2018-12-14:
Hi there I have the same issue the facetime camera (macbook pro 2014) works in cheese and in hangouts, But doesnt work on skype in chrome, firefox or on the app its self.  Is there something I have missed?

arun-gurung, 2018-12-15:
Facetime HD works with Viber on Linux, but still no with "skypeforlinux". Still waiting...

closed 2019-03-20 by arun-gurung