patjak/facetimehd #146: Problem with Ubuntu 17.10

mtashackori, 2018-01-06:
after the command:
dkms mkdeb -m facetimehd -v 0.1 --source-only

I gave the below error:

DKMS: mkdeb completed.
Moving built files to /var/lib/dkms/facetimehd/0.1/deb...mv: cannot stat '/tmp/dkms.87AjjZ/facetimehd-dkms_0.1_amd64.deb': No such file or directory
(bad exit status: 1)

xkill, 2018-02-20:
I have the same problem, but seems a problem with mkdeb, not a problem of bcwc_pcie (Works fine on Ubuntu 16.04)

cbluth, 2018-03-15:
@mtashackori use this patch against /usr/bin/dkms
https://bugs.debian.org/cgi-bin/bugreport.cgi?att=1;bug=832558;filename=revert_all.patch;msg=5

Elettrotecnica, 2019-06-19:
Sorry for necroposting, but as I have stumbled in the issue myself I just wanted to suggest against patching dkms, just specifying the **-k all** flag on the command line does the trick