patjak/facetimehd #147: modprobe: FATAL: Module facetimehd not found in directory /lib/modules/4.13.0-26-generic

antiNeuron, 2018-01-12:
My initial problem is as I switched from OS to Ubuntu and now using a MacBook Air machine my built-in webcam wasn't detected by any cam software (cheese, hangout, ..etc). So, I tried to use the facetimehd module.

I did my best to follow all the steps to extract and then install it on Ubuntu 16.4.03 LTS (4.10.0.42 kernal).
In the last step, where I run mplay tv://, I get errors. The part of suspending and reloading I honestly didn't get and not sure if it does make a difference here.

Thanks in advance!

alex-shpilkin, 2018-01-12:
Try rebooting once and then redoing the install process from scratch (delete the old files!). This looks like what you'd get if you have a newer kernel (4.13.0) installed after a system update, but you haven't yet rebooted so your system is still running the older one (4.10.0). A kernel update is one of the very few situations where you actually have to reboot a Linux system.

antiNeuron, 2018-01-14:
Yes, it did work! Rebooting and re-installing .. Thank you!

closed 2018-01-14 by antiNeuron

alex-shpilkin, 2018-01-14:
@antiNeuron Note that, unless you set up DKMS (also described in the instructions), you'll have to rebuild & reinstall manually after each kernel update. This is the nature of working with out-of-tree kernel modules. (I'm also not sure what Ubuntu does with modules for nonexistent kernels—you might need to delete old builds manually once in a while.)