patjak/facetimehd #150: Patrik Jakobsson are you here :)

fakbill, 2018-01-30:
Hello,

It looks like the author of this driver is now working on something else.
It is a bit sad because this driver could end up in the mainline kernel if someone take the job over.
The risk is that changes in the kernel will slowly kill this driver and all the work will be lost.
I don't know enough about these Linux drivers to continue the work. However, we could ask on the video mailing list if someone is interested in maintaining bcwc_pcie. Maybe patjak is still here and he plans to keep this driver inline with the kernel API.
What do you think?

rodrigoaguilera, 2018-02-27:
From the last comments here https://github.com/patjak/bcwc_pcie/issues/9
I tought @sschnelle was going to take over the project with a second hand macbook.

fakbill, 2018-03-09:
Very good but I see no progress in the code so far :(

closed 2018-03-09 by fakbill

fakbill, 2018-07-04:
Looks like nothing is going on on @sschnelle side :(
I would be a bit sad to let this driver die (it will slowly die out if no one has time to fix it each time the kernel API is going to change).

cengizIO, 2018-07-04:
Why would it die? We're here to fork and update it. Right?

BTW: does the license allow this to be distributed in mainline?

fakbill, 2018-08-07:
If no-one as time to update it then it will slowly die because it won't compile within up to date kernels.

Yes the licence is OK for mainline inclusion. However, to push it into the mainline, someone needs to write the patches, post them on the lkml, answer the numerous comments and to find the time to maintain it (it should no be a lot of work once it is in mainline).