patjak/facetimehd #154: Can't init facetimehd

otisliao1, 2018-05-09:
Hi there,

Thanks for the hard work of creating this driver. I installed it on my Macbook Pro 15" (Middle 2015) and I encounter this issue of PCIe link failure, could somebody do me a favor?

> [  707.667246] Linux video capture interface: v2.00
> [  707.692278] facetimehd 0000:04:00.0: Found FaceTime HD camera with device id: 1570
> [  707.717650] facetimehd 0000:04:00.0: Setting 64bit DMA mask
> [  707.721040] facetimehd 0000:04:00.0: Failed to init S2 PCIe link: ffffffff
> [  707.721127] facetimehd: probe of 0000:04:00.0 failed with error -5

otisliao1, 2018-05-09:
I reboot my laptop and the init process goes further and then stop at error -22
```

[   57.622994] Linux video capture interface: v2.00
[   57.646230] facetimehd 0000:04:00.0: Found FaceTime HD camera with device id: 1570
[   57.648637] facetimehd 0000:04:00.0: Setting 64bit DMA mask
[   57.658781] facetimehd 0000:04:00.0: S2 PCIe link init succeeded
[   57.658834] facetimehd 0000:04:00.0: Refclk: 25MHz (0xa)
[   57.670841] facetimehd 0000:04:00.0: PLL reset finished
[   57.670843] facetimehd 0000:04:00.0: Waiting for S2 PLL to lock at 450 MHz
[   58.048512] facetimehd 0000:04:00.0: Failed to lock S2 PLL: 0xc902c902
[   58.068592] facetimehd 0000:04:00.0: DDR40 PHY PLL locked on safe settings
[   58.068618] facetimehd 0000:04:00.0: STRAP valid
[   58.068619] facetimehd 0000:04:00.0: Configuring DDR PLLs for 450 MHz
[   58.068753] facetimehd 0000:04:00.0: DDR40 PLL is locked after 0 us
[   58.068767] facetimehd 0000:04:00.0: First DDR40 VDL calibration completed after 1 us
[   58.068783] facetimehd 0000:04:00.0: Second DDR40 VDL calibration completed after 1 us
[   58.068784] facetimehd 0000:04:00.0: Using step size 153
[   58.068793] facetimehd 0000:04:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[   58.068801] facetimehd 0000:04:00.0: Virtual VTT enabled
[   58.089318] facetimehd 0000:04:00.0: S2 DRAM memory address: 0x22159559
[   58.089346] facetimehd 0000:04:00.0: Rewrite DDR mode registers succeeded
[   58.089840] facetimehd 0000:04:00.0: Full memory verification succeeded! (0)
[   58.638407] facetimehd 0000:04:00.0: Loaded firmware, size: 1392kb
[   58.785628] facetimehd 0000:04:00.0: ISP woke up after 0ms
[   58.785635] facetimehd 0000:04:00.0: Number of IPC channels: 7, queue size: 44865
[   58.785638] facetimehd 0000:04:00.0: Firmware requested heap size: 3072kb
[   58.795625] facetimehd 0000:04:00.0: ISP second int after 0ms
[   58.795628] facetimehd 0000:04:00.0: Channel description table at 00800000
[   58.807853] facetimehd 0000:04:00.0: magic value: 00000000 after 0 ms
[   58.807855] facetimehd 0000:04:00.0: Enabling interrupts
[   58.808314] FWMSG: 
[   58.808433] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[   58.808445] FWMSG: 
[   58.808482] FWMSG:   Restart count : 0
[   58.808584] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[   58.808636] FWMSG:   System Clock : 200000000 Hz
[   58.808701] FWMSG:   Processor mode : SUPERVISOR
[   58.808762] FWMSG:   Cache architecture type : SEPARATE
[   58.808805] FWMSG:   Cache type : WRITEBACK
[   58.808857] FWMSG:   Data Cache Line : 32 [0x20]
[   58.808907] FWMSG:   Boot arguments entries : 2
[   58.808961] FWMSG:     0000: 0x00000000 0x00000000
[   58.809052] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[   58.809110] FWMSG:   Main memory :   base : 0x00000000
[   58.809195] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[   58.809342] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[   58.809428] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[   58.809509] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[   58.809600] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[   58.809660] FWMSG:   Shared memory : base : 0x00800000
[   58.809811] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[   58.809920] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[   58.810021] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[   58.810089] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[   58.810159] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[   58.810252] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[   58.810303] FWMSG:   Application : adc [release]
[   58.810364] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[   58.810410] FWMSG:   Release : S2ISP-01.43.00
[   58.810449] FWMSG:   H4ISPAPPLE : 11536
[   58.810482] FWMSG:   H4ISPCD : 4081
[   58.810510] FWMSG:   ffw : 4143
[   58.810585] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[   58.810598] FWMSG: 
[   58.810715] FWMSG: -------------------------------------------------------------------------------
[   58.810726] FWMSG: 
[   58.900176] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[   58.900510] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[   58.900907] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[   58.901275] no set file for sensorid 0000 0000 found
[   58.901282] facetimehd 0000:04:00.0: fthd_hw_deinit
[   58.901496] facetimehd: probe of 0000:04:00.0 failed with error -22

```

otisliao1, 2018-05-09:
Cheese says "No Device found"
I am on OpenSUSE 42.3 Leap

patjak, 2019-12-10:
Looks like your camera doesn't report a proper sensor id. We've change so that this is no longer fatal. The latest version in master branch will fix your problem.

closed 2019-12-10 by patjak