patjak/facetimehd #155: Can't compile after upgrade to F28

dariotislar, 2018-05-11:
I'm getting error after upgrading to F28 and Kernel 4.16:
`make -C /lib/modules/4.16.5-300.fc28.x86_64/build M=/home/dario/code/util/bcwc_pcie modules
make[1]: Entering directory '/usr/src/kernels/4.16.5-300.fc28.x86_64'
  CC [M]  /home/dario/code/util/bcwc_pcie/fthd_hw.o
In file included from ./include/linux/delay.h:22,
                 from /home/dario/code/util/bcwc_pcie/fthd_hw.c:21:
./include/linux/kernel.h:6:10: fatal error: stdarg.h: No such file or directory
 #include <stdarg.h>
          ^~~~~~~~~~
compilation terminated.
make[2]: *** [scripts/Makefile.build:325: /home/dario/code/util/bcwc_pcie/fthd_hw.o] Error 1
make[1]: *** [Makefile:1565: _module_/home/dario/code/util/bcwc_pcie] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.16.5-300.fc28.x86_64'
make: *** [Makefile:9: all] Error 2
`

abustany, 2018-05-11:
`stdarg.h` should come with your gcc installation:

```
$ rpm -qf /usr/lib/gcc/x86_64-redhat-linux/8/include/stdarg.h
gcc-8.1.1-1.fc28.x86_64
```

dariotislar, 2018-05-11:
That package is installed, it might be that some configuration got broken during the upgrade?
How can I debug/fix this?

abustany, 2018-05-11:
Is the `stdarg.h` file present on your system ? If not, I'd suggest reinstalling the package. If yes, I'd first start off a clean repository and check if it works. If it doesn't, try to compile a trivial program using stdarg.h:

```c
#include <stdarg.h>
int main(int argc, char **argv) { return 0; }
```

fisxoj, 2018-05-17:
I had this problem and, after I found this issue and verified `stdarg.h` existed, running `make clean` before `make` was enough to sort things out for me!

dariotislar, 2018-05-18:
Wooow, I'm such a noob :)
`make clean` did the trick
Thanks alot.

closed 2018-05-18 by dariotislar

ktdreyer, 2018-05-29:
By the way, for Fedora I'm building RPMs at https://ktdreyer.fedorapeople.org/macbook/ . Packaging and code at https://github.com/ktdreyer/facetimehd-kmod-rpm