patjak/facetimehd #156: "Refused to change power state" with kernel 4.16.7

abustany, 2018-05-11:
Running on `4.16.7-300.fc28.x86_64`, I get when trying to load facetimehd:

```
[ 3768.285523] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[ 3768.297451] facetimehd 0000:02:00.0: Refused to change power state, currently in D3
[ 3768.297707] facetimehd 0000:02:00.0: Failed to enable MSI
[ 3768.297753] facetimehd: probe of 0000:02:00.0 failed with error -22
```

This happens with both master and mainline branches.

damaestro, 2018-05-14:
On the same kernel, I also am unable to use this device.
```
[   44.705524] Linux video capture interface: v2.00
[   44.720333] facetimehd: loading out-of-tree module taints kernel.
[   44.720430] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[   44.721519] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[   44.721776] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[   44.726766] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[   44.726792] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[   44.736816] facetimehd 0000:02:00.0: PLL reset finished
[   44.736817] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[   44.736829] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[   44.746863] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[   44.766939] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[   44.766952] facetimehd 0000:02:00.0: STRAP valid
[   44.766953] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[   44.766960] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[   44.766969] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[   44.766979] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[   44.766980] facetimehd 0000:02:00.0: Using step size 153
[   44.766985] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[   44.766989] facetimehd 0000:02:00.0: Virtual VTT enabled
[   44.787549] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[   44.787561] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[   44.787755] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[   44.787932] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/firmware.bin failed with error -2
[   44.788053] facetimehd: probe of 0000:02:00.0 failed with error -2
```

damaestro, 2018-05-17:
I didn't follow the directions. When I properly downloaded the firmware and then built, the device works fine.

@abustany `4.16.8-300.fc28.x86_64` is working for me.