patjak/facetimehd #157: modprobe: ERROR: could not insert 'facetimehd': Exec format error with kernel 4.16.13-300.fc28.x86_64

seungjin, 2018-06-05:
It worked well with kernel 4.16.12-300.fc28.x86_64.
I updated kernel to new 4.16.13-300.fc28.x86_64 a few days ago and it shows now "modprobe: ERROR: could not insert 'facetimehd': Exec format error"

cengizIO, 2018-06-05:
Can you share the output of this?

```
$ file /usr/lib/modules/$(uname -r)/extra/facetimehd.ko 
```

seungjin, 2018-06-05:
$ file /usr/lib/modules/4.16.13-300.fc28.x86_64/extra/facetimehd.ko
/usr/lib/modules/4.16.13-300.fc28.x86_64/extra/facetimehd.ko: ELF 64-bit LSB relocatable, x86-64, version 1 (SYSV), BuildID[sha1]=9fe05e31c2e08c52b4728a0c0304b5901d9dbe24, with debug_info, not stripped

cengizIO, 2018-06-05:
Interesting. This might be similar to this: https://github.com/patjak/bcwc_pcie/issues/91

cengizIO, 2018-06-05:
And please make sure that you've followed Fedora specific steps: 

https://github.com/patjak/bcwc_pcie/wiki/Get-Started#get-started-on-fedora

seungjin, 2018-06-06:
I deleted a previous(exisitng, the one I used for my previous kernel) source folder (completely removed) and ran "git clone" (FRESH COPY)
then 
$ cd firmware
$ make
$ make install
then followed https://github.com/patjak/bcwc_pcie/wiki/Get-Started#get-started-on-fedora

Now it works!

Seems when I update fedora kernel I shouldn't recycle existing git source. Instead, delete them and re git clone then install it.

antmachine, 2018-06-18:
Same issue here on Fedora 28. Fresh git clone of repo and following the install instructions again fixes it.

andybrown668, 2018-08-08:
you can just do a 'git clean -fxd' on an existing clone the run the make and install steps - worked on Ubuntu 18.04

ivanperez-keera, 2019-06-29:
If this issue is solved by cleaning the repo and old files, could it be closed, @seungjin ?