patjak/facetimehd #160: Failed to make install on Mint 19

hxingpax, 2018-09-09:
~~~
sudo make install
make -C /lib/modules/4.15.0-33-generic/build M=/home/xing/workspace/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.15.0-33-generic'
  INSTALL /home/xing/workspace/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.15.0-33-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-33-generic'
~~~

System information:

~~~
uname -a
Linux xing-MacBookPro 4.15.0-33-generic #36-Ubuntu SMP Wed Aug 15 16:00:05 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
~~~

hxingpax, 2018-09-09:
Well, so I manually put the facetimed into the system moduels file directory, and loaded it through `modprobe`, and maked sure it is loaded with `lsmod`:

~~~
lsmod | grep facetime
facetimehd             90112  0
videobuf2_dma_sg       16384  1 facetimehd
videobuf2_v4l2         24576  1 facetimehd
videobuf2_core         40960  2 facetimehd,videobuf2_v4l2
videodev              184320  3 facetimehd,videobuf2_core,videobuf2_v4l2
~~~

But , seems like the webcam still does not work when I run cheeese? What should I try next?

hxingpax, 2018-09-09:
Update:

After a reinstall and relaod, finally it works.

closed 2018-09-09 by hxingpax

jpretori, 2019-03-25:
I just encountered the same problem on Ubuntu 18.04LTS (bionic beaver).

```
$ sudo make install
make -C /lib/modules/4.18.0-16-generic/build M=/home/johan/src/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.18.0-16-generic'
  INSTALL /home/johan/src/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:74
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:81
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.18.0-16-generic
make[1]: Leaving directory '/usr/src/linux-headers-4.18.0-16-generic'
```

After completing the rest of the steps in the wiki, the "mplayer test" fails with this message:
```$ mplayer tv://
Creating config file: /home/johan/.mplayer/config
MPlayer 1.3.0 (Debian), built with gcc-7 (C) 2000-2016 MPlayer Team
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing tv://.
TV file format detected.
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: unable to open '/dev/video0': No such file or directory
v4l2: ioctl set mute failed: Bad file descriptor
v4l2: 0 frames successfully processed, 0 frames dropped.

Exiting... (End of file)
```
I tried some things but could not figure it out.  In the end, dmesg gives me this right at the end of the log:
```[20271.577352] media: Linux media interface: v0.10
[20271.581845] videodev: Linux video capture interface: v2.00
[20271.586224] facetimehd: loading out-of-tree module taints kernel.
[20271.586260] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
```

It seems to me that the two things are somehow related.  Is there something I can do to help the next person avoid this difficulty?