patjak/facetimehd #162: MacBook Pro 13.3 Retina problem with Ubuntu 18.04

mtashackori, 2018-10-03:
I've been used the driver on Ubuntu 16.04 perfectly but after upgrade to 18.04 I've got these errors on compiling process:
==========
Makefile:34: recipe for target 'AppleCameraInterface' failed
make: *** [AppleCameraInterface] Error 1
done

Installing firmware
Copying firmware into '//lib/firmware/facetimehd'
install: cannot stat 'firmware.bin': No such file or directory
Makefile:45: recipe for target 'install' failed
make: *** [install] Error 1
========
The ubuntu 18.04 that I used installed cleanly.

mtashackori, 2018-10-07:
I've found the answer:
https://github.com/patjak/bcwc_pcie/issues/41#issuecomment-170334594

trendsetter37, 2018-10-18:
I had to manually change the `Makefile` by hard coding `/usr/lib/firmware/facetimehd` in the install section

closed 2018-10-18 by mtashackori