patjak/facetimehd #163: module verification failed: signature and/or required key missing - tainting kernel

LeoIannacone, 2018-10-10:
Hi,

is this a know issue ?
```
Oct 10 17:01:22 plentific kernel: [62091.587072] brcmfmac: brcmf_inetaddr_changed: fail to get arp ip table err:-52
Oct 10 17:39:01 plentific kernel: [64351.000968] media: Linux media interface: v0.10
Oct 10 17:39:01 plentific kernel: [64351.009743] videodev: Linux video capture interface: v2.00
Oct 10 17:39:01 plentific kernel: [64351.014479] facetimehd: loading out-of-tree module taints kernel.
Oct 10 17:39:01 plentific kernel: [64351.014515] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
```

```
Linux plentific 4.18.0-8-generic #9-Ubuntu SMP Mon Sep 10 16:12:10 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

cengizIO, 2018-10-10:
`loading out-of-tree module taints kernel.`

since this is an out of tree module, it will taint the kernel.

unless you're going to debug the kernel itself, you wouldn't have any problems with that warning.

LeoIannacone, 2018-10-10:
The point is: I can't get this to work.
```
$ ls -l /dev/video0                                                                                                    
ls: cannot access '/dev/video0': No such file or directory
```

LeoIannacone, 2018-10-10:
Ohh found, finally!

I had to add myself to the `video` group. Thanks!

closed 2018-10-10 by LeoIannacone