patjak/facetimehd #169: Build fail for kernel 4.19.3

sunshine69, 2018-11-21:
I got build error, not sure how to fix it

root@stevek-macbook /h/s/s/bcwc_pcie# make
make -C /lib/modules/4.19.3core2h1k-rt/build M=/home/stevek/src/bcwc_pcie modules
make[1]: Entering directory '/mnt/sda4/tmp/linux-4.19'
  CC [M]  /home/stevek/src/bcwc_pcie/fthd_ddr.o
In file included from /home/stevek/src/bcwc_pcie/fthd_ddr.c:21:0:
/home/stevek/src/bcwc_pcie/fthd_drv.h:121:5: warning: "LINUX_VERSION_CODE" is not defined, evaluates to 0 [-Wundef]
 #if LINUX_VERSION_CODE < KERNEL_VERSION(4,8,0)
     ^~~~~~~~~~~~~~~~~~
/home/stevek/src/bcwc_pcie/fthd_drv.h:121:26: warning: "KERNEL_VERSION" is not defined, evaluates to 0 [-Wundef]
 #if LINUX_VERSION_CODE < KERNEL_VERSION(4,8,0)
                          ^~~~~~~~~~~~~~
/home/stevek/src/bcwc_pcie/fthd_drv.h:121:40: error: missing binary operator before token "("
 #if LINUX_VERSION_CODE < KERNEL_VERSION(4,8,0)
                                        ^
scripts/Makefile.build:305: recipe for target '/home/stevek/src/bcwc_pcie/fthd_ddr.o' failed
make[2]: *** [/home/stevek/src/bcwc_pcie/fthd_ddr.o] Error 1
Makefile:1517: recipe for target '_module_/home/stevek/src/bcwc_pcie' failed
make[1]: *** [_module_/home/stevek/src/bcwc_pcie] Error 2
make[1]: Leaving directory '/mnt/sda4/tmp/linux-4.19'
Makefile:12: recipe for target 'all' failed
make: *** [all] Error 2
root@stevek-macbook /h/s/s/bcwc_pcie# 

Thanks

nhed, 2018-11-30:
@sunshine69 

Had the same issue
Now works for me  on 4.19.4-300.fc29.x86_64 once I checked-out the `mainline` branch (instead of `master`)

arun-gurung, 2018-12-27:
I use the latest mainline elrepo kernel (kernel-ml-4.20.0-1.el7.elrepo.x86_64) on CentOS and I get the same output: 

make -C /lib/modules/4.20.0-1.el7.elrepo.x86_64/build M=/root/tmp/bcwc_pcie-master modules
make[1]: Entering directory `/usr/src/kernels/4.20.0-1.el7.elrepo.x86_64'
  CC [M]  /root/tmp/bcwc_pcie-master/fthd_ddr.o
In file included from /root/tmp/bcwc_pcie-master/fthd_ddr.c:21:0:
/root/tmp/bcwc_pcie-master/fthd_drv.h:121:5: warning: "LINUX_VERSION_CODE" is not defined [-Wundef]
 #if LINUX_VERSION_CODE < KERNEL_VERSION(4,8,0)
     ^
/root/tmp/bcwc_pcie-master/fthd_drv.h:121:26: warning: "KERNEL_VERSION" is not defined [-Wundef]
 #if LINUX_VERSION_CODE < KERNEL_VERSION(4,8,0)
                          ^
/root/tmp/bcwc_pcie-master/fthd_drv.h:121:40: error: missing binary operator before token "("
 #if LINUX_VERSION_CODE < KERNEL_VERSION(4,8,0)
                                        ^
make[2]: *** [/root/tmp/bcwc_pcie-master/fthd_ddr.o] Error 1
make[1]: *** [_module_/root/tmp/bcwc_pcie-master] Error 2
make[1]: Leaving directory `/usr/src/kernels/4.20.0-1.el7.elrepo.x86_64'
make: *** [all] Error 2

arun-gurung, 2019-01-01:
I went back to kernel 4.18 and it works. Must be the Makefiles who must get updated.

closed 2019-08-11 by sunshine69