patjak/facetimehd #174: Missing file - MackBook Air

m0rph84, 2019-03-01:
I'm trying to install the driver on a MacBook Air mid-2013 with last version of Pop_Os! Linux but I got this error after the last "make install" before "sudo depmod":

`make -C /lib/modules/4.18.0-15-generic/build
 M=/etc/local/src/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-4.18.0-15- generic'
      INSTALL /etc/local/src/bcwc_pcie/facetimehd.ko
 At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  4.18.0-15-generic
    make[1]: Leaving directory '/usr/src/linux-headers-4.18.0-15-generic'`

a-piece-of-garbage, 2019-03-04:
Same problem, using Fedora 29.

omer531998, 2019-03-08:
same here.. using ubuntu 4.18.0-15-generic #16~18.04.1-Ubuntu SMP x86_64 x86_64 x86_64 GNU/Linux

patjak, 2019-03-12:
Is this actually an error that prevents the module from compiling or being loaded? Looks like the same problem as https://github.com/umlaeute/v4l2loopback/issues/139

sorenmarkert, 2019-03-21:
Same problem here:
`$ uname -a
Linux sm-macbook 4.18.0-16-generic #17-Ubuntu SMP Fri Feb 8 00:06:57 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

m0rph84, 2019-06-02:
Sorry for the very late reply.
I tried the solution posted by ROBERT-MCDOWELL in [umlaeute/v4l2loopback#139](https://github.com/umlaeute/v4l2loopback/issues/139) but it didn't solve the issue. Still getting the same error when trying to compile the module following these steps:

`$ cd /etc/local/src`
`$ git clone https://github.com/patjak/bcwc_pcie.git`
`$ cd bcwc_pcie/firmware`
`$ sudo make`
`$ sudo make install`
`$ cd ..`
`$ sudo make`
`$ sudo make install`
`$ sudo depmod`
`$ sudo modprobe -r bdc_pci`
`$ sudo modprobe facetimehd`

At the last `sudo make install` before `sudo depmod` I got:

`make -C /lib/modules/5.0.0-15-generic/build M=/etc/local/src/bcwc_pcie modules_install`
`make[1]: Entering directory '/usr/src/linux-headers-5.0.0-15-generic
INSTALL /etc/local/src/bcwc_pcie/facetimehd.ko`
`At main.c:160:`
`- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72`
`- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79`
`sign-file: certs/signing_key.pem: No such file or directory`
`DEPMOD  5.0.0-15-generic`
`Warning: modules_install: missing 'System.map' file. Skipping depmod.`
`make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-15-generic'`

Am I missing something?
My system is Pop!_Os 19.04 an Ubuntu based distro with kernel 5.0.0-15-generic

comzeradd, 2019-06-06:
The missing files don't seem to create a problem. I managed to make the camera work by just removing `bdc_pci` first.

So:
```
modprobe -r bdc_pci
```
and then:
```
modprobe facetimehd
```

For making this permanent, just blacklist that module. Create a file under `/etc/modprobe.d/blacklist-camera.conf` with this:
```
blacklist bdc_pci
```

deansay, 2019-06-16:
I got this error.
[ 2363.663753] facetimehd: Unknown symbol mcount (err 0)

deansay, 2019-06-16:
> The missing files don't seem to create a problem. I managed to make the camera work by just removing `bdc_pci` first.
>
> So:
>
> ```
> modprobe -r bdc_pci
> ```
>
> and then:
>
> ```
> modprobe facetimehd
> ```
>
> For [...]
I got this error.
[ 2363.663753] facetimehd: Unknown symbol mcount (err 0)

gedster314, 2020-04-30:
Same issue here.  Was working with earlier kernel but broke with a Ubuntu Kernel update.

I'm just a script kiddie so I don't know the long term ramification of making this change.  I copied it from this page.

https://medium.com/@victorleungtw/install-ubuntu-20-04-lts-on-macbookpro14-1-9f1b6afde43d

After you do the firmware step and cd to the parent folder, edit the make file.  

install: 
  cp facetimehd.ko /lib/modules/$(shell uname -r)/extra/depmod -a

After you do the edit, do the make, the make install and then reboot.  Skip the modprobe stuff.

alexpop, 2021-04-05:
You are missing a signing key to sign the module: `sign-file: certs/signing_key.pem: No such file or directory`

Create the key like this:
```bash
cd /lib/modules/$(uname -r)/build/certs

sudo tee x509.genkey > /dev/null << 'EOF'
[ req ]
default_bits = 4096
distinguished_name = req_distinguished_name
prompt = no
string_mask = utf8only
x509_extensions = myexts
[ req_distinguished_name ]
CN = Modules
[ myexts ]
basicConstraints=critical,CA:FALSE
keyUsage=digitalSignature
subjectKeyIdentifier=hash
authorityKeyIdentifier=keyid
EOF

sudo openssl req -new -nodes -utf8 -sha512 -days 36500 -batch -x509 -config x509.genkey -outform DER -out signing_key.x509 -keyout signing_key.pem

```

closed 2024-06-27 by m0rph84