patjak/facetimehd #178: MacBook 12", 2017: How to configure facetime camera resolution?

marbetschar, 2019-04-07:
Hi there!

Thanks for providing this awesome driver!! However, I'm currently struggling with one small issue: I suspect that the screen resolutions Cheese uses in my environment are too high and therefore it is not able to capture any video.

Here's what I get when querying for supported resolutions:

```
$ sudo lspci -s 03:00.0 -v
03:00.0 Multimedia controller: Broadcom Inc. and subsidiaries 720p FaceTime HD Camera
	Subsystem: Broadcom Inc. and subsidiaries 720p FaceTime HD Camera
	Flags: bus master, fast devsel, latency 0, IRQ 29
	Memory at 92600000 (64-bit, non-prefetchable) [size=64K]
	Memory at b0000000 (64-bit, prefetchable) [size=256M]
	Memory at 92500000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [48] Power Management version 3
	Capabilities: [58] MSI: Enable+ Count=1/1 Maskable- 64bit+
	Capabilities: [68] Vendor Specific Information: Len=44 <?>
	Capabilities: [ac] Express Endpoint, MSI 00
	Capabilities: [100] Advanced Error Reporting
	Capabilities: [13c] Device Serial Number 00-00-00-ff-ff-00-00-00
	Capabilities: [150] Power Budgeting <?>
	Capabilities: [160] Virtual Channel
	Capabilities: [1b0] Latency Tolerance Reporting
	Capabilities: [220] Resizable BAR <?>
	Kernel driver in use: facetimehd
	Kernel modules: bdc_pci
$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Stepwise 320x240 - 1280x720 with step 8/1
	[1]: 'YVYU' (YVYU 4:2:2)
		Size: Stepwise 320x240 - 1280x720 with step 8/1
```

And this is how Cheese looks like:

[image]

Please note I'm not able to select any other resolution in Cheese. The one selected is the only one available.

Any help would be highly appreciated.

marbetschar, 2019-04-11:
Any ideas how I might fix the webcam issue? I'd happily provide a pull request or anything else which takes this further - but I don't know where to start unfortunately. A little hint to get things rolling would be awesome! :roller_coaster:

marbetschar, 2019-04-11:
One more thing: The green webcam light is flashing as soon as the driver gets loaded (`modprobe facetimehd`). This happens even if I manually load the driver. The full `dmesg` output for driver loading is as follows:

```
[  207.109769] media: Linux media interface: v0.10
[  207.165815] videodev: Linux video capture interface: v2.00
[  207.212689] Found FaceTime HD camera with device id: 1570
[  207.212879] facetimehd 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[  209.455366] Loaded firmware, size: 1392kb
[  227.232894] media: Linux media interface: v0.10
[  227.263342] videodev: Linux video capture interface: v2.00
[  227.299196] Found FaceTime HD camera with device id: 1570
[  227.299330] facetimehd 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[  229.541596] Loaded firmware, size: 1392kb
```

marbetschar, 2019-04-22:
Is this driver still under development?

Asking this not to offend anyone, we all get busy from time to time. But I'm wondering because there doesn't seem to be much activity in this repo by the owner. Lots of open issues and no new commits.

That said, I'm wondering if there's any chance to get help at all.

maximilianduell, 2019-05-10:
Probably that is a different camera model. Does it use the same firmware as the 720p cameras?

marbetschar, 2019-05-10:
@maximilianduell thanks for getting back on this 👍

Found the `AppleCameraInterface` binary within `/System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/` thanks to the [firmware extraction script](https://github.com/patjak/facetimehd-firmware/blob/e3bebf37334f3886e7cdbcefaec0938c91516fdf/extract-firmware.sh#L88) as you pointed out.

If I'm on the right track you are suggesting to use `AppleCameraInterface` file from my macOS install, rename it to `facetimehd` and [replace the default `/usr/lib/firmware/facetimehd` under Linux](https://github.com/patjak/bcwc_pcie/wiki/Get-Started#firmware-extraction) .... right?

maximilianduell, 2019-05-10:
If I interpret the extraction script correctly, the file `AppleCameraInterface` is not just the firmware but also contains OSX driver code. You can't just rename it to `firmware.bin` From this file the firmware has to be extracted in some way. 

Perhaps 

     extract-firmware -x AppleCameraInterface 

or 

     extract-firmware --ignore-hashes -x AppleCameraInterface 

can help with this step.

marbetschar, 2019-05-14:
Unfortunately no luck. If I run the script as provided, it fails since my AppleCameraInterface file does not match any known hash. In addition, I changed the script a bit so it ignores the checks (version attached) and exports the firmware regardless if the hash is known or not.

When I run the script, it will export the following files:

```
-rw-r--r-- 1 664K firmware-wnd_105.bin
-rw-r--r-- 1 657K firmware-wnd_138.bin
-rw-r--r-- 1 1.4M firmware-osx_140.bin
-rw-r--r-- 1 1.4M firmware-osx_143_1.bin
-rw-r--r-- 1 1.4M firmware-osx_143_2.bin
-rw-r--r-- 1 1.4M firmware-osx_143_3.bin
```

Now I replaced the `/usr/lib/firmware/facetimehd/firmware.bin` with each one of the osx firmware versions and after each replacement made a reboot. After reach reboot I ran `mpv tv://` and the only thing that indeed improved is the fact, that the green flash light is on as long as the command is running. Other than that nothing changed. In addition I'm only able to execute the command once. The second time nothing happens anymore.

Here's the changed script:
[extract-firmware.sh.txt](https://github.com/patjak/bcwc_pcie/files/3179596/extract-firmware.sh.txt)

Any other ideas? :)

maximilianduell, 2019-05-17:
I do not have much experience with developing drivers. Unfortunately I do not have a Macbook with a 480p camera I could use for testing. If I had one I would try to compare the various functionalities and initialization routines of the driver on the two different hardware models and see if I can get it running. Perhaps I could borrow such a Macbook for a short time in one or two months.

Otherwise without access to the hardware it is hard for me to say what else one could try. But my guess is that it likely is a deeper issue and that some initialization routines perhaps or data conversion routines have to be adapted or implemented differently.

patjak, 2019-12-09:
As Maximilian say, the Macbook 480p camera is likely a model we don't support. Not much I can do without actually having a device to work with. Sorry.

closed 2019-12-09 by patjak

maximeplante, 2020-08-04:
For people encountering this issue: see #196 for a fix. You have to change some values in fthd_isp.c and fthd_v4l2.c.

doctor, 2026-08-17:
The camera on a **MacBook10,1 (12-inch, 2017)** now works fully — 848x588 at
30fps, no SIF errors. Writing up the complete recipe since it needs three
separate things, and any one of them missing looks like "unsupported hardware".

Incidentally the sensor is not 480p as this issue's title guesses: its active
area is **848x588**. The 640x480 that gets reported early on is an ISP fallback
that appears when calibration is missing.

## 1. Firmware 1.43.0 predates these machines

This is the piece that made everything else look hopeless. `facetimehd-firmware`
extracts firmware 1.43.0 from OS X 10.11.5 (2016); the MacBook10,1 shipped in
June 2017 with macOS 10.12.5. The ISP boots perfectly, loads firmware, completes
DDR calibration, answers every command — and finds no sensor:

```
facetimehd 0000:03:00.0: Loaded firmware, size: 1392kb
facetimehd 0000:03:00.0: ISP woke up after 0ms
FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 2667: Sensor is null after hNVStorage Validate
sensor id: 0000 0000
sensor count: 0
```

Since `sensor_id1` is 0, `fthd_isp_cmd_set_loadfile()` matches no case, so no
set file is ever requested either — the two failures compound.

Firmware **5.60.0** from macOS Sierra 10.12.6 fixes it:

```
sensor id: 0005 9774
sensor count: 1
```

PR: patjak/facetimehd-firmware#12

## 2. The 1675 set file is shipped by nobody

With the sensor visible as `0005 9774`, the driver asks for `1675_01XX.dat`.
`facetimehd-data` ships four of the eleven set files, and the Boot Camp driver
those come from does not contain this one at all.

They are all present in macOS, but in the userspace DAL plugin's helper
(`AppleCamera.plugin/Contents/Resources/AppleCameraAssistant`), not in the kext.
Offsets for all eleven, plus the check that validates them, are in
patjak/facetimehd-firmware#3.

Loading it is what makes the ISP report the sensor's real geometry — before it,
a 640x480 fallback; after it, `within [0, 0][848, 588]`.

## 3. The hardcoded 1280x720 crop — already fixed here

@thomas-shirley's 98b55fd ("detect sensor resolution instead of hardcoding
1280x720") handles this correctly by reading geometry from the per-channel
camera config. Without it, the driver commands a 1280x720 crop on the 848x588
array and the sensor interface faults on every frame:

```
CROP -> [0, 0][1280, 720] within [0, 0][848, 588]
FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x804!
facetimehd 0000:03:00.0: IO: timeout
```

Worth flagging for anyone testing: **98b55fd is not in any tagged release**, so
distro packages built from 0.7.0.2 (the Arch `facetimehd-dkms` package, for
instance) still have this bug. Building from master is required for now.

## Result

With all three, on kernel 7.1.8:

```
$ v4l2-ctl -d /dev/video0 --list-formats-ext
	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 848x588
			Interval: Discrete 0.033s (30.000 fps)

CROP -> [0, 0][848, 588] within [0, 0][848, 588]
```

Frames captured cleanly, zero `sifIrq` errors. Happy to test patches on this
hardware if useful — and if someone with a MacBook8,1 or 9,1 can confirm, the
same recipe should apply, though I have only verified 10,1.

doctor, 2026-08-17:
Follow-up to my earlier comment: I've put the Arch packaging for the firmware
and calibration side up at https://github.com/doctor/facetimehd-macos so nobody
has to redo the extraction by hand.

It is a split package that downloads the macOS 10.12.6 update from Apple's CDN
at build time and carves out firmware 5.60.0 plus all 11 sensor set files —
nothing Apple-owned is redistributed, same approach as the existing
`facetimehd-firmware` package uses for 10.11.5.

```
yay -S facetimehd-dkms-git
git clone https://github.com/doctor/facetimehd-macos
cd facetimehd-macos && makepkg -si
```

`facetimehd-dkms-git` is the important part on the driver side: it tracks
master, so it picks up 98b55fd. The released `facetimehd-dkms` (0.7.0.2) does
not.

Its `check()` re-derives the four set files `facetimehd-data` already ships and
asserts they match byte for byte, which is what validates the offsets used for
the other seven.

Fair warning: this is a personal repo, not an AUR package, and it depends on
that Apple CDN URL staying put. It needs ~2 GB of download and ~4.5 GB of build
space. Verified only on a MacBook10,1 so far — if someone with a MacBook8,1 or
9,1 tries it, I'd be glad to hear whether the same set file (1675) and geometry
apply.

patjak, 2026-08-19:
https://github.com/patjak/facetimehd/commit/98b55fd3bd0312d805df941b3905193a0e0fbc0d is now included in the latest tagged release (0.7.0.3) so any non-default resolutions should be supported when using that version.