patjak/facetimehd #179: cannot stat 'System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface': No such file or directory

Alirezaies, 2019-04-10:
after trying to do `make` under the `firmware/` directory I get this error:

```
Checking dependencies for driver download...
/usr/bin/curl
/usr/bin/xzcat
/bin/cpio

Downloading the driver, please wait...
mv: cannot stat 'System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface': No such file or directory
Makefile:34: recipe for target 'AppleCameraInterface' failed
make: *** [AppleCameraInterface] Error 1
```

closed 2019-04-10 by Alirezaies

miguelmota, 2019-10-30:
Same issue, how did you manage to resolve it?

```bash
$ make

Checking dependencies for driver download...
/usr/bin/curl
/usr/bin/xzcat
/bin/cpio

Downloading the driver, please wait...
mv: cannot stat 'System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface': No such file or directory
Makefile:34: recipe for target 'AppleCameraInterface' failed
make: *** [AppleCameraInterface] Error 1
```

miguelmota, 2019-10-30:
@patjak @Alirezaies

tmartiro, 2020-06-25:
Try to change curl arguments in the Makefile 
-       @(curl -s -L -r "$(RANGE)" "$(URL)" | xzcat -q | cpio --format odc -i -d "./$(FILE)") &> /dev/null || true
+       @(curl -k -s -L -r "$(RANGE)" "$(URL)" | xzcat -q | cpio --format odc -i -d "./$(FILE)") &> /dev/null || true