patjak/facetimehd #181: Error Building On Fedora 30

Alirezaies, 2019-05-15:
after extracting the module, on the `make install` in the projects root I get these output:

[root@AlirezaieS bcwc_pcie]# make install
make -C /lib/modules/5.0.14-300.fc30.x86_64/build M=/home/sadegh/devel/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/kernels/5.0.14-300.fc30.x86_64'
  INSTALL /home/sadegh/devel/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.0.14-300.fc30.x86_64
make[1]: Leaving directory '/usr/src/kernels/5.0.14-300.fc30.x86_64'

the `dmesg`  output is:

[12799.233216] facetimehd: version magic '5.0.9-301.fc30.x86_64 SMP mod_unload ' should be '5.0.14-300.fc30.x86_64 SMP mod_unload '
[12853.233796] media: Linux media interface: v0.10
[12853.252010] videodev: Linux video capture interface: v2.00
[12853.263262] facetimehd: version magic '5.0.9-301.fc30.x86_64 SMP mod_unload ' should be '5.0.14-300.fc30.x86_64 SMP mod_unload '
[13072.416813] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[13072.418695] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[13072.423916] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[13072.423949] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[13072.433970] facetimehd 0000:02:00.0: PLL reset finished
[13072.433974] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[13072.433987] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[13072.443995] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[13072.464032] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[13072.464045] facetimehd 0000:02:00.0: STRAP valid
[13072.464046] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[13072.464053] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[13072.464064] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 3 us
[13072.464074] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[13072.464075] facetimehd 0000:02:00.0: Using step size 146
[13072.464079] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x10119
[13072.464083] facetimehd 0000:02:00.0: Virtual VTT enabled
[13072.484604] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[13072.484617] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[13072.484811] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[13072.600852] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[13072.752518] facetimehd 0000:02:00.0: ISP woke up after 0ms
[13072.752523] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[13072.752642] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[13072.762658] facetimehd 0000:02:00.0: ISP second int after 0ms
[13072.762661] facetimehd 0000:02:00.0: Channel description table at 00800000
[13072.772946] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[13072.772947] facetimehd 0000:02:00.0: Enabling interrupts
[13072.773116] FWMSG: 
[13072.773148] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[13072.773157] FWMSG: 
[13072.773171] FWMSG:   Restart count : 0
[13072.773247] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[13072.773264] FWMSG:   System Clock : 200000000 Hz
[13072.773281] FWMSG:   Processor mode : SUPERVISOR
[13072.773301] FWMSG:   Cache architecture type : SEPARATE
[13072.773317] FWMSG:   Cache type : WRITEBACK
[13072.773334] FWMSG:   Data Cache Line : 32 [0x20]
[13072.773359] FWMSG:   Boot arguments entries : 2
[13072.773400] FWMSG:     0000: 0x00000000 0x00000000
[13072.773427] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[13072.773447] FWMSG:   Main memory :   base : 0x00000000
[13072.773504] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[13072.773530] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[13072.773575] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[13072.773599] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[13072.773659] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[13072.773679] FWMSG:   Shared memory : base : 0x00800000
[13072.773742] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[13072.773808] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[13072.773859] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[13072.773881] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[13072.773926] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[13072.773954] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[13072.773972] FWMSG:   Application : adc [release]
[13072.773993] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[13072.774011] FWMSG:   Release : S2ISP-01.43.00
[13072.774028] FWMSG:   H4ISPAPPLE : 11536
[13072.774043] FWMSG:   H4ISPCD : 4081
[13072.774057] FWMSG:   ffw : 4143
[13072.774113] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[13072.774123] FWMSG: 
[13072.774157] FWMSG: -------------------------------------------------------------------------------
[13072.774167] FWMSG: 
[13072.847375] FWMSG: [ISP] CMD = 0x0004 [CISP_CMD_PRINT_ENABLE]
[13072.847538] FWMSG: [ISP] CMD = 0x0003 [CISP_CMD_CONFIG_GET]
[13072.847735] FWMSG: [ISP] CH = 0 CMD = 0x010d [CISP_CMD_CH_INFO_GET]
[13072.847929] facetimehd 0000:02:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2

LyesSaadi, 2019-05-16:
Hi ! I have exactly the same problem !

Alirezaies, 2019-05-16:
some one referenced to #104 which lead me to an kernel upgrade, but back then it was version 4.4 I think, we live in 2019 with version 5.0 kernel :)

---- On Thu, 16 May 2019 21:44:57 +0430 notifications@github.com wrote ----

Hi ! I have exactly the same problem !

evoblue, 2019-05-18:
I have the same issue on fedora 30. I think that the ssl certificate problem has always been there also in previous versions. But it used to work anyway on fedora 27 for example.  I think we have a different problem now that is  more subtle.

patjak, 2019-06-07:
I believe you're missing the certificate for signing the module. Depending on your distro/config you may not be able to load unsigned modules. I would suggest looking into how your distro handles non-signed out of tree kernel modules.

Alirezaies, 2019-06-08:
well, secure boot is disabled on my device and the module could be loaded.

the out put of  `lsmod | grep -i 'facetimehd'` shows:

`➜  ~ lsmod | grep -i facetimehd`
`facetimehd             86016  0`
`videobuf2_dma_sg       16384  1 facetimehd`
`videobuf2_v4l2         28672  1 facetimehd`
`videobuf2_common       57344  2 videobuf2_v4l2,facetimehd`
`videodev              233472  3 videobuf2_v4l2,facetimehd,videobuf2_common`

and for the `dmesg` I still get `Direct firmware load for facetimehd/1871_01XX.dat failed with error -2` which according to the wiki is nothing to be worry about but I sstill have no video

patjak, 2019-06-08:
@Alirezaies 
[12799.233216] facetimehd: version magic '5.0.9-301.fc30.x86_64 SMP mod_unload ' should be '5.0.14-300.fc30.x86_64 SMP mod_unload '

Are you compiling against the wrong headers? Also, which program did you test the camera with? I use qv4l2 for my testing.

damaestro, 2019-06-23:
I don't have this issue, but some applications are unable to use the camera. Chromium and Zoom work fine but Cheese doesn't.

```
$ uname -r
5.1.11-300.fc30.x86_64
$ rpm -qa kernel\*|grep $(uname -r)|sort
kernel-5.1.11-300.fc30.x86_64
kernel-core-5.1.11-300.fc30.x86_64
kernel-devel-5.1.11-300.fc30.x86_64
kernel-modules-5.1.11-300.fc30.x86_64
kernel-modules-extra-5.1.11-300.fc30.x86_64
$ rpm -q chromium zoom cheese
chromium-73.0.3683.86-2.fc30.x86_64
zoom-2.8.252201.0616-1.x86_64
cheese-3.32.1-1.fc30.x86_64
$ lsmod|grep facetimehd
facetimehd             98304  0
videobuf2_dma_sg       16384  1 facetimehd
videobuf2_v4l2         28672  1 facetimehd
videobuf2_common       57344  2 videobuf2_v4l2,facetimehd
videodev              233472  3 videobuf2_v4l2,facetimehd,videobuf2_common
```
This has been the case since a kernel update in f27+. I don't recall when it stopped exactly.

damaestro, 2019-06-23:
Please check that you followed the wiki. Sometimes I forget to download the firmware.

https://github.com/patjak/bcwc_pcie/wiki/Get-Started#get-started-on-fedora
https://github.com/patjak/bcwc_pcie/wiki/Get-Started#firmware-extraction

damaestro, 2019-06-23:
Latest kernel is working fine too:
```
# make clean
make -C /lib/modules/5.1.12-300.fc30.x86_64/build M=/root/bcwc_pcie clean
make[1]: Entering directory '/usr/src/kernels/5.1.12-300.fc30.x86_64'
  CLEAN   /root/bcwc_pcie/.tmp_versions
  CLEAN   /root/bcwc_pcie/Module.symvers
make[1]: Leaving directory '/usr/src/kernels/5.1.12-300.fc30.x86_64'
# make
make -C /lib/modules/5.1.12-300.fc30.x86_64/build M=/root/bcwc_pcie modules
make[1]: Entering directory '/usr/src/kernels/5.1.12-300.fc30.x86_64'
  CC [M]  /root/bcwc_pcie/fthd_ddr.o
  CC [M]  /root/bcwc_pcie/fthd_hw.o
  CC [M]  /root/bcwc_pcie/fthd_drv.o
  CC [M]  /root/bcwc_pcie/fthd_ringbuf.o
  CC [M]  /root/bcwc_pcie/fthd_isp.o
  CC [M]  /root/bcwc_pcie/fthd_v4l2.o
  CC [M]  /root/bcwc_pcie/fthd_buffer.o
  CC [M]  /root/bcwc_pcie/fthd_debugfs.o
  LD [M]  /root/bcwc_pcie/facetimehd.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /root/bcwc_pcie/facetimehd.mod.o
  LD [M]  /root/bcwc_pcie/facetimehd.ko
make[1]: Leaving directory '/usr/src/kernels/5.1.12-300.fc30.x86_64'
# make install
make -C /lib/modules/5.1.12-300.fc30.x86_64/build M=/root/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/kernels/5.1.12-300.fc30.x86_64'
  INSTALL /root/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.1.12-300.fc30.x86_64
make[1]: Leaving directory '/usr/src/kernels/5.1.12-300.fc30.x86_64'
# uname -r
5.1.12-300.fc30.x86_64
# rpm -qa kernel\*|grep $(uname -r)|sort
kernel-5.1.12-300.fc30.x86_64
kernel-core-5.1.12-300.fc30.x86_64
kernel-devel-5.1.12-300.fc30.x86_64
kernel-headers-5.1.12-300.fc30.x86_64
kernel-modules-5.1.12-300.fc30.x86_64
kernel-modules-extra-5.1.12-300.fc30.x86_64
# modprobe facetimehd
# lsmod|grep facetimehd
facetimehd             98304  0
videobuf2_dma_sg       16384  1 facetimehd
videobuf2_v4l2         28672  1 facetimehd
videobuf2_common       57344  2 videobuf2_v4l2,facetimehd
videodev              233472  3 videobuf2_v4l2,facetimehd,videobuf2_common
```

kernel-headers got installed for something else I'm working on and does not seem to be required for building this driver.

LyesSaadi, 2019-06-23:
Hi, something weird happened. I reinstalled the module, still had the error. Then, I tested it with Cheese and Skype and it didn't worked either. But I followed @damaestro 's advice... and tested it with Firefox, and it worked ¯\\_(ツ)_/¯, who knows, maybe it was working from the beginning...

patjak, 2019-12-09:
Skype is known to have issues because we support features it doesn't understand. Cheese should be fixed but if you're running an old version it might not have the proper fixes. Cheese 3.26.0 works fine here. Chrome also have a fix for this but older version might not. The best way to test that the driver is working is to use qv4l2.

evoblue, 2019-12-09:
Is there any hope for a skype support in the future  ?
Anyway I confirm that it works with firefox on fedora 30 and also linphone.

patjak, 2019-12-09:
@evoblue yes I plan to remove/hide the features that Skype and some browsers doesn't support (which causes problems). The correct way of doing it would be to let Skype and others to fix their stuff but I don't see that happening anytime soon :(

evoblue, 2019-12-09:
I know that the ideal way to deal with this problem is another one, but
your plan to fix this issue on "our" side is anyway a good news for the
community! Thanks for your support!

Caboose30, 2020-02-16:
I ran across the same error. According to modprobe the driver is loaded, however nothing works - not firefox, Cheese, discord, or mplayer. Am I missing something?
Also I am terrible at anything CLI. GUIs are my friends.