patjak/facetimehd #186: Ubuntu 19.10 not working

mcamou, 2019-07-31:
I recently upgraded to Ubuntu 19.10 and the facetimehd driver does not compile:

```
$ lsb_release -a
LSB Version:    core-10.2019031300ubuntu1-noarch:printing-10.2019031300ubuntu1-noarch:security-10.2019031300ubuntu1-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 19.04
Release:        19.04
Codename:       disco

mario at tumbolia in ~/src/macbuntu/facetimehd (master●)
$ uname -a
Linux tumbolia 5.0.0-21-generic #22-Ubuntu SMP Tue Jul 2 13:27:33 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

$ make
make -C /lib/modules/5.0.0-21-generic/build M=/home/mario/src/macbuntu/facetimehd modules
make[1]: Entering directory '/usr/src/linux-headers-5.0.0-21-generic'
  CC [M]  /home/mario/src/macbuntu/facetimehd/fthd_ddr.o
In file included from /home/mario/src/macbuntu/facetimehd/fthd_ddr.c:21:
/home/mario/src/macbuntu/facetimehd/fthd_drv.h:121:5: warning: "LINUX_VERSION_CODE" is not defined, evaluates to 0 [-Wundef]
 #if LINUX_VERSION_CODE < KERNEL_VERSION(4,8,0)
     ^~~~~~~~~~~~~~~~~~
/home/mario/src/macbuntu/facetimehd/fthd_drv.h:121:26: warning: "KERNEL_VERSION" is not defined, evaluates to 0 [-Wundef]
 #if LINUX_VERSION_CODE < KERNEL_VERSION(4,8,0)
                          ^~~~~~~~~~~~~~
/home/mario/src/macbuntu/facetimehd/fthd_drv.h:121:40: error: missing binary operator before token "("
 #if LINUX_VERSION_CODE < KERNEL_VERSION(4,8,0)
                                        ^
make[2]: *** [scripts/Makefile.build:286: /home/mario/src/macbuntu/facetimehd/fthd_ddr.o] Error 1
make[1]: *** [Makefile:1606: _module_/home/mario/src/macbuntu/facetimehd] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-21-generic'
make: *** [Makefile:12: all] Error 2
done
```

wvengen, 2019-10-08:
That is really odd, as `linux/version.h` [is included](https://github.com/patjak/bcwc_pcie/blob/master/fthd_drv.h#L27). I can only guess that the kernel headers are not fully installed or configured.

patjak, 2019-12-09:
Should be fixed now in the master branch

closed 2019-12-09 by patjak