patjak/facetimehd #191: Missing osxupd10.11.3.dmg

gdotdesign, 2019-10-29:
I'm trying to run it on Macbook Pro with Ubuntu 18.04.

The file it tries to download no is longer available https://support.apple.com/downloads/DL1858/en_US/osxupd10.11.3.dmg  so the `make` fails.

Is there way to get it from somewhere else? It worked like two weeks ago.

offlinehacker, 2019-11-05:
See https://github.com/patjak/facetimehd-firmware/issues/1

patjak, 2019-11-05:
@offlinehacker 's fix is now available in https://github.com/patjak/facetimehd-firmware

closed 2019-11-05 by patjak

gdotdesign, 2019-11-05:
Sorry I don't know how to apply that, I run the script and it downloaded the firmware what should I do after that?

patjak, 2019-11-05:
@gdotdesign I've updated the wiki, see: https://github.com/patjak/bcwc_pcie/wiki/Get-Started#firmware-extraction

gdotdesign, 2019-11-07:
Thanks, I can download the firmware without problems.

After that I tried to follow the instructions for Ubuntu, the firmware part `bcwc_pcie/firmware` still does not work (I guess it's not even needed now?) then the `make` works:
```
make -C /lib/modules/5.0.0-32-generic/build M=/home/gdot/Lib/bcwc_pcie modules
make[1]: Entering directory '/usr/src/linux-headers-5.0.0-32-generic'
  CC [M]  /home/gdot/Lib/bcwc_pcie/fthd_ddr.o
  CC [M]  /home/gdot/Lib/bcwc_pcie/fthd_hw.o
  CC [M]  /home/gdot/Lib/bcwc_pcie/fthd_drv.o
  CC [M]  /home/gdot/Lib/bcwc_pcie/fthd_ringbuf.o
  CC [M]  /home/gdot/Lib/bcwc_pcie/fthd_isp.o
  CC [M]  /home/gdot/Lib/bcwc_pcie/fthd_v4l2.o
  CC [M]  /home/gdot/Lib/bcwc_pcie/fthd_buffer.o
  CC [M]  /home/gdot/Lib/bcwc_pcie/fthd_debugfs.o
  LD [M]  /home/gdot/Lib/bcwc_pcie/facetimehd.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/gdot/Lib/bcwc_pcie/facetimehd.mod.o
  LD [M]  /home/gdot/Lib/bcwc_pcie/facetimehd.ko
make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-32-generic'
```
Then the `sudo make install` fails:
```
make -C /lib/modules/5.0.0-32-generic/build M=/home/gdot/Lib/bcwc_pcie modules_install
make[1]: Entering directory '/usr/src/linux-headers-5.0.0-32-generic'
  INSTALL /home/gdot/Lib/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.0.0-32-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Leaving directory '/usr/src/linux-headers-5.0.0-32-generic'
```

The funny thing is that it has worked once before on probably a different kernel version, so maybe that messes up something, but I still don't see the connection with the firmware in other repo.

Sorry about being difficult, I'll probably give up soon and use an external webcam.