patjak/facetimehd #192: Corrupt Output: Everything Is Green

ty3sc, 2019-11-26:
Hi, I think I've followed the instructions properly but whenever I run `mplayer tv://` I get a corrupt video. Everything is green! Has anyone bumped into this before?

```
foo@foo-macbook-pro:~$ mplayer tv://
MPlayer 1.3.0 (Debian), built with gcc-7 (C) 2000-2016 MPlayer Team
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing tv://.
TV file format detected.
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: your device driver does not support VIDIOC_G_STD ioctl, VIDIOC_G_PARM was used instead.
Selected device: Apple Facetime HD
 Capabilities:  video capture  read/write  streaming
 supported norms:
 inputs: 0 = Camera;
 Current input: 0
 Current format: YUYV
tv.c: norm_from_string(pal): Bogus norm parameter, setting default.
v4l2: ioctl enum norm failed: Inappropriate ioctl for device
Error: Cannot set norm!
Selected input hasn't got a tuner!
v4l2: ioctl set mute failed: Invalid argument
Failed to open VDPAU backend libvdpau_i965.so: cannot open shared object file: No such file or directory
[vdpau] Error when calling vdp_device_create_x11: 1
==========================================================================
Opening video decoder: [raw] RAW Uncompressed Video
Could not find matching colorspace - retrying with -vf scale...
Opening video filter: [scale]
Movie-Aspect is undefined - no prescaling applied.
[swscaler @ 0x5594ccf6c5e0] bicubic scaler, from yuyv422 to yuv420p using MMXEXT
[swscaler @ 0x5594ccf6c5e0] using unscaled yuyv422 -> yuv420p special converter
VO: [xv] 640x480 => 640x480 Planar YV12 
Selected video codec: [rawyuy2] vfm: raw (RAW YUY2)
==========================================================================
Audio: no sound
Starting playback...
V:   0.0   2/  2 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0   4/  4 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0   6/  6 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0   8/  8 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0  10/ 10 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0  12/ 12 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0  14/ 14 ??% ??% ??,?% 0 0 
v4l2: select timeout

MPlayer interrupted by signal 2 in module: video_read_frame
V:   0.0  15/ 15 ??% ??% ??,?% 0 0 
v4l2: select timeout
v4l2: ioctl set mute failed: Invalid argument
v4l2: 0 frames successfully processed, 1 frames dropped.

Exiting... (Quit)
```

[image]

```
foo@foo-macbook-pro:~$ sudo dmidecode | grep -A3 "^System Information"
System Information
	Manufacturer: Apple Inc.
	Product Name: MacBookPro11,4
	Version: 1.0
```

```
foo@foo-macbook-pro:~$ uname -a
Linux foo-macbook-pro 4.15.0-70-generic #79-Ubuntu SMP Tue Nov 12 10:36:11 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

```

ty3sc, 2019-11-26:
I'm pretty sure this is my MacBook Pro if that helps.

[image]

ty3sc, 2019-11-26:
I should also add, the green LED next to my webcam does turn on. The "Cheese" program can also see the webcam under "Preferences > Device". I'm not sure if that's relevant.
[image]

ty3sc, 2019-11-27:
I turned on my laptop this morning and my webcam is mysteriously working now... I cannot explain this.

Thank you for this repo though. It works! :)

ty3sc, 2019-12-10:
Interestingly, this happened again to me yesterday. I upgraded the kernel and had to reinstall this driver. Again, I got green output after completing the process, but after leaving my laptop overnight the webcam is working perfectly.

Could this have anything to do with the PRAM or NVRAM or SMC expiring on my MacBook Pro overnight?

ty3sc, 2019-12-10:
Sorry, I can't seem to run that command on my machine. Am I doing something wrong?

```
foo@foo-macbook-pro:~$ sudo modprobe dyndbg==pmf
modprobe: FATAL: Module dyndbg==pmf not found in directory /lib/modules/4.15.0-72-generic
```

patjak, 2019-12-10:
oh sorry, I meant
```modprobe facetimehd dyndb==pmf```
If the module is already loaded you also first need to run
```modprobe -r facetimehd```

ty3sc, 2019-12-10:
After running those commands I'm now getting the green screen again. Here's the output of my console:

```
foo@foo-macbook-pro:~$ sudo modprobe -r facetimehd
[sudo] password for foo: 
foo@foo-macbook-pro:~$ sudo modprobe facetimehd dyndb==pmf
foo@foo-macbook-pro:~$ mplayer tv://
MPlayer 1.3.0 (Debian), built with gcc-7 (C) 2000-2016 MPlayer Team
do_connect: could not connect to socket
connect: No such file or directory
Failed to open LIRC support. You will not be able to use your remote control.

Playing tv://.
TV file format detected.
Selected driver: v4l2
 name: Video 4 Linux 2 input
 author: Martin Olschewski <olschewski@zpr.uni-koeln.de>
 comment: first try, more to come ;-)
v4l2: your device driver does not support VIDIOC_G_STD ioctl, VIDIOC_G_PARM was used instead.
Selected device: Apple Facetime HD
 Capabilities:  video capture  read/write  streaming
 supported norms:
 inputs: 0 = Camera;
 Current input: 0
 Current format: YUYV
tv.c: norm_from_string(pal): Bogus norm parameter, setting default.
v4l2: ioctl enum norm failed: Inappropriate ioctl for device
Error: Cannot set norm!
Selected input hasn't got a tuner!
v4l2: ioctl set mute failed: Invalid argument
Failed to open VDPAU backend libvdpau_i965.so: cannot open shared object file: No such file or directory
[vdpau] Error when calling vdp_device_create_x11: 1
==========================================================================
Opening video decoder: [raw] RAW Uncompressed Video
Could not find matching colorspace - retrying with -vf scale...
Opening video filter: [scale]
Movie-Aspect is undefined - no prescaling applied.
[swscaler @ 0x55bace6155e0] bicubic scaler, from yuyv422 to yuv420p using MMXEXT
[swscaler @ 0x55bace6155e0] using unscaled yuyv422 -> yuv420p special converter
VO: [xv] 640x480 => 640x480 Planar YV12 
Selected video codec: [rawyuy2] vfm: raw (RAW YUY2)
==========================================================================
Audio: no sound
Starting playback...
V:   0.0   2/  2 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0   4/  4 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0   6/  6 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0   8/  8 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0  10/ 10 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0  12/ 12 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0  14/ 14 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0  16/ 16 ??% ??% ??,?% 0 0 
v4l2: select timeout
V:   0.0  18/ 18 ??% ??% ??,?% 0 0 
v4l2: select timeout
v4l2: ioctl set mute failed: Invalid argument
v4l2: 0 frames successfully processed, 1 frames dropped.

Exiting... (Quit)
foo@foo-macbook-pro:~$ 
```

patjak, 2019-12-10:
What is the output from running dmesg?

ty3sc, 2019-12-10:
Sorry, here it is:

```
[20300.425443] device vethc257978 entered promiscuous mode
[20300.426313] IPv6: ADDRCONF(NETDEV_UP): vethc257978: link is not ready
[20301.080751] eth0: renamed from veth40c8778
[20301.116757] IPv6: ADDRCONF(NETDEV_CHANGE): vethc257978: link becomes ready
[20301.116842] br-3ab91b5d0470: port 6(vethc257978) entered blocking state
[20301.116846] br-3ab91b5d0470: port 6(vethc257978) entered forwarding state
[20317.471091] br-3ab91b5d0470: port 6(vethc257978) entered disabled state
[20317.471188] veth40c8778: renamed from eth0
[20317.560086] br-3ab91b5d0470: port 6(vethc257978) entered disabled state
[20317.561836] device vethc257978 left promiscuous mode
[20317.561839] br-3ab91b5d0470: port 6(vethc257978) entered disabled state
[20347.310677] br-3ab91b5d0470: port 6(veth497e676) entered blocking state
[20347.310680] br-3ab91b5d0470: port 6(veth497e676) entered disabled state
[20347.310769] device veth497e676 entered promiscuous mode
[20347.310924] IPv6: ADDRCONF(NETDEV_UP): veth497e676: link is not ready
[20347.977843] eth0: renamed from veth73a8290
[20348.022091] IPv6: ADDRCONF(NETDEV_CHANGE): veth497e676: link becomes ready
[20348.022193] br-3ab91b5d0470: port 6(veth497e676) entered blocking state
[20348.022198] br-3ab91b5d0470: port 6(veth497e676) entered forwarding state
[20364.638378] br-3ab91b5d0470: port 6(veth497e676) entered disabled state
[20364.638463] veth73a8290: renamed from eth0
[20364.720921] br-3ab91b5d0470: port 6(veth497e676) entered disabled state
[20364.722680] device veth497e676 left promiscuous mode
[20364.722683] br-3ab91b5d0470: port 6(veth497e676) entered disabled state
[20455.148700] FWMSG: [ISP] CMD = 0x0001 [CISP_CMD_STOP]
[20455.159719] facetimehd 0000:04:00.0: fthd_hw_deinit
[20458.597707] media: Linux media interface: v0.10
[20458.604082] Linux video capture interface: v2.00
[20458.611344] facetimehd: unknown parameter 'dyndb' ignored
[20458.611402] facetimehd 0000:04:00.0: Found FaceTime HD camera with device id: 1570
[20458.611638] facetimehd 0000:04:00.0: Setting 64bit DMA mask
[20458.616673] facetimehd 0000:04:00.0: S2 PCIe link init succeeded
[20458.616701] facetimehd 0000:04:00.0: Refclk: 25MHz (0xa)
[20458.626633] facetimehd 0000:04:00.0: PLL reset finished
[20458.626635] facetimehd 0000:04:00.0: Waiting for S2 PLL to lock at 450 MHz
[20458.626647] facetimehd 0000:04:00.0: S2 PLL is locked after 10 us
[20458.636573] facetimehd 0000:04:00.0: S2 PLL is in bypass mode
[20458.656459] facetimehd 0000:04:00.0: DDR40 PHY PLL locked on safe settings
[20458.656472] facetimehd 0000:04:00.0: STRAP valid
[20458.656473] facetimehd 0000:04:00.0: Configuring DDR PLLs for 450 MHz
[20458.656480] facetimehd 0000:04:00.0: DDR40 PLL is locked after 0 us
[20458.656490] facetimehd 0000:04:00.0: First DDR40 VDL calibration completed after 2 us
[20458.656499] facetimehd 0000:04:00.0: Second DDR40 VDL calibration completed after 1 us
[20458.656500] facetimehd 0000:04:00.0: Using step size 150
[20458.656505] facetimehd 0000:04:00.0: VDL set to: coarse=0x10008, fine=0x1011a
[20458.656509] facetimehd 0000:04:00.0: Virtual VTT enabled
[20458.676865] facetimehd 0000:04:00.0: S2 DRAM memory address: 0x22159559
[20458.676889] facetimehd 0000:04:00.0: Rewrite DDR mode registers succeeded
[20458.677137] facetimehd 0000:04:00.0: Full memory verification succeeded! (0)
[20459.138577] facetimehd 0000:04:00.0: Loaded firmware, size: 1392kb
[20459.284709] facetimehd 0000:04:00.0: ISP woke up after 0ms
[20459.284714] facetimehd 0000:04:00.0: Number of IPC channels: 7, queue size: 44865
[20459.284717] facetimehd 0000:04:00.0: Firmware requested heap size: 3072kb
[20459.294654] facetimehd 0000:04:00.0: ISP second int after 0ms
[20459.294656] facetimehd 0000:04:00.0: Channel description table at 00800000
[20459.304753] facetimehd 0000:04:00.0: magic value: 00000000 after 0 ms
[20459.304754] facetimehd 0000:04:00.0: Enabling interrupts
[20459.304920] FWMSG: 
[20459.304942] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[20459.304953] FWMSG: 
[20459.304965] FWMSG:   Restart count : 0
[20459.305054] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[20459.305068] FWMSG:   System Clock : 200000000 Hz
[20459.305082] FWMSG:   Processor mode : SUPERVISOR
[20459.305097] FWMSG:   Cache architecture type : SEPARATE
[20459.305111] FWMSG:   Cache type : WRITEBACK
[20459.305124] FWMSG:   Data Cache Line : 32 [0x20]
[... 245 lines cut ...]
[20466.821515] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4FSM.cpp, 449: FlowIC00: MIPI error (interrupt source = 0x8)
[20466.821545] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 2547: FlowIC00: Perform error recovery (89).
[20466.873510] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4FSM.cpp, 449: FlowIC00: MIPI error (interrupt source = 0x8)
[20466.873540] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 2547: FlowIC00: Perform error recovery (90).
[20466.925460] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4FSM.cpp, 449: FlowIC00: MIPI error (interrupt source = 0x8)
[20466.925498] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 2547: FlowIC00: Perform error recovery (91).
[20466.977394] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4FSM.cpp, 449: FlowIC00: MIPI error (interrupt source = 0x8)
[20466.977421] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 2547: FlowIC00: Perform error recovery (92).
[... cut ...]
[20467.228364] FWMSG: [ISP] CH = 0 CMD = 0x0104 [CISP_CMD_CH_BUFFER_RETURN]
[20467.229033] FWMSG: [ISP] CH = 0 CMD = 0x0d01 [CISP_CMD_CH_FACE_DETECTION_STOP]
[20467.229248] FWMSG: [ISP] CH = 0 CMD = 0x0d04 [CISP_CMD_CH_FACE_DETECTION_DISABLE]
[20467.229472] FWMSG: [ISP] CH = 0 CMD = 0xc114 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_DISABLE]
[20467.229646] FWMSG: [ISP] CH = 0 CMD = 0xc103 [CISP_CMD_APPLE_CH_MOTION_HISTORY_STOP]
[20467.229804] FWMSG: [ISP] CH = 0 CMD = 0xc101 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_STOP]
foo@foo-macbook-pro:~$ 
```

patjak, 2019-12-10:
Ok so we shouldn't be seeing:
```
[20467.185446] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4FSM.cpp, 449: FlowIC00: MIPI error (interrupt source = 0x8)
[20467.185477] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 2547: FlowIC00: Perform error recovery (96).
```
But I don't know why this is happening. Can you also try qv4l2 to make sure it's not a problem with mplayer.

ty3sc, 2019-12-10:
It's suddenly working again. I cannot explain why this is!

Thank you for your help and thank you for authoring these drivers.

I will try and keep a log of what triggers this problem. If I collate anything useful I will reopen this.

closed 2019-12-10 by ty3sc

patjak, 2019-12-10:
Yes, try to figure out if anything in particular is triggering this. Also, make sure you're running the latest version of the driver and perhaps try reinstalling the firmware to make sure it's not corrupted. qv4l2 is the best way to test since it supports features that other webcam programs do not. Thanks

mark-piper, 2024-11-12:
I also had this issue, with everything showing up pink and green on a Macbook Air 6,2.  It started occurring after some system updates, but I'm unsure exactly why.

The root cause is the pixel format being negotiated as YVYU when it needs to be YUYV instead. I was able to verify this by using `qv4l2` to try both formats.

Recompiling/reinstalling the driver after commenting out YVYU support in `fthd_v4l2_ioctl_querycap` fixed the issue for me.

In https://github.com/patjak/facetimehd/blob/master/fthd_v4l2.c#L364 I changed it to this:

```
static int fthd_v4l2_ioctl_enum_fmt_vid_cap(struct file *filp, void *priv,
				   struct v4l2_fmtdesc *fmt)
{
	char *desc = NULL;

	switch (fmt->index) {
	case 0:
		fmt->pixelformat = V4L2_PIX_FMT_YUYV;
		desc = "YUYV";
		break;
	/* *** FIX: DISABLING YVYU FOR MACBOOK AIR 6,2 ***
        case 1:
		fmt->pixelformat = V4L2_PIX_FMT_YVYU;
		desc = "YVYU";
		break;
	*/
	/* We don't support the mplane yet
	case 2:
		fmt->pixelformat = V4L2_PIX_FMT_NV16;
		desc = "NV16";
		break;
	*/
	default:
		return -EINVAL;
	}
```