patjak/facetimehd #193: Failed to build on Archlinux lts kernel

adbjesus, 2019-12-09:
Hi,

I get an error when trying to build the latest [commit](https://github.com/patjak/bcwc_pcie/tree/9d7ff828ebe9c455108e6039b45cd06c693597a2) from the `mainline` branch against the latest `linux-lts` package from archlinux (version `4.19.88-1-lts`). This builds fine on the latest `linux` package (version `5.4.2-arch1-1`). Logs follow:

```
$ sudo dkms install bcwc-pcie/r259.9d7ff82 -k 4.19.88-1-lts

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
make -j4 KERNELRELEASE=4.19.88-1-lts -C /usr/lib/modules/4.19.88-1-lts/build M=/var/lib/dkms/bcwc-pcie/r259.9d7ff82/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.88-1-lts (x86_64)
Consult /var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/make.log for more information.
```

```
$ cat /var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/make.log
DKMS make.log for bcwc-pcie-r259.9d7ff82 for kernel 4.19.88-1-lts (x86_64)
Mon 09 Dec 2019 11:01:40 AM CET
make: Entering directory '/usr/lib/modules/4.19.88-1-lts/build'
  CC [M]  /var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_drv.o
  CC [M]  /var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_hw.o
  CC [M]  /var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_ringbuf.o
  CC [M]  /var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_isp.o
/var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_drv.c: In function ‘fthd_pci_init’:
/var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_drv.c:379:2: error: implicit declaration of function ‘pci_disable_link_state’; did you mean ‘pci_disable_ats’? [-Werror=implicit-function-declaration]
  379 |  pci_disable_link_state(pdev, PCIE_LINK_STATE_L0S | PCIE_LINK_STATE_L1 |
      |  ^~~~~~~~~~~~~~~~~~~~~~
      |  pci_disable_ats
/var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_drv.c:379:31: error: ‘PCIE_LINK_STATE_L0S’ undeclared (first use in this function)
  379 |  pci_disable_link_state(pdev, PCIE_LINK_STATE_L0S | PCIE_LINK_STATE_L1 |
      |                               ^~~~~~~~~~~~~~~~~~~
/var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_drv.c:379:31: note: each undeclared identifier is reported only once for each function it appears in
  CC [M]  /var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_v4l2.o
/var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_drv.c:379:53: error: ‘PCIE_LINK_STATE_L1’ undeclared (first use in this function)
  379 |  pci_disable_link_state(pdev, PCIE_LINK_STATE_L0S | PCIE_LINK_STATE_L1 |
      |                                                     ^~~~~~~~~~~~~~~~~~
/var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_drv.c:380:11: error: ‘PCIE_LINK_STATE_CLKPM’ undeclared (first use in this function)
  380 |           PCIE_LINK_STATE_CLKPM);
      |           ^~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:303: /var/lib/dkms/bcwc-pcie/r259.9d7ff82/build/fthd_drv.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:1525: _module_/var/lib/dkms/bcwc-pcie/r259.9d7ff82/build] Error 2
make: Leaving directory '/usr/lib/modules/4.19.88-1-lts/build'
```

patjak, 2019-12-09:
Hi,
For older kernels please use the master branch instead of the mainline branch.

closed 2019-12-09 by patjak

adbjesus, 2019-12-09:
@patjak okay, a note in the wiki/readme regarding this could be useful, at least I didn't find that information anywhere. Moreover, in the wiki here it describes the use of the `bcwc-pcie-git` aur package on archlinux, which will not work for the `linux-lts` kernel so a note about that would also be useful.

patjak, 2019-12-09:
Thanks for testing. Actually you can use the master branch for all kernels. I have my upstreaming work in the mainline branch so it's considered a development branch. Better to always use the master branch.

The wiki is writable for everyone so please update with any improvements / corrections you think is necessary. I don't keep very good track of what's in there. It's mostly for users to help other users.