patjak/facetimehd #195: Error installing on Macbook 10.1 running Ubuntu 18.04

marcusenglert, 2019-12-28:
Hi,
I try to get facetimehd running on my macbook10.1 running Ubuntu 18.04 LTS.
Pls. see the output of sudo make install:
make -C /lib/modules/5.0.0-37-generic/build M=/home/marcus/bcwc_pcie modules_install
make[1]: Verzeichnis „/usr/src/linux-headers-5.0.0-37-generic“ wird betreten
  INSTALL /home/marcus/bcwc_pcie/facetimehd.ko
At main.c:160:
- SSL error:02001002:system library:fopen:No such file or directory: ../crypto/bio/bss_file.c:72
- SSL error:2006D080:BIO routines:BIO_new_file:no such file: ../crypto/bio/bss_file.c:79
sign-file: certs/signing_key.pem: No such file or directory
  DEPMOD  5.0.0-37-generic
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[1]: Verzeichnis „/usr/src/linux-headers-5.0.0-37-generic“ wird verlassen

Thanks for your help!

Best

Marcus

leifliddy, 2019-12-29:
It's so much easier to just use ```dkms``` to install a module.
Unfortunately, the instructions listed at:
https://github.com/patjak/bcwc_pcie/wiki/Get-Started
...are a mess

you just need to make sure you have the dkms package installed
**ubuntu**
```apt-get install dkms```
**fedora**
```dnf install dkms```

then run:
```
git clone https://github.com/patjak/bcwc_pcie.git /usr/src/facetimehd-0.1
dkms install -m facetimehd -v 0.1
```
...that's it. Let me know if you run into any issues. 
On another note, looking the issues page, it doesn't appear that anyone's managed to get this driver working on the macbook12 inch models {8,1 + 9,1 + 10,1}

leifliddy, 2019-12-29:
In any case, that error is harmless. It's trying to sign the module, but you don't have a key setup to sign it with.
You could get rid of that error by not using ```modules_install``` in your ```Makefile```
ie
you change this:
```
install:
	$(MAKE) -C $(KDIR) M=$(PWD) modules_install
```
to
```
install:
	cp facetimehd.ko /lib/modules/$(shell uname -r)/extra
	depmod -a
```
..you'd obviously need to verify that the ```/lib/modules/$(shell uname -r)/extra``` exists. 
..also ensure that the lines following ```install``` are indented using tab

mingen-pan, 2020-04-01:
> It's so much easier to just use `dkms` to build + install a module.
> Unfortunately, the dkms install instructions listed at:
> https://github.com/patjak/bcwc_pcie/wiki/Get-Started
> ...are a mess
> 
> you just need to make sure you have the dkms package installed
> **ubuntu**
> `apt-get install dkms`
> 
> **fedora**
> `dnf install dkms`
> 
> then run:
> 
> ```
> git clone https://github.com/patjak/bcwc_pcie.git /usr/src/facetimehd-0.1
> dkms install -m facetimehd -v 0.1
> ```
> 
> ...that's it. You will however, have to install the firmware as a separate step:
> https://github.com/patjak/facetimehd-firmware
> Let me know if you run into any issues.
> 
> On another note, looking the issues page, it doesn't appear that anyone's managed to get this driver working on the macbook 12 inch models {8,1 + 9,1 + 10,1}

Hello, I am new to this. How could I check if the driver is installed successfully.

earboxer, 2020-04-12:
I recently used these instructions. (in ubuntu studio 19.10)

The big thing I noticed after doing it is that

```
sudo modprobe -r bdc_pci
sudo modprobe facetimehd
```

The second command no longer gave an error (it did before it was installed)

victorleungtw, 2020-04-24:
I tried the solution by @leifliddy and it works. Thanks a lot