patjak/facetimehd #196: macbook 12-inch driver issue

leifliddy, 2019-12-30:
This driver still doesn't work on the macbook 12 inch models: {8,1 + 9,1 + 10,1}
I've read a number of issues where people have touched on this topic, but no real effort was made to diagnose issue. 

**I'm running Fedora 30 w/ kernel 5.3 on a macbook 9,1 (2016 model)

Some people have suggested that the firmware might be different for this type of Macbook (due to its 480p camera)

Sooo...I decided to extract the firmware myself (unfortunately the same behavior was observed, but I'll walk you through the steps anyway)

First, I copied over the ```AppleCameraInterface``` file from macOS to Linux
**copied from
```
/System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface
```
then I extracted firmware.bin, and copied it to ```/lib/firmware/facetimehd```

```
[root@mac tmp]$ binwalk -e AppleCameraInterface 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
81920         0x14000         gzip compressed data, maximum compression, from Unix, NULL date (1970-01-01 00:00:00)
748290        0xB6B02         Certificate in DER format (x509 v3), header length: 4, sequence length: 1038
749332        0xB6F14         Certificate in DER format (x509 v3), header length: 4, sequence length: 1211
750007        0xB71B7         Certificate in DER format (x509 v3), header length: 4, sequence length: 260
750547        0xB73D3         Certificate in DER format (x509 v3), header length: 4, sequence length: 1339
751227        0xB767B         Certificate in DER format (x509 v3), header length: 4, sequence length: 271
752160        0xB7A20         XML document, version: "1.0"

[root@mac _AppleCameraInterface.extracted]$ cp _AppleCameraInterface.extracted/14000 /lib/firmware/facetimehd/firmware.bin
```

Then I ran ```modprobe facetimehd dyndbg==pmf```
```
[Mon Dec 30 03:27:58 2019] mc: Linux media interface: v0.10
[Mon Dec 30 03:27:58 2019] videodev: Linux video capture interface: v2.00
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Found FaceTime HD camera with device id: 1570
[Mon Dec 30 03:27:58 2019] facetimehd:fthd_pci_reserve_mem: Allocated S2 regs (BAR 0). 65536 bytes at 0x00000000b56696f1
[Mon Dec 30 03:27:58 2019] facetimehd:fthd_pci_reserve_mem: Allocated S2 mem (BAR 2). 268435456 bytes at 0x0000000002affd27
[Mon Dec 30 03:27:58 2019] facetimehd:fthd_pci_reserve_mem: Allocated ISP regs (BAR 4). 1048576 bytes at 0x000000002df644cb
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Setting 64bit DMA mask
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: S2 PCIe link init succeeded
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Refclk: 24MHz (0x2
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: PLL reset finished
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Waiting for S2 PLL to lock at 450 MHz
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: S2 PLL is locked after 10 us
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: S2 PLL is in bypass mode
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: DDR40 PHY PLL locked on safe settings
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: STRAP valid
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Configuring DDR PLLs for 450 MHz
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: DDR40 PLL is locked after 0 us
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: First DDR40 VDL calibration completed after 2 us
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Second DDR40 VDL calibration completed after 2 us
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Using step size 156
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: VDL set to: coarse=0x10008, fine=0x1011b
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Virtual VTT enabled
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: S2 DRAM memory address: 0x22159559
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Rewrite DDR mode registers succeeded
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Full memory verification succeeded! (0)
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Loaded firmware, size: 1384kb
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: ISP woke up after 0ms
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Number of IPC channels: 7, queue size: 44865
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Firmware requested heap size: 3072kb
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: ISP second int after 0ms
[Mon Dec 30 03:27:58 2019] facetimehd 0000:03:00.0: Channel description table at 00800000
[Mon Dec 30 03:27:58 2019] facetimehd:isp_fill_channel_info: Channel 0: TERMINAL, type 2, source 0, size 512, offset 800700
[Mon Dec 30 03:27:58 2019] facetimehd:isp_fill_channel_info: Channel 1: IO, type 0, source 1, size 8, offset 808700
[Mon Dec 30 03:27:58 2019] facetimehd:isp_fill_channel_info: Channel 2: DEBUG, type 0, source 1, size 8, offset 808900
[Mon Dec 30 03:27:58 2019] facetimehd:isp_fill_channel_info: Channel 3: BUF_H2T, type 0, source 2, size 64, offset 808b00
[Mon Dec 30 03:27:58 2019] facetimehd:isp_fill_channel_info: Channel 4: BUF_T2H, type 1, source 3, size 64, offset 809b00
[... 301 lines cut ...]
[Mon Dec 30 03:36:05 2019] facetimehd:fthd_v4l2_ioctl_try_fmt_vid_cap: fthd_v4l2_ioctl_try_fmt_vid_cap: 640x480
[Mon Dec 30 03:36:05 2019] facetimehd:fthd_v4l2_ioctl_try_fmt_vid_cap: fthd_v4l2_ioctl_try_fmt_vid_cap: 640x480
[Mon Dec 30 03:36:05 2019] facetimehd:fthd_v4l2_ioctl_try_fmt_vid_cap: fthd_v4l2_ioctl_try_fmt_vid_cap: 640x480
[Mon Dec 30 03:36:05 2019] facetimehd:fthd_v4l2_ioctl_try_fmt_vid_cap: fthd_v4l2_ioctl_try_fmt_vid_cap: 640x480
[Mon Dec 30 03:36:05 2019] facetimehd:fthd_v4l2_ioctl_try_fmt_vid_cap: fthd_v4l2_ioctl_try_fmt_vid_cap: 640x480
[Mon Dec 30 03:36:05 2019] facetimehd:fthd_v4l2_ioctl_try_fmt_vid_cap: fthd_v4l2_ioctl_try_fmt_vid_cap: 640x480
[Mon Dec 30 03:36:05 2019] facetimehd:fthd_v4l2_ioctl_try_fmt_vid_cap: fthd_v4l2_ioctl_try_fmt_vid_cap: 640x480
[Mon Dec 30 03:36:05 2019] facetimehd:fthd_v4l2_ioctl_try_fmt_vid_cap: fthd_v4l2_ioctl_try_fmt_vid_cap: 640x480
[Mon Dec 30 03:36:05 2019] facetimehd:fthd_v4l2_ioctl_try_fmt_vid_cap: fthd_v4l2_ioctl_try_fmt_vid_cap: 640x480
[Mon Dec 30 03:36:07 2019] facetimehd:fthd_v4l2_ioctl_try_fmt_vid_cap: fthd_v4l2_ioctl_try_fmt_vid_cap: 640x480
[Mon Dec 30 03:36:07 2019] facetimehd:fthd_v4l2_ioctl_s_fmt_vid_cap: YUYV
[Mon Dec 30 03:36:08 2019] facetimehd:fthd_v4l2_ioctl_g_fmt_vid_cap: fthd_v4l2_ioctl_g_fmt_vid_cap
[Mon Dec 30 03:36:08 2019] facetimehd:fthd_buffer_queue_setup: using 4 buffers
[Mon Dec 30 03:36:08 2019] facetimehd:fthd_buffer_prepare: 00000000fa4f4f70
[Mon Dec 30 03:36:08 2019] facetimehd:fthd_buffer_cleanup: 00000000fa4f4f70
[Mon Dec 30 03:36:08 2019] facetimehd:fthd_buffer_cleanup: 000000009c205222
[Mon Dec 30 03:36:08 2019] facetimehd:fthd_buffer_cleanup: 00000000d2f26eb9
[Mon Dec 30 03:36:08 2019] facetimehd:fthd_buffer_cleanup: 00000000d11b7a7f
[Mon Dec 30 03:36:08 2019] facetimehd:fthd_v4l2_ioctl_g_fmt_vid_cap: fthd_v4l2_ioctl_g_fmt_vid_cap
[Mon Dec 30 03:36:08 2019] facetimehd:fthd_v4l2_ioctl_g_fmt_vid_cap: fthd_v4l2_ioctl_g_fmt_vid_cap
```

At I needed to reload the driver with ```modprobe -r facetimehd && modprobe facetimehd``` to get out of this hung state.

Does anyone have any ideas as to why this driver isn't functioning properly?

leifliddy, 2019-12-30:
I should have probably mentioned that I modified these lines in 
```fthd_v4l2.c``` to lower the max resolution to 480p:
```
#define FTHD_MAX_WIDTH 640
#define FTHD_MAX_HEIGHT 480
```

leifliddy, 2019-12-31:
Since this camera is only 480p, should I modify the values in size/height values in these files?
```
fthd_isp.c:	    dev_priv->fmt.fmt.height < 720) {
fthd_isp.c:	ret = fthd_isp_cmd_channel_crop_set(dev_priv, 0, x1, 0, x2, 720);
fthd_v4l2.c:	dev_priv->fmt.fmt.sizeimage = 1280 * 720 * 2;
fthd_v4l2.c:	dev_priv->fmt.fmt.height = 720;
fthd_v4l2.c.orig:#define FTHD_MAX_HEIGHT 720
```

leifliddy, 2019-12-31:
holy s**t I got the camera working on my macbook9,1!!!!!

just needed to change the following values to :
**fthd_isp.c**
```
        if (dev_priv->fmt.fmt.width < 640 ||
            dev_priv->fmt.fmt.height < 480) {
                x1 = 80;
                x2 = 480;
        } else {
                x1 = 0;
                x2 = 640;
        }
    
    ret = fthd_isp_cmd_channel_crop_set(dev_priv, 0, x1, 0, x2, 480);
```
**fthd_v4l2.c**
```
        dev_priv->fmt.fmt.sizeimage = 640 * 480 * 2;
        dev_priv->fmt.fmt.pixelformat = V4L2_PIX_FMT_YUYV;
        dev_priv->fmt.fmt.width = 640;
        dev_priv->fmt.fmt.height = 480;
```

leifliddy, 2019-12-31:
I also found a way to extract the sensor calibration files from a modern version of ```AppleCamera.sys```
https://github.com/patjak/facetimehd-firmware/issues/3

Where are these three files:
```
1671_01XX.dat
1674_01XX.dat
1675_01XX.dat
```
I might write a script in the future that will extract these files automatically from any version of ```AppleCamera.sys``` 
It'll do this by analyzing a hexdump of this file and matching on these file signatures.

leifliddy, 2019-12-31:
@patjak would there a way to modify the logic in ```fthd_isp.c``` and ```fthd_v4l2.c``` to use the boardname value 
```board = dmi_get_system_info(DMI_BOARD_NAME);```
to match the model of macbook being used and then set the height/width/size parameters accordingly?

leifliddy, 2019-12-31:
[image]

leifliddy, 2019-12-31:
So, qv4ls works perfect and shows a frame rate of 30 fps. 
However, cheese doesn't launch due to frame rate issues:
```
[42876.829973] facetimehd:fthd_v4l2_ioctl_enum_frameintervals: fthd_v4l2_ioctl_enum_frameintervals
```
```mpv av://v4l2:/dev/video0```  works, but video is really choppy

How can I specify a 30fps frame rate in the driver?

leifliddy, 2020-01-01:
I managed to get the frame rate issue sorted out by changing the frame time value in ```fthd_drv.c```
to 33 (1000 / 33 = 30.3)
``` 
dev_priv->frametime = 33; /* 30 fps */
```
...but the frame dropping issue continues in mpv. I'm still trying to figure out how this driver works. 
One of the issues is that this camera records in 480p, but the aspect ratio of the screen is 16:9
The ```qv4l2``` application can display the video in a 16:9 ratio window, but I'm wondering how I would do that in the driver itself. Does anyone know how I could change the video aspect ratio in the driver?
**I'm surprised Apple put such a low quality camera in this macbook. The overall image quality is really bad.

leifliddy, 2020-01-01:
[image]
I've made some progress on getting ```cheese``` to work .
So, cheese will work properly for the root user, but not for my normal user.
Here is the dconf cheese config of my normal user. 
It always sets the video-y-resolution to ```1280```
If I manually change this value, the next time I open ```cheese``` it changes the value back.

leifliddy, 2020-01-01:
[image]
For the root user, the video-y-resolution is set to ```720```
I don't understand where cheese is getting these values from. 
Why are these values different between my normal user and root user?

leifliddy, 2020-01-01:
Ahhhh, ok I see
It should be setting this value according to the max values in ```fthd_v4l2.c```
```
#define FTHD_MAX_WIDTH 640
#define FTHD_MAX_HEIGHT 480
#define FTHD_MIN_WIDTH 320
#define FTHD_MIN_HEIGHT 240
```
So, I've just changed the max_width + max_height from 1280*720 to 640*480
Now root's dconf cheese config file shows 640*480 for the x+y resolution
However, the y-resolution for my normal user is now ```640``` it should be ```480```
weird, it's somehow translating the x+y max resolution wrong.
It looks fine from the command line:
 ```
v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Stepwise 320x240 - 640x480 with step 8/1
	[1]: 'YVYU' (YVYU 4:2:2)
		Size: Stepwise 320x240 - 640x480 with step 8/1
```

```
ffmpeg -f video4linux2 -list_formats all -i /dev/video0
[video4linux2,v4l2 @ 0x556115fdad00] Raw       :     yuyv422 :           YUYV 4:2:2 : {320-640, 8}x{240-480, 1}
[video4linux2,v4l2 @ 0x556115fdad00] Raw       : Unsupported :           YVYU 4:2:2 : {320-640, 8}x{240-480, 1}
```

patjak, 2020-01-03:
Hi, great work!
The firmware you have extracted is version 1.57. The one we've been using is 1.43. The reason we use the old firmware is because it's the only one we can download from Apple and extract. We are not allowed to redistribute the firmware. Can you try the 1.43 version and see if that works as well? It would be great if we could find a way to download the latest bootcamp driver without having to install bootcamp first.

Ok, as you say, we could check DMI and change max resolution based on that but perhaps the ISP can tell us the same thing. It might be as simple as to check the sensor id. I'll have a look at this.

Older versions of cheese (and plenty of other apps) doesn't support stepwise frame sizes and/or stepwise frame intervals. The only solution to get support for all applications seems to be to go back to discrete values. I'm planning to do this when time allows.

leifliddy, 2020-01-03:
Cools, thanks! Yes, firmware 1.43 works without issue on the macbook9,1.

maximeplante, 2020-08-04:
The change to fthd_isp.c and fthd_v4l2.c worked. Thanks!!!

xaiki, 2020-12-31:
I have a 10,1, and the changes proposed here don't work on it.
but even without the patches i can't get the light to turn on when i try to use the camera
```
┌[~/src/Mac/bcwc_pcie:master⚡☡]
└[pedrera] sudo insmod ./facetimehd.ko                                                         06:07:34
[ 4384.172881] facetimehd 0000:03:00.0: Found FaceTime HD camera with device id: 1570
[ 4384.173210] facetimehd 0000:03:00.0: can't disable ASPM; OS doesn't have ASPM control
[ 4384.173421] facetimehd 0000:03:00.0: Setting 64bit DMA mask
[ 4384.180235] facetimehd 0000:03:00.0: S2 PCIe link init succeeded
[ 4384.180268] facetimehd 0000:03:00.0: Refclk: 24MHz (0x2
[ 4384.190283] facetimehd 0000:03:00.0: PLL reset finished
[ 4384.190284] facetimehd 0000:03:00.0: Waiting for S2 PLL to lock at 450 MHz
[ 4384.190297] facetimehd 0000:03:00.0: S2 PLL is locked after 10 us
[ 4384.200346] facetimehd 0000:03:00.0: S2 PLL is in bypass mode
[ 4384.220426] facetimehd 0000:03:00.0: DDR40 PHY PLL locked on safe settings
[ 4384.220439] facetimehd 0000:03:00.0: STRAP valid
[ 4384.220440] facetimehd 0000:03:00.0: Configuring DDR PLLs for 450 MHz
[ 4384.220447] facetimehd 0000:03:00.0: DDR40 PLL is locked after 0 us
[ 4384.220456] facetimehd 0000:03:00.0: First DDR40 VDL calibration completed after 2 us
[ 4384.220463] facetimehd 0000:03:00.0: Second DDR40 VDL calibration completed after 1 us
[ 4384.220464] facetimehd 0000:03:00.0: Using step size 137
[ 4384.220468] facetimehd 0000:03:00.0: VDL set to: coarse=0x10007, fine=0x10117
[ 4384.220472] facetimehd 0000:03:00.0: Virtual VTT enabled
[ 4384.241077] facetimehd 0000:03:00.0: S2 DRAM memory address: 0x22159559
[ 4384.241091] facetimehd 0000:03:00.0: Rewrite DDR mode registers succeeded
[ 4384.241297] facetimehd 0000:03:00.0: Full memory verification succeeded! (0)
[ 4384.241807] facetimehd 0000:03:00.0: firmware: direct-loading firmware facetimehd/firmware.bin
[ 4384.352174] facetimehd 0000:03:00.0: Loaded firmware, size: 1392kb
[ 4384.509192] facetimehd 0000:03:00.0: ISP woke up after 0ms
[ 4384.509199] facetimehd 0000:03:00.0: Number of IPC channels: 7, queue size: 44865
[ 4384.509285] facetimehd 0000:03:00.0: Firmware requested heap size: 3072kb
[ 4384.519352] facetimehd 0000:03:00.0: ISP second int after 0ms
[ 4384.519355] facetimehd 0000:03:00.0: Channel description table at 00800000
[ 4384.529814] facetimehd 0000:03:00.0: magic value: 00000000 after 0 ms
[ 4384.529815] facetimehd 0000:03:00.0: Enabling interrupts
[ 4384.530124] FWMSG: 
[ 4384.530163] FWMSG: - APOLLO-ISP-APPLE ------------------------------------------------------------
[ 4384.530173] FWMSG: 
[ 4384.530191] FWMSG:   Restart count : 0
[ 4384.530225] FWMSG:   Platform : mode = TARGET, ID = 0x00000000, HW = 00020018.00000000
[ 4384.530246] FWMSG:   System Clock : 200000000 Hz
[ 4384.530265] FWMSG:   Processor mode : SUPERVISOR
[ 4384.530286] FWMSG:   Cache architecture type : SEPARATE
[ 4384.530304] FWMSG:   Cache type : WRITEBACK
[ 4384.530322] FWMSG:   Data Cache Line : 32 [0x20]
[ 4384.530341] FWMSG:   Boot arguments entries : 2
[ 4384.530360] FWMSG:     0000: 0x00000000 0x00000000
[ 4384.530387] FWMSG:   Physical memory base : 0x00000000 [TLB base 0x00160000]
[ 4384.530407] FWMSG:   Main memory :   base : 0x00000000
[ 4384.530435] FWMSG:                   size : 8388608 [0x00800000] [8.0 MB]
[ 4384.530462] FWMSG:   Extra heap :    base : 0x2080b000 [phy = 0x0080b000]
[ 4384.530496] FWMSG:                   size : 3145728 [0x00300000] [3.0 MB]
[ 4384.530521] FWMSG:   Shared window : base : 0x00800000 [static wiring]
[ 4384.530636] FWMSG:                   size : 125829120 [0x07800000] [120.0 MB]
[ 4384.530657] FWMSG:   Shared memory : base : 0x00800000
[ 4384.530687] FWMSG:                   size : 260046848 [0x0f800000] [248.0 MB]
[ 4384.530724] FWMSG:   TEXT : 1421992 [0x15b2a8] - text 667256, cstring  38749, const 715528
[ 4384.530808] FWMSG:   DATA : 8076 [0x1f8c] - data 0, bss 460, common 1452, noinit 6144
[ 4384.530832] FWMSG:   Heap free space : 10002626 [0x0098a0c2]
[ 4384.530854] FWMSG:   Heap allocated space : 94656 [0x000171c0]
[ 4384.530888] FWMSG:   Disclaimer : Copyright (c) APPLE Inc. All Rights Reserved.
[ 4384.530911] FWMSG:   Application : adc [release]
[ 4384.530935] FWMSG:   Linked on : Jul 25 2015 - 08:48:55
[ 4384.530959] FWMSG:   Release : S2ISP-01.43.00
[ 4384.530981] FWMSG:   H4ISPAPPLE : 11536
[ 4384.530997] FWMSG:   H4ISPCD : 4081
[ 4384.531011] FWMSG:   ffw : 4143
[ 4384.531038] FWMSG:   Tool-chain : iPhone OS - 7.0.3 [clang/clang++]
[ 4384.531049] FWMSG: 
[ 4384.531084] FWMSG: -------------------------------------------------------------------------------
[... 31 lines cut ...]
[ 4387.016693] FWMSG: [ISP] CH = 0 CMD = 0x0208 [CISP_CMD_CH_AE_FRAME_RATE_MAX_SET]
[ 4387.016799] FWMSG: [DSC] CH = 0   AE Frame Rate Max Set [10240] 
[ 4387.016944] FWMSG: [ISP] CH = 0 CMD = 0x020a [CISP_CMD_CH_AE_FRAME_RATE_MIN_SET]
[ 4387.017116] FWMSG: [DSC] CH = 0   AE Frame Rate Min Set [10240] 
[ 4387.017499] FWMSG: [ISP] CH = 0 CMD = 0xc100 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_START]
[ 4387.017887] FWMSG: [ISP] CH = 0 CMD = 0xc102 [CISP_CMD_APPLE_CH_MOTION_HISTORY_START]
[ 4387.018255] FWMSG: [ISP] CH = 0 CMD = 0xc113 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_ENABLE]
[ 4387.018407] FWMSG: [ISP] CH = 0 CMD = 0x8106 [CISP_CMD_APPLE_CH_STREAMING_MODE_SET]
[ 4387.018430] FWMSG: [DSC] CH = 0   Streaming Mode Set [0]
[ 4387.018590] FWMSG: [ISP] CH = 0 CMD = 0x0b02 [CISP_CMD_CH_SCALER_BRIGHTNESS_SET]
[... cut ...]
Exiting... (Quit)
[ 4393.005849] FWMSG: [ISP] CH = 0 CMD = 0x0d04 [CISP_CMD_CH_FACE_DETECTION_DISABLE]
[ 4393.006395] FWMSG: [ISP] CH = 0 CMD = 0xc114 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_DISABLE]
[ 4393.006722] FWMSG: [ISP] CH = 0 CMD = 0xc103 [CISP_CMD_APPLE_CH_MOTION_HISTORY_STOP]
[ 4393.007051] FWMSG: [ISP] CH = 0 CMD = 0xc101 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_STOP]
┌[~/src/Mac/bcwc_pcie:master⚡☡]
└[pedrera]                                                                                     06:07:46

```

taro-watanabe, 2023-08-13:
thanks for the help, I've also managed to make it work on macbook8,1, and the only additional change I had to apply was the 

```
#define FTHD_MAX_WIDTH 640
#define FTHD_MAX_HEIGHT 480
```

to fthd_v4l2.c file, so maybe give it a shot if anyone's struggling.

roogler, 2024-05-21:
I got the camera working on a Macbook8,1 running Ubuntu 24.04, kernel  6.8.0-31.

The firmware was extracted from `/System/Library/Extensions/AppleCameraInterface.kext/Contents/MacOS/AppleCameraInterface` taken from `Catalina`, using `binwalk` as per [](https://github.com/patjak/facetimehd/issues/196#issue-543699055).

`fthd_isp.c` and `fthd_v4l2.c` were edited as per [](https://github.com/patjak/facetimehd/issues/196#issuecomment-569858722).

With that firmware and those edits to the driver source, the camera works in `cheese` and `qv4l2`.

After the firmware has loaded it looks for `1675_01XX.dat,` and reports a failure if it's not present.  Although it's not necessary, I find that the .dat file does improve the colours, and may make the picture slightly less noisy.  It can be obtained as follows:

Download the `FaceTime Camera Driver Update` from [](https://download.info.apple.com/Mac_OS_X/031-29274-201500728-1C1B8542-358F-11E5-A861-8B973953DB92/AppleBcUpdate.exe).

Unpack AppleBcUpdate.exe:  `unrar x AppleBcUpdate.exe` .

Unpack `AppleCamera64.exe`: `unrar x AppleCamera64.exe`.

As far as I can tell, `1675_01XX.dat` is the data in AppleCamera.sys between offset 0x00183eb0 and 0x00188530:

`dd if=AppleCamera.sys of=1675_01XX.dat bs=1 skip=1588912 count=18048`

Copy `1675_01XX.dat `to /`usr/lib/firmware/facetimehd`.

y-tzk, 2025-02-22:
It's not working kernel 6.13 endeavour os & manjaro macbook 12 inch.

It works good before I was updating
but now did'nt change flame sizes to 640*480.
V4L2 Test Bench outputs green screen.

fthd_v4l2.c and fthd_isp.c are still modified 1280*720 to 640*480

```

$ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YVYU' (YVYU 4:2:2)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)

$ v4l2-ctl --list-devices    
Apple Facetime HD (PCI:0000:03:00.0):
        /dev/video0

```

```

dmesg
[ 1385.816274] FWMSG: ERR: ./H4ISPCD/filters/IC/CImageCaptureH4.cpp, 777: FlowIC00: SIF errors: sifIrq = 0x804!
[ 1387.803221] facetimehd 0000:03:00.0: IO: timeout

```

I'm not sure how to change it.
Any ideas?

y-tzk, 2025-02-27:
solved 

use
"""
https://github.com/patjak/facetimehd/compare/master...jakintosh:facetimehd:master
"""

tkalant, 2025-03-28:
The change to fthd_isp.c and fthd_v4l2.c also worked for me on a Macbook8,1 early 2015, running ubuntu 24.04 and linux 6.8 kernel. Thank you so much!

brniederberger, 2025-11-22:
@BrewCoffeeAddict 
Sure thing, the extraction process involved time with a hex editor and binwalk and bit of patience, but here's the basic steps.

In the facetimehd-firmware repository there is a  script https://github.com/patjak/facetimehd-firmware/blob/master/extract-firmware.sh that identifies the header and footer signature of the firmware.  First thing I did was to search for those using binwalk.

```
# Header Search - this is a very distinct string which occurs in bytes 4-32 of the firmware
$ binwalk -R "\xfe\xff\xff\xea\xfe\xff\xff\xea\xfe\xff\xff\xea\xfe\xff\xff\xea\xfe\xff\xff\xea\xfe\xff\xff\xea\xfe\xff\xff\xea" AppleCamera.sys 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
86836         0x15334         Raw signature (\xfe\xff\xff\xea\xfe\xff\xff\xea\xfe\xff\xff\xea\xfe\xff\xff\xea\xfe\xff\xff\xea\xfe\xff\xff\xea\xfe\xff\xff\xea)

# Footer Search, this string occurs more commonly
$ binwalk -R "\x00\x00\x00\x00\xff\xff\xff\xff" AppleCamera.sys 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
825020        0xC96BC         Raw signature (\x00\x00\x00\x00\xff\xff\xff\xff)
825084        0xC96FC         Raw signature (\x00\x00\x00\x00\xff\xff\xff\xff)
< Many rows removed for brevity>
842308        0xCDA44         Raw signature (\x00\x00\x00\x00\xff\xff\xff\xff)
1504044       0x16F32C        Raw signature (\x00\x00\x00\x00\xff\xff\xff\xff)
1505680       0x16F990        Raw signature (\x00\x00\x00\x00\xff\xff\xff\xff)
1508956       0x17065C        Raw signature (\x00\x00\x00\x00\xff\xff\xff\xff)
1982068       0x1E3E74        Raw signature (\x00\x00\x00\x00\xff\xff\xff\xff)
1982164       0x1E3ED4        Raw signature (\x00\x00\x00\x00\xff\xff\xff\xff)
1982316       0x1E3F6C        Raw signature (\x00\x00\x00\x00\xff\xff\xff\xff)
```
To select find the end point of firmware, I knew the firmware should be around 1.4M in size which meant it could be any of the three matches with a decimal offset of 150k.  I then examined those three locations using a hex editor.  I knew the firmware is padded out with nulls at the end, so the correct end signature should be preceded by nulls.  The only match to meet that criteria was at offset 1504044.

So we now know the start point (header match - 4, because signature is 4th byte) 86836 - 4 = 86832.  The total size is (Footer Offset + 8) - Start , (1504044 + 8) - 86832 = 1417220.  Which gives the command:  
`dd  bs=1 skip=86832 count=1417220 if=AppleCamera.sys of=firmware.bin`

Sensor calibration files were a little harder to find, but the files names are included in the firmware which is a of a bit clue. If you run binwalk without arguments you'll get the following output. 

```
$ binwalk AppleCamera.sys

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
0             0x0             Microsoft executable, portable (PE)
71531         0x1176B         ESP Image (ESP32): segment count: 6, flash mode: QUOUT, flash speed: 40MHz, flash size: 1MB, entry address: 0x848b4800, hash: none
84944         0x14BD0         Unix path: /usr/local/share/firmware/isp/9112_01XX.dat
90944         0x16340         Copyright string: "Copyright (c) APPLE Inc. All Rights Reserved."
919478        0xE07B6         JBOOT STAG header, image id: 0, timestamp 0x3C0000, image size: 759693312 bytes, image JBOOT checksum: 0x0, header JBOOT checksum: 0x1980
926350        0xE228E         JBOOT STAG header, image id: 0, timestamp 0x1C0000, image size: 759693312 bytes, image JBOOT checksum: 0x0, header JBOOT checksum: 0xB00
936326        0xE4986         JBOOT STAG header, image id: 0, timestamp 0x3C0000, image size: 724041728 bytes, image JBOOT checksum: 0x0, header JBOOT checksum: 0x4
947906        0xE76C2         JBOOT STAG header, image id: 0, timestamp 0x140000, image size: 724041728 bytes, image JBOOT checksum: 0x0, header JBOOT checksum: 0x0
< Many rows removed for brevity>
```
The third entry is a unix path. If you go to that offset in the file, you'll find a listing of all the calibration files in the firmware library.  I was unable to find a signature in the calibration files to figure out where they are, but the filename contains the key.  The first three hex bytes of sensor data are used to make the filename.  So the file 1871_01XX.dat starts with bytes \x18\x71\x01, go search for those.
```
$ binwalk -R "\x18\x71\x01" AppleCamera.sys 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
1511184       0x170F10        Raw signature (\x18\x71\x01)
1743168       0x1A9940        Raw signature (\x18\x71\x01)
```
There are two locations because the sensor files are repeated in the driver library, with all the files concatenated and the repeated a second time.  If you search for all of the sensor files you'll get a list of offsets and if you put them in sequential order you can figure out where they start and end.  For example, the second sensor file was 1874_01XX.dat.

```
$ binwalk -R "\x18\x74\x01" AppleCamera.sys 

DECIMAL       HEXADECIMAL     DESCRIPTION
--------------------------------------------------------------------------------
1530224       0x175970        Raw signature (\x18\x74\x01)
1762208       0x1AE3A0        Raw signature (\x18\x74\x01)
```
The difference between the start of the first file and start of the second file will give you the size of the sensor file, allowing you to extract.  1530224 − 1511184 (or 1762208 - 1743168) = 19040. Allowing you to create the dd command:

`dd bs=1 skip=1511184 count=19040 if=AppleCamera.sys of=1871_01XX.dat`

As a bonus, if you go to binwalk location which is described as the Apple Copyright string, you can see the driver version number (e.g 01.55.00).

doctor, 2026-08-17:
The 12-inch MacBook camera works now — verified end to end on a MacBook10,1
(848x588 at 30fps, no SIF errors). Rather than duplicate it here, the full
writeup is in #178.

Short version, since this issue is where people tend to land:

1. **Firmware 1.43.0 predates these machines** and never detects the sensor
   (`sensor count: 0`, "Sensor is null after hNVStorage Validate"). Firmware
   5.60.0 from macOS 10.12.6 is required — patjak/facetimehd-firmware#12.
2. **`1675_01XX.dat` is shipped by nobody.** It is not in the Boot Camp driver
   everyone extracts from; all eleven set files are in macOS's
   `AppleCameraAssistant` instead. Offsets and validation in
   patjak/facetimehd-firmware#3.
3. **The hardcoded 1280x720 crop** is already fixed by 98b55fd, but that commit
   is in no tagged release — so distro packages built from 0.7.0.2 still fail
   with `sifIrq = 0x804` followed by `IO: timeout`. Build from master.

All three are needed together; any one missing still yields no video, which is
probably why this looked like unsupported hardware for so long.