patjak/facetimehd #200: Fedora error: "There was an error playing video from the webcam"

miguelmota, 2020-02-20:
Hi, followed instructions as explained in [installation wiki](https://github.com/patjak/bcwc_pcie/wiki/Get-Started) but when I run the cheese program I get "There was an error playing video from the webcam"

I'm on Fedora 31

Any help is appreciated. Thanks

GiovanniGrieco, 2020-03-31:
Here I am on Fedora 31 with Kernel 5.5.11, MacBook Pro 13" Early 2015. Same issue: Cheese prints out an error, Microsoft Teams shows black screen, but `mplayer` and Google Chrome work nominally.

Any hint on how we can debug such problem? `dmesg` doesn't seem to print out anything suspicious.

<details> 
  <summary><code>facetimehd</code> output </summary>
<pre>
[  128.708968] facetimehd: loading out-of-tree module taints kernel.
[  128.709029] facetimehd: module verification failed: signature and/or required key missing - tainting kernel
[  128.709575] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[  128.709713] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[  128.714742] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[  128.714769] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[  128.724781] facetimehd 0000:02:00.0: PLL reset finished
[  128.724782] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[  128.724794] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[  128.734800] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[  128.754836] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[  128.754849] facetimehd 0000:02:00.0: STRAP valid
[  128.754850] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[  128.754857] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[  128.754866] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[  128.754876] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[  128.754876] facetimehd 0000:02:00.0: Using step size 159
[  128.754881] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011c
[  128.754885] facetimehd 0000:02:00.0: Virtual VTT enabled
[  128.775402] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[  128.775416] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[  128.775610] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[  128.893170] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[  129.039678] facetimehd 0000:02:00.0: ISP woke up after 0ms
[  129.039683] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[  129.039686] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[  129.049700] facetimehd 0000:02:00.0: ISP second int after 0ms
[  129.049702] facetimehd 0000:02:00.0: Channel description table at 00800000
[  129.059949] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[  129.059950] facetimehd 0000:02:00.0: Enabling interrupts
[ 1524.160524] facetimehd 0000:02:00.0: fthd_hw_deinit
[ 1530.084704] facetimehd 0000:02:00.0: Found FaceTime HD camera with device id: 1570
[ 1530.084842] facetimehd 0000:02:00.0: Setting 64bit DMA mask
[ 1530.089894] facetimehd 0000:02:00.0: S2 PCIe link init succeeded
[ 1530.089921] facetimehd 0000:02:00.0: Refclk: 25MHz (0xa)
[ 1530.099932] facetimehd 0000:02:00.0: PLL reset finished
[ 1530.099934] facetimehd 0000:02:00.0: Waiting for S2 PLL to lock at 450 MHz
[ 1530.099946] facetimehd 0000:02:00.0: S2 PLL is locked after 10 us
[ 1530.109952] facetimehd 0000:02:00.0: S2 PLL is in bypass mode
[ 1530.129993] facetimehd 0000:02:00.0: DDR40 PHY PLL locked on safe settings
[ 1530.130009] facetimehd 0000:02:00.0: STRAP valid
[ 1530.130011] facetimehd 0000:02:00.0: Configuring DDR PLLs for 450 MHz
[ 1530.130019] facetimehd 0000:02:00.0: DDR40 PLL is locked after 0 us
[ 1530.130029] facetimehd 0000:02:00.0: First DDR40 VDL calibration completed after 2 us
[ 1530.130040] facetimehd 0000:02:00.0: Second DDR40 VDL calibration completed after 2 us
[ 1530.130041] facetimehd 0000:02:00.0: Using step size 159
[ 1530.130046] facetimehd 0000:02:00.0: VDL set to: coarse=0x10008, fine=0x1011c
[ 1530.130051] facetimehd 0000:02:00.0: Virtual VTT enabled
[ 1530.150575] facetimehd 0000:02:00.0: S2 DRAM memory address: 0x22159559
[ 1530.150588] facetimehd 0000:02:00.0: Rewrite DDR mode registers succeeded
[ 1530.150791] facetimehd 0000:02:00.0: Full memory verification succeeded! (0)
[ 1530.265562] facetimehd 0000:02:00.0: Loaded firmware, size: 1392kb
[ 1530.417974] facetimehd 0000:02:00.0: ISP woke up after 0ms
[ 1530.417980] facetimehd 0000:02:00.0: Number of IPC channels: 7, queue size: 44865
[ 1530.417996] facetimehd 0000:02:00.0: Firmware requested heap size: 3072kb
[ 1530.428016] facetimehd 0000:02:00.0: ISP second int after 0ms
[ 1530.428019] facetimehd 0000:02:00.0: Channel description table at 00800000
[ 1530.438427] facetimehd 0000:02:00.0: magic value: 00000000 after 0 ms
[ 1530.438430] facetimehd 0000:02:00.0: Enabling interrupts
</pre>
</details>

<details> 
[... 440 lines cut ...]
[ 2453.726005] FWMSG: [ISP] CH = 0 CMD = 0x0115 [CISP_CMD_CH_SIF_PIXEL_FORMAT_SET]
[ 2453.726129] FWMSG: [ISP] CH = 0 CMD = 0x0124 [CISP_CMD_CH_CAMERA_ERR_HANDLE_CONFIG]
[ 2453.726245] FWMSG: [ISP] CH = 0 CMD = 0x0d05 [CISP_CMD_CH_FACE_DETECTION_ENABLE]
[ 2453.726441] FWMSG: [ISP] CH = 0 CMD = 0x0d00 [CISP_CMD_CH_FACE_DETECTION_START]
[ 2453.726652] FWMSG: [ISP] CH = 0 CMD = 0x0208 [CISP_CMD_CH_AE_FRAME_RATE_MAX_SET]
[ 2453.726807] FWMSG: [DSC] CH = 0   AE Frame Rate Max Set [8448] 
[ 2453.726920] FWMSG: [ISP] CH = 0 CMD = 0x020a [CISP_CMD_CH_AE_FRAME_RATE_MIN_SET]
[ 2453.727025] FWMSG: [DSC] CH = 0   AE Frame Rate Min Set [8448] 
[ 2453.727153] FWMSG: [ISP] CH = 0 CMD = 0xc100 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_START]
[ 2453.727385] FWMSG: [ISP] CH = 0 CMD = 0xc102 [CISP_CMD_APPLE_CH_MOTION_HISTORY_START]
[... cut ...]
[ 2465.530210] FWMSG: [ISP] CH = 0 CMD = 0x0104 [CISP_CMD_CH_BUFFER_RETURN]
[ 2465.530935] FWMSG: [ISP] CH = 0 CMD = 0x0d01 [CISP_CMD_CH_FACE_DETECTION_STOP]
[ 2465.531133] FWMSG: [ISP] CH = 0 CMD = 0x0d04 [CISP_CMD_CH_FACE_DETECTION_DISABLE]
[ 2465.531317] FWMSG: [ISP] CH = 0 CMD = 0xc114 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_DISABLE]
[ 2465.531533] FWMSG: [ISP] CH = 0 CMD = 0xc103 [CISP_CMD_APPLE_CH_MOTION_HISTORY_STOP]
[ 2465.531703] FWMSG: [ISP] CH = 0 CMD = 0xc101 [CISP_CMD_APPLE_CH_TEMPORAL_FILTER_STOP]
</pre>
</details>

kmzby, 2020-04-07:
Exactly the same as @GiovanniGrieco reports: Manjaro, 5.4.28-1-MANJARO, MBP 13" Early 2015 (12.1). Camera works just fine in Blink-based browsers, mplayer and V4L Test Bench. However, Teams, VLC and Kamoso only show a black screen. VLC stderr: ```Invalid PCR value in ES_OUT_SET_(GROUP_)PCR !```

toger5, 2020-04-14:
Dont knwo if that is of any help but I have the same issue.

toger5, 2020-05-23:
This issue only happend to me on cheese app. I updated to fedora 32 and cheese 3.34. Now it works fine. So this is most probably not an issue with bcwc_pcie but with cheese which is fixed now.

keplersj, 2020-10-15:
I'm experiencing this issue on Manjaro as well. Gnome Cheese and Blink-based browsers are able to get video from the camera, while Microsoft Teams (AUR and Flatpak) aren't. It seems strange strange to me given the Teams app is just an Electron application.

Please let me know what is needed to debug this issue and how we can fix it.

kmzby, 2021-01-25:
Same reported in https://github.com/patjak/bcwc_pcie/issues/198