patjak/facetimehd #206: Fedora packaging

gaul, 2020-04-07:
The [Fedora packaging](https://github.com/ktdreyer/facetimehd-kmod-rpm) maintainer no longer updates his repo.  I wonder if someone else in the community can take this up, preferably in the main Fedora repo if possible, otherwise in the non-free RPM Fusion?

frgt10, 2020-05-16:
Hi! I made dkms package. It can be installed from [copr repo](https://copr.fedorainfracloud.org/coprs/frgt10/facetimehd-dkms/).

gaul, 2020-05-18:
That's great, thank you!  Can you update the wiki:

https://github.com/patjak/bcwc_pcie/wiki/Get-Started#get-started-on-fedora

After you get some feedback, do you think you could try to get this into the main Fedora repo?

frgt10, 2020-05-18:
> That's great, thank you! Can you update the wiki:
> 
> https://github.com/patjak/bcwc_pcie/wiki/Get-Started#get-started-on-fedora
 
Wiki updated.

> After you get some feedback, do you think you could try to get this into the main Fedora repo?

Not sure if the package can be added to the main Fedora repository. I will explore the possibility.

LyesSaadi, 2020-06-30:
Hello :)!

> After you get some feedback, do you think you could try to get this into the main Fedora repo?

This is prohibited:
https://docs.fedoraproject.org/en-US/packaging-guidelines/what-can-be-packaged/#_no_external_kernel_modules

We could (should?) try contacting the Kernel team, but I don't know how probable it is they accept it.

But it should be allowed on rpmfusion! Copr is also a great choice generally for hosting kernel modules.