patjak/facetimehd #212: macbook pro mid 2014, ubuntu, failure on direct firmware load

TeaWolf, 2020-05-02:
After following the getting started guide, I'm left with the following error in dmesg

```
 facetimehd 0000:04:00.0: Direct firmware load for facetimehd/1871_01XX.dat failed with error -2
```

I'm on a Macbook pro mid 2014 15"
Ubuntu 18.04.2 LTS
Linux kernel version 5.3.0-51-generic

The full output from dmesg is in this file [dmesg-out.txt](https://github.com/patjak/bcwc_pcie/files/4567599/dmesg-out.txt) (Error in question is at the very end).
Here's the output from lsmod [lsmod-out.txt](https://github.com/patjak/bcwc_pcie/files/4567603/lsmod-out.txt)

I'm fairly new to linux so I don't know how to go about this...

maxgio92, 2020-05-11:
@TeaWolf you could ignore that. Does the driver then works fine for you?

TeaWolf, 2020-05-11:
it wasn't working at the time.. but has magically started to now....

closed 2020-05-11 by TeaWolf