patjak/facetimehd #213: Module facetimehd not found in directory /lib/modules/5.4.0-7626-generic

azcn2503, 2020-05-15:
Hello,

Firstly, thank you for taking the time to reverse engineer this driver and provide as much as possible to the community.

I'm having an issue setting this up on a Pop!_OS 20.04, a Debian/Ubuntu derived distribution.

Here's my terminal input and output, if it helps:

```
casey@pop-os:~/git$ git clone https://github.com/patjak/facetimehd-firmware.git
Cloning into 'facetimehd-firmware'...
remote: Enumerating objects: 1, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 886 (delta 0), reused 1 (delta 0), pack-reused 885
Receiving objects: 100% (886/886), 290.76 KiB | 791.00 KiB/s, done.
Resolving deltas: 100% (585/585), done.
casey@pop-os:~/git$ cd facetimehd-firmware/
casey@pop-os:~/git/facetimehd-firmware$ make

Checking dependencies for driver download...
/usr/bin/curl
/usr/bin/xzcat
/usr/bin/cpio

Downloading the driver, please wait...

Found matching hash from OS X, El Capitan 10.11.5
==> Extracting firmware...
 --> Decompressing the firmware using gzip...
 --> Deleting temporary files...
 --> Extracted firmware version 1.43.0

casey@pop-os:~/git/facetimehd-firmware$ sudo make install
Copying firmware into '//usr/lib/firmware/facetimehd'
casey@pop-os:~/git/facetimehd-firmware$ sudo apt install linux-headers-5.4.0-7626-generic git kmod libssl-dev checkinstall
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.25.1-1ubuntu3).
kmod is already the newest version (27-1ubuntu2).
libssl-dev is already the newest version (1.1.1f-1ubuntu2).
checkinstall is already the newest version (1.6.2+git20170426.d24a630-2ubuntu1).
linux-headers-5.4.0-7626-generic is already the newest version (5.4.0-7626.30~1588169883~20.04~bbe668a).
The following packages were automatically installed and are no longer required:
  fonts-liberation2 fonts-opensymbol gstreamer1.0-gtk3 libabw-0.1-1
  libboost-date-time1.71.0 libboost-filesystem1.71.0
  libboost-iostreams1.71.0 libboost-locale1.71.0 libcdr-0.1-1
  libclucene-contribs1v5 libclucene-core1v5 libcmis-0.5-5v5
  libcolamd2 libe-book-0.1-1 libeot0 libepubgen-0.1-1
  libetonyek-0.1-1 libexttextcat-2.0-0 libexttextcat-data
  libfreehand-0.1-1 libgpgmepp6 libjuh-java libjurt-java
  liblangtag-common liblangtag1 libmhash2 libmspub-0.1-1
  libmwaw-0.3-3 libmythes-1.2-0 libneon27-gnutls libodfgen-0.1-1
  liborcus-0.15-0 libpagemaker-0.0-0 libraptor2-0 librasqal3 librdf0
  librevenge-0.0-0 libridl-java libsuitesparseconfig5 libuno-cppu3
  libuno-cppuhelpergcc3-3 libuno-purpenvhelpergcc3-3 libuno-sal3
  libuno-salhelpergcc3-3 libunoloader-java libvisio-0.1-1
  libwpd-0.10-10 libwpg-0.3-3 libwps-0.4-4 libxmlsec1 libxmlsec1-nss
  libyajl2 lp-solve uno-libs-private ure
Use 'sudo apt autoremove' to remove them.
0 to upgrade, 0 to newly install, 0 to remove and 101 not to upgrade.
casey@pop-os:~/git/facetimehd-firmware$ sudo depmod
casey@pop-os:~/git/facetimehd-firmware$ sudo modprobe facetimehd
modprobe: FATAL: Module facetimehd not found in directory /lib/modules/5.4.0-7626-generic
```

There is some duplication of the instructions provided in the following section: https://github.com/patjak/bcwc_pcie/wiki/Get-Started#get-started-on-debian - some of these are already detailed above in the Firmware extraction section: https://github.com/patjak/bcwc_pcie/wiki/Get-Started#get-started-on-debian - so I skipped over the commands I had already run.

Is the sensor calibration an essential part of the installation?

Kind regards.

patjak, 2020-05-22:
Hi, you have missed the step where you install the driver (you're only installing the firmware)

The firmware repo is: https://github.com/patjak/facetimehd-firmware.git
The driver repo is: https://github.com/patjak/bcwc_pcie.git

Have a look at the docs again and you'll figure it out :)

azcn2503, 2020-05-22:
Thanks :smile:

closed 2020-05-22 by azcn2503