patjak/facetimehd #215: CPU usage?

gaul, 2020-05-21:
Has anyone compared this driver against the macOS one?  I routinely see 90+ degC temperatures during Google Meet and Zoom calls.  This throttles my MacBook 2013 down to 800 MHz which drops video and more annoyingly audio.

patjak, 2020-06-07:
Just displaying the video stream with qv4l2 shows cpu usage at ~15%. Doing the same with Photo Booth in OSX shows similar numbers. Perhaps video scaling for incoming / outgoing streams is not accelerated in Linux and this gives high CPU load?

gaul, 2020-06-09:
I did some tests with cheese and Firefox and it seems that h.264 is hardware accelerated for both.  However it seems that Google Meet uses VP9 or some other codec that my aging laptop does not hardware accelerate.

closed 2020-06-09 by gaul